zwave-js/zwave-js #4150: No Security on battery driven devices

wallenium, 2022-01-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Reinterview of insecure devices to force security. 

What do you observe?
The devices are included with "no security" but support S2 security. All of them are battery powered devices.

- Abus Flood Detector SHWM10000
- Popp Mold detector POPE701202
- Popp CO Detector 004407

What did you expect to happen?
Ehoud use S2 security or S0 if s2 is not supported

Steps to reproduce the behavior:
  1. Go to advanced, interview
  2. Use checkbox "Reset security classes"
  3. Go to the device and open tamper proof and start including mode to wake it up
  4. wait for the interview to finalize

### Device information

manufacturer: ABUS Security-Center GmbH & Co. KG (0x0403)
product type: 0x02
product id:   0x02

Node ID in your network: 008

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.11.1
`zwavejs2mqtt` branch: 6.4.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2022-01-31.log](https://github.com/zwave-js/node-zwave-js/files/7974817/zwavejs_2022-01-31.log)

AlCalzone, 2022-02-01:
> Reinterview of insecure devices to force security.

That's not how it works. The checkbox you mean is meant to reset the information zwave-js has about the security status, e.g. when it was incorrectly discovered.
To change the security status, you need to exclude and re-include the devices.

Aside from that, it looks like you're affected by https://github.com/zwave-js/node-zwave-js/issues/3906 and should do a firmware upgrade on the stick.

wallenium, 2022-02-01:
Hi,
Stick is upgraded.

Ok. Hmm. I read that you are planning to avoid exclude and include of the device in a future update...? I thought it is this checkbox 

https://github.com/zwave-js/node-zwave-js/issues/3227

So all I can do at the moment is to follow the dirty route.. I will try to do so ;)

AlCalzone, 2022-02-01:
> I read that you are planning to avoid exclude and include of the device in a future update...?

This is not possible - that's how Z-Wave works. The linked issue is also only a workaround which involves factory-resetting the device and replacing it with itself (replace failed node). You can do that already, but you should shut down the z-wave application you're using first, otherwise the reset could result in the node being removed.

closed 2022-02-05 by zwave-js-assistant[bot]