zwave-js/zwave-js #4165: Incorrect device config for Leviton DZMX1

dra2885, 2022-02-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Including a new Leviton DZMX1 or re-interviewing an existing DZMX1.

What do you observe?
The device is configured as a Scene Capable Push On/Off device when in fact it is a Scene Capable Push On/Off Dimmer. Re-interviewing doesn't help, as this is defined in the source config in Z-Wave JS DB.

What did you expect to happen?
Expected the device to be included as a dimmer and not a switch.

Steps to reproduce the behavior:
DB device link for ref: https://devices.zwave-js.io/?jumpTo=0x001d:0x1b03:0x0334:0.7#

I'd take a shot at editing the config file and submitting a PR, but I'm way out of my depth with z-wave device configuration...

### Device information

Manufacturer: Leviton
Model name: DZMX1

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.11.1
`zwavejs2mqtt` branch: 6.4.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

I don't have a relevant log file

dra2885, 2022-02-01:
Disregard .. while incorrect, upon further testing the issue appears to be merely cosmetic.

AlCalzone, 2022-02-01:
@dra2885 it might be that these two device fingerprints are actually two different devices:
https://github.com/zwave-js/node-zwave-js/blob/27e4bd7271fda55ce51d1def1796caf458552aaa/packages/config/config/devices/0x001d/dzmx1.json#L6-L16

The change would be cosmetical, but it still makes sense to have correct device labels. Which product ID/type does yours have and which device is it? Or is there just one and we have the wrong label?

dra2885, 2022-02-12:
@AlCalzone -- apologies for the delay in responding. The device name is correct (DZMX1), it's just the type that is wrong. This is a scene-capabale dimmer, not a switch.

Happy to provide other details if needed.

closed 2022-03-04 by AlCalzone