zwave-js/zwave-js #4169: [Bug]: Associating a second dimmer with a scene controller button-press adds unwanted commands to first dimmer

pethans, 2022-02-01:


AlCalzone, 2022-02-01:
Associations are handled by the devices without interaction of the controller. Z-Wave JS is only involved when setting up the associations. So this can only be a bug in the scene controller's firmware.

It could be that it confuses the endpoints of its associations, i.e. instead of sending a command to RGBW endpoint 5 and Dimmer endpoint 0, it sends the commands to RGBW endpoint 0 (i.e. all channels) and Dimmer endpoint 0.

pethans, 2022-02-01:
Thank you for this, I have submitted a ticket to Zooz with the above information and will report back. If they determine something is wrong with the way z2m is configuring the devices' associations, I will let you know.

closed 2022-02-19 by zwave-js-assistant[bot]