zwave-js/zwave-js #4170: [question] NVM size is 0 - How is the size determined?

andyboeh, 2022-02-01:


AlCalzone, 2022-02-01:
0 is not the NVM size, that's the length of the method argument array.

> Unknown NVM size

is because the response by the stick doesn't match any of the defined NVM sizes. 
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly`, try the backup again and attach the log here as a file.

andyboeh, 2022-02-01:
I redacted the home ID; looks like the manufacturer is reported as 0xff
[zwavejs_2022-01-31.log](https://github.com/zwave-js/node-zwave-js/files/7976588/zwavejs_2022-01-31.log)

AlCalzone, 2022-02-01:
```
2022-01-31T20:52:50.678Z DRIVER « [RES] [GetNVMId]
                                    manufacturer: 0xff
                                    memory type:  EEPROM
                                    memory size:  Unknown
```
That's the intersting bit. Z-Wave JS can't know how large the memory is, so it doesn't know how much it can read.
I'm not sure what the correct behavior in this case is and how the device would react to an out-of-bounds read (return nothing? wrap around?).

andyboeh, 2022-02-01:
Hm...does Z-Wave JS identify the memory size based on the ID returned by the chip? Is the memory external or on the SoC?

The Z-Wave module is a separate PCB, it looks like this: https://www.digikey.be/en/products/detail/silicon-labs/ZM5304AE-CME3R/4135431

I'm not sure what the smaller IC is. The markings are S8U S78 77, I couldn't find any information on that.

AlCalzone, 2022-02-01:
> does Z-Wave JS identify the memory size based on the ID returned by the chip

No, the size information is part of the response (the third `0xff`).
https://github.com/zwave-js/node-zwave-js/blob/3b116f8c95535ef6e561fa862bec9d0ddaf1a3f9/packages/zwave-js/src/lib/serialapi/nvm/GetNVMIdMessages.ts#L23-L37

AFAIK, the `get_NVM_id` command is the only way to get any information about the (external) NVM on 500 series controllers:
[image]

andyboeh, 2022-02-01:
Yeah, looks so. 
If I understood the concept correctly, the manufacturer should have programmed this information into the NVR flash page of the Z-Wave chip. I suppose that this information is then returned by the serial API and in my case the manufacturer didn't do that. Since there is then no reliable way of detecting the missing information, I'm screwed.

andyboeh, 2022-02-01:
I just found a tool "TestNVM.py" on github that should be able to dump the NVM. They seem to just dump 2Mbit: https://github.com/drzwave/TestNVM/blob/a57c42406c54b24609b3aa5e74d5ebf494f0506f/TestNVM.py#L411

I'll modify it for TCP and see what happens.

closed 2022-02-05 by zwave-js-assistant[bot]