zwave-js/zwave-js #4172: Missing device configuration: Kwikset 620 zwave 700 lock

MoogleTroupe, 2022-02-01:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Kwikset 620

### Manufacturer ID

0x0090

### Product Type

0x0811

### Product ID

0x23a8

### Firmware Version

7.20

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4184

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/4184/5069369-003.pdf

The biggest change from the earlier kwikset locks is that this one supports 250 user codes and has a zwave 700 chipset

AlCalzone, 2022-02-01:
Looks like we can just add this fingerprint to https://devices.zwave-js.io/?jumpTo=0x0090:0x0811:0x03a8:0.0

MoogleTroupe, 2022-02-01:
@AlCalzone yes it has 1 2 3 4 10 40 46 in the config parameters

MoogleTroupe, 2022-02-01:
it appears kwikset incremented both of the 620 locks. 0x0811:0x03a8 (Camelot style) and 0x0811:0x13a8 (contemporary style) show on the zwave website. being that 0x0811:0x23a8 is a new revision of the Camelot style, I don't have a new instance of the contemporary style lock in front of me but most likely 0x0811:0x33a8 is also going to be a valid fingerprint.

AlCalzone, 2022-02-01:
> most likely 0x0811:0x33a8 is also going to be a valid fingerprint

I think we're going to wait until someone confirms.

@zwave-js-bot add fingerprint 0x0811:0x23a8 to 0x0090:0x0811:0x03a8:0.0

PR #4173 (2022-02-01): fix(config): add fingerprint `0x0811:0x23a8` to "Kwikset HC620"
fixes: #4172
closed 2022-02-01 by AlCalzone