zwave-js/zwave-js #4174: [Bug]: Does not reconnect when the TCP connection is broken

andyboeh, 2022-02-01:


andyboeh, 2022-02-25:
We had a partial power outage today where both of my Z-Wave controllers lost power but the server running zwave-js stayed online (thanks to its UPS). Once again, I had to manually reconnect to the controllers.

vlycop, 2022-04-07:
So i found something, debugging with F11 is very inconsistent, i hate it.
When starting the debuger with a fake/down tcp ip, the `serial.on("close", ()` immediately trigger, but since it have a `[if (this.isOpen) {`](https://github.com/zwave-js/node-zwave-js/blob/966e51fc81eb7a1686f9ac94971a06bab5ed26f1/packages/serial/src/ZWaveSocket.ts#L21) i doesn't do anything more and silently failed.
I believe this cause a lack of 'error' that the UI should catch to try a restart.
Removing the if sadly still don't emit the error in the log but at least to go downsteam. i'll keep digging when i have some time

vlycop, 2022-04-07:
it is re-filtered by https://github.com/zwave-js/node-zwave-js/blob/966e51fc81eb7a1686f9ac94971a06bab5ed26f1/packages/zwave-js/src/lib/driver/Driver.ts#L788
```
				if (this.isSoftResetting && !this.serial?.isOpen) {
					// A disconnection while soft resetting is to be expected
					return;
				} else if (!this._isOpen) {
					// tryOpenSerialport takes care of error handling
					return;
				}
```
Removing it to does allow the error to come through.

```
Debugger attached.
Debugger attached.
Debugger attached.
13:42:50.368 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
13:42:50.371 DRIVER   version 9.0.1
13:42:50.372 DRIVER   
13:42:50.372 DRIVER   starting driver...
13:42:50.381 DRIVER   opening serial port tcp://192.168.20.63:5896
13:42:55.740 DRIVER   Serial port errored: connect ECONNREFUSED 192.168.20.63:5896
ZWaveError: Serial port errored: connect ECONNREFUSED 192.168.20.63:5896 (ZW0100)
    at ZWaveSocket.<anonymous> (/home/vlycop/Documents/Dev/node-zwave-js/packages/zwave-js/src/lib/driver/Driver.ts:792:19)
    at ZWaveSocket.emit (node:events:520:28)
    at ZWaveSocket.emit (node:domain:475:12)
    at Socket.<anonymous> (/home/vlycop/Documents/Dev/node-zwave-js/packages/serial/src/ZWaveSerialPortBase.ts:112:9)
    at Socket.emit (node:events:520:28)
    at Socket.emit (node:domain:475:12)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  code: 100,
  context: undefined,
  transactionSource: undefined
}
13:42:55.746 DRIVER   destroying driver instance...
13:42:55.748 DRIVER   driver instance destroyed
ZWaveError: Serial port errored: The socket closed unexpectedly! (ZW0100)
    at ZWaveSocket.<anonymous> (/home/vlycop/Documents/Dev/node-zwave-js/packages/zwave-js/src/lib/driver/Driver.ts:792:19)
    at ZWaveSocket.emit (node:events:520:28)
    at ZWaveSocket.emit (node:domain:475:12)
    at Socket.<anonymous> (/home/vlycop/Documents/Dev/node-zwave-js/packages/serial/src/ZWaveSocket.ts:21:13)
    at Socket.emit (node:events:520:28)
    at Socket.emit (node:domain:475:12)
    at TCP.<anonymous> (node:net:687:12)
    at TCP.callbackTrampoline (node:internal/async_hooks:130:17) {
  code: 100,
  context: undefined,
  transactionSource: undefined
}
Waiting for the debugger to disconnect...
Waiting for the debugger to disconnect...
Waiting for the debugger to disconnect...
```

vlycop, 2022-04-11:
i have tried a lot of thing but this is overall to much for me.
I don't understand what happenes that cause everything to stop rather than continue after the socket.open is called.
I hope what i found could be of any use as i really need auto restart to work in order to move from openzwave to this project.

closed 2022-04-21 by AlCalzone

vlycop, 2022-04-21:
Thank-you very much !

PR #4492 (2022-04-21): fix: retry TCP socket reconnection on startup
Retrying the initial connection on TCP sockets is now handled the same way as for the serial port

fixes: #4174