zwave-js/zwave-js #4175: ShenZhen Sunricher Technology, Ltd. SR-ZV9101FX-RU

Inoy-D, 2022-02-01:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

ShenZhen Sunricher Technology, Ltd. SR-ZV9101FX-RU

### Manufacturer ID

0x0330

### Product Type

 0x021A

### Product ID

0xD002

### Firmware Version

1.5

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2759

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=Manuals/2759/Z-Wave%20LED%20Dimmer%20SR-ZV9101FX-RU%20User%20Manual.pdf

AlCalzone, 2022-02-02:
@zwave-js-bot import config 2759 from zwa

AlCalzone, 2022-02-02:
The bot has created a PR but the file only serves as a starter template. It needs to be cleaned up to conform with our style guide, see above.

For anyone who wants to help out:
- fork the repo if you haven't already
- edit branch `import-config-1643837042` in your fork
- create a pull request with your branch `import-config-1643837042` as the source and our branch `import-config-1643837042` as the target

Inoy-D, 2022-02-03:
Hi, i don’t see  branch import-config-1643837042 in my fork:  https://github.com/Inoy-D/node-zwave-js/branches/all?query=import-config-1643837042

> The bot has created a PR but the file only serves as a starter template. It needs to be cleaned up to conform with our style guide, see above.
> For anyone who wants to help out:
> *  fork the repo if [...]
 
 
С уважением, Дмитрий Сулимов.

Inoy-D, 2022-02-03:
I found a similar device in the configuration database for another country

closed 2022-02-03 by Inoy-D

AlCalzone, 2022-02-03:
> Hi, i don’t see  branch import-config-1643837042 in my fork

You might have to sync the fork first.