zwave-js/zwave-js #4183: VSCode Extension for authoring config files

AlCalzone, 2022-02-03:
**Is your feature request related to a problem? Please describe.**
Authoring config files is quite complex and although we have a JSON Schema to help, it becomes unwieldy as soon as imports across multiple files are involved. Its not easy to see how the resulting config looks like and whether it has any errors.

**Describe the solution you'd like**
A VSCode extension that displays a WYSIWYG style preview of the resulting config file on the right, just like the markdown extension. Bonus points for highlighting linter errors, proper auto-completion even if imports are involved, etc...

**Describe alternatives you've considered**
-

**Additional context**
-

AlCalzone, 2022-02-03:
I've never worked on VSCode extensions, so if anyone with some experience wants to help out, please get in touch!

darkbasic, 2022-02-03:
I (briefly) [did work](https://youtu.be/qBh6-Pd5Otw) on VSCode extensions, but I can't assure I will be able to tackle this because my experience is limited and I'm quite short on free time right now. Anyway if you decide to give it a try feel free to ping me.

AlCalzone, 2023-01-21:
Update: I'm actively working on this now. An early version has been released to the official and the Open VSX marketplace, so there's Gitpod support too.

AlCalzone, 2025-05-09:
This has existed for a while now

closed 2025-05-09 by AlCalzone