zwave-js/zwave-js #4187: Minimum value of Parameter range for HomeSeer Floodlight Sensor HS-FLS100+ is rejected

COCottonwood, 2022-02-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Previous PR#3701 was CLOSED but NOT fixed.  [https://github.com/zwave-js/node-zwave-js/issues/3701](url)

What causes the bug?
Unable to input parameter value of zero.  It is still using MIN of 10 and MAX of 900

For HomeSeer HS-FLS100+, Lux_Trigger_Threshold, allowed values are 0, 30-200, 255. Value 0 will NOT activate floodlight. Value 30 represents 10 lux, Value 200 represents 900 lux. Value 255 will activate floodlight regardless of ambient light level

What do you observe?
Zwave JS HomeSeer HS-FLS100+ parameter options:
[image]

This does NOT match the Z-Wave JS Database

[image]

What did you expect to happen?

Per the HomeSeer HS-FLS100+ Manual (located at - https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2947/HS-FLS100-user-manual2b-A501112779R.pdf
[image]

Parameter file needs to be updated as follows:

"lux_trigger_threshold": {
	"label": "Light Level Threshold",
	"description": "Allowed values are 0, 30-200, 255. Value 0 will NOT activate floodlight. Value 30 represents 10 lux, Value 200 represents 900 lux. Value 255 will activate floodlight regardless of light level",
	"valueSize": 2,
	"minValue": 0,
	"maxValue": 255,
	"defaultValue": 50,
	"unsigned": true
},

What did you see?
[image]

### Device information

Manufacturer: HomeSeer
Model name: HS-FLS100+
Node ID in your network:

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

No Log file needed

COCottonwood, 2022-02-04:
FYI - I have updated to Currently on CORE 2022.2.1; SUPERVISOR 2022.01.1; HOME ASSISTANT OS 7.2 today, and I re-interviewed the node, and still have this failure.  

Do I need to remove this device and re-add it, because I would have to climb up on a ladder to re-add it as it is mounted on the corner of my house (and I have 5 of these)?

COCottonwood, 2022-02-05:
PLEASE DO NOT TAKE ANY ACTION ON THIS YET... I might have opened this PT too early. 

I was jsut informed that I need to wait for the Z-Wave JS addon to pickup the right zwavejs version, then update it, then reinterview.  See https://github.com/zwave-js/node-zwave-js/issues/3701

AlCalzone, 2022-02-06:
Let me know how it goes

COCottonwood, 2022-02-21:
Issue resolved, I did open this by mistake.

closed 2022-02-21 by COCottonwood