zwave-js/zwave-js #4188: GE/Jasco 1274/ZW3003 Dimmer Switch Missing Some Manual Actuations

CeasarPrime, 2022-02-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Manual actuation of the dimmer switch sometimes does not get reflected in the HA UI..  This appears to be very similar to another issue named:  Jasco 45609 missing some manual switch presses #4008.  However it appears that the "solution" was to replace the devices with new ones - but am not sure because the issue was closed.  I have three of the devices and they all seem to have the same issue.  A week ago they all seemed to work OK using my VERA Edge hub.  

What do you observe?
HA UI does not always follow the actual operation of the switch when manually changed.  HA UI follows state of switch fine when changed from the UI or by an automation.  

What did you expect to happen?
I expected the switches to operate the same as it did when used with VERA recently.  With VERA, the switches worked just fine - as expected.  Even manual switch operations were immediately reflected in the UI. 

### Device information

Manufacturer:   GE/Jasco 
Model name: 12724/ZW3003
Node ID in your network: 29, 39, 40

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: HA:  2022.2.0
`node-zwave-js` branch: 8.11.2
`zwavejs2mqtt` branch:  6.5.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Devices worked previously with VERA hub

### Attach Driver Logfile

[config_entry-zwave_js-d7e4374b64cdf223fd69cc2e06859182.json (1).txt](https://github.com/zwave-js/node-zwave-js/files/8005895/config_entry-zwave_js-d7e4374b64cdf223fd69cc2e06859182.json.1.txt)

AlCalzone, 2022-02-04:
> A week ago they all seemed to work OK using my VERA Edge hub.

This means the only relevant thing that changed is the controller hardware - and maybe its placement?
Please read and apply the things mentioned under general troubleshooting here:
https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

---

On a side note:

> config_entry-zwave_js-d7e4374b64cdf223fd69cc2e06859182.json (1).txt

This isn't a **driver logfile**. Maybe the explanation above the textbox where you paste it isn't obvious enough yet?
[image]

CeasarPrime, 2022-02-04:
Thanks for your very quick response AlCalzone.  The z wave stick I am using is a Aeotec Gen5+ that is at the end of a 6 foot USB extension cord to keep it well away from the Raspberry Pi 4 I am using to run Home Assistant.  The Raspberry Pi is within a foot, or so, of the Vera Edge hub.  I have not had any problem getting the z wave stick to communicate with devices all over the house - up to 50 ft away and through multiple walls.  All devices communicate directly with the stick - no hops through repeaters.  One of my dimmer switches that I am having difficulty with is within 6 ft of the z wave stick so signal strength should not be an issue.  

Sorry about the file type.   I will try again.  Thanks for your patience with me, I am just part way up the Home Assistant learning curve.  

[zwavejs2mqtt-store (2).zip](https://github.com/zwave-js/node-zwave-js/files/8006821/zwavejs2mqtt-store.2.zip)

CeasarPrime, 2022-02-05:
Just as a follow up to this topic, I have a newer in wall switch that seems to work perfectly.  It is a GE/Jasco 14291/ZW4005.  This is not a dimmer but just an ON / OFF paddle switch, located in the garage and is the farthest away of any z wave devices I have.  **However I believe it is a Z-wave Plus**.   It is shown as connected to the "lifeline" which is Node1 - the USB Z wave stick.  It would appear that the dimmer switches cannot be so connected.

AlCalzone, 2022-02-06:
Can you please re-interview (note this **does not** mean re-include!) the problematic device, then actuate it manually 1-2 times and post a log of just that? I wonder if the association just failed to be set up correctly the first time.
Don't forget to mention the node ID then, so I know what to look for.

CeasarPrime, 2022-02-06:
OK, will do,  I will have a chance to try that later today.  I have tried to re-interview Node 39 once today but for some reason, after doing that, my z wave stick seemed to loose all connection to **all** z wave devices.  A Home Assistant restart did not solve it but a reload from a full back up did.  I am not sure how that happened?  Anyway I will try again later and give you the result with a log.

CeasarPrime, 2022-02-07:
I re-interviewed Node 39 (which is a GE/Jasco 1274/ZW3003 Dimmer Switch) then started a verbose log which I have attached.   This time I had no trouble after the re-interview.  I toggled the switch manually a couple of times from fully off to fully bright while the log was underway.  I hope this helps.  
Recall that I have three of these switches.  They all seem to operate the same and the node data shown all seems the same to me.  I will  understand if these older switches are not fully supported now and will never be fully supported.  There are literally thousands of Z wave devices out there.  I can imagine that supporting even the newer ones is a huge task.!!  
Only one of the three switches is "mission critical" to me from the standpoint of needing to know the state of the device (even if changed manually) for the purpose of an important automation.  I would plan to change that switch to a new version if these older ones are not supported.  

Kind regards,

  
[zwavejs2mqtt-store (3).zip](https://github.com/zwave-js/node-zwave-js/files/8017031/zwavejs2mqtt-store.3.zip)
.

AlCalzone, 2022-02-08:
Ok, Node 39 is one of those devices that needs polling to work around the old instant reporting patent. In order to avoid unnecessary queries, the driver only polls devices under very specific circumstances. Prior to the re-interview we might not have had all the information to determine that, likely due to a timeout in the previous interview. It does look like this is working now:
```
2022-02-07T16:34:35.698Z DRIVER « [Node 039] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x270a04110126277370867277
2022-02-07T16:34:35.702Z CNTRLR « [Node 039] Received updated node info
2022-02-07T16:34:35.704Z CNTRLR   [Node 039] Node does not send unsolicited updates; refreshing actuator and sen
                                  sor values...
2022-02-07T16:34:35.709Z CNTRLR » [Node 039] requesting current switch state...
... state is queried and returned...
```

This means if you're still losing some switch actuations now, its most likely a connectivity issue.

CeasarPrime, 2022-02-09:
Thanks for looking into this.  It does look like Home Assistant is still not following manual actuations of Node 39, but I am satisfied that we have done what we can do on this topic.  
Again thanks for your help.

AlCalzone, 2022-02-09:
If we can trust the above log, the information at least reaches the driver, so the question is why it doesn't show in HomeAssistant... Maybe https://github.com/home-assistant/core can help you?

CeasarPrime, 2022-02-10:
Agreed.  I will at least post this as an issue for the lads developing Home Assistant Core.

AlCalzone, 2022-02-10:
There recently was another issue opened about the same device: https://github.com/zwave-js/node-zwave-js/issues/4220

The verdict there was that the manual activations DO reach us, but the new status has to be polled and sometimes this happens too quickly while transitioning, so the status is outdated (e.g. almost off instead of off). Could that be your problem aswell?

CeasarPrime, 2022-02-10:
> within the “store” directory using the Z Wave JS UI in Home Assistant
>
> To be sure - you are using zwavejs2mqtt, right? Then you can create a file in that folder, name it, paste the file contents on the right side and save it.
>  <https://user-images.githubusercontent.com/17641229/153493103-6d17e8eb-f4e1-4f2d-8788-ac8c87db7904.png>
>
> Don't forget to restart z2m afterwards.

OK I have created a file “test.json” and pasted the contents of the file, as requested.  I then saved and did a soft reset (restart).  None of the z wave devices now respond.  I get a message like “Failed to call service light.turn_on.  Zwave error 102.  The driver is not ready or has been destroyed ZW0103”.  See attachment.

It looks like I am going to recover using a backup????

 

 

 

 

Gerald Kiffiak

 

(403) 347-4464  ●   (403) 348-3471 – cell  ●  (403) 755-9671 – fax  ●    ***@***.***> ***@***.***

CeasarPrime, 2022-02-10:
When trying the soft reset again I get the attached error.

 

 

Gerald Kiffiak

 

(403) 347-4464  ●   (403) 348-3471 – cell  ●  (403) 755-9671 – fax  ●    ***@***.***> ***@***.***

 

 <https://user-images.githubusercontent.com/17641229/153493103-6d17e8eb-f4e1-4f2d-8788-ac8c87db7904.png>

AlCalzone, 2022-02-11:
You can't attach things to e-mails. You need to upload them via the Github UI.

> I then saved and did a soft reset (restart)

I meant an actual restart as in stop zwavejs2mqtt and start it again. If your stick isn't responsive, you might have to unplug it for a few seconds aswell.

CeasarPrime, 2022-02-12:
I am prepared to try this again but I will need some detailed instructions.  Please elaborate on:

_an actual restart as in stop zwavejs2mqtt and start it again_

How exactly is it best to do that?

AlCalzone, 2022-02-15:
I don't know? How are you running zwavejs2mqtt? Is there not a way to stop that?

Anyways, it looks like this is a recurring issue on this particular device. Can you follow the instructions there please? https://github.com/zwave-js/node-zwave-js/issues/4220#issuecomment-1040353844

closed 2022-03-05 by zwave-js-assistant[bot]

AlCalzone, 2022-07-07:
We're working with Jasco to get these things cleaned up soon.