zwave-js/zwave-js #4189: 700 Series network issues after firmware update (un-responsive network?)

RJ-Make, 2022-02-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
It's extremely random, no concise cause.

What do you observe?
System will work for many hours and then very long > 10-20 min of devices not responding. 

What did you expect to happen?
The z-wave devices to function properly. 

Steps to reproduce the behavior:
See above

### Device information

Manufacturer: Zooz
Model name: ZST10 700 W/ Latest Firmware

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.35.0
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-02-04.zip](https://github.com/zwave-js/node-zwave-js/files/8006573/zwavejs_2022-02-04.zip)

AlCalzone, 2022-02-04:
I can't open that zip file :(

RJ-Make, 2022-02-04:
> I can't open that zip file :(

Didn't get my email with the password?

AlCalzone, 2022-02-04:
It didn't even ask me for a password, but I can try again with a different program tomorrow.

RJ-Make, 2022-02-04:
> It didn't even ask me for a password, but I can try again with a different program tomorrow.

Ok, I'll re-upload tomorrow. I'm sure I screwed something up.

RJ-Make, 2022-02-05:
I've re-upped the zip. Same password.

AlCalzone, 2022-03-04:
Sorry, this slipped through the cracks - can you send me the password again, I don't have an email with one.

RJ-Make, 2022-03-04:
No problem, I see that you are very busy..  Will send. 

The network has been fairly stable, with only 2 cases of long response times or dead nodes. ~2 weeks ago and today. Today it was ~6 nodes in the red, unfortunately I didn't have logging turned on. (I know, I know :-( )  That said a restart of zwavejs2mqtt 'fixed' all the dead nodes.

RJ-Make, 2022-03-06:
This morning 2 nodes down (89 and 112) but network still responsive. Ping brought both back. Password is the same sent. 

Please let me know if a password is even needed for this data.

Thanks 

[zwavelogs_node_89_112_dead_network_ok_03-06-2022.zip](https://github.com/zwave-js/node-zwave-js/files/8192521/zwavelogs_node_89_112_dead_network_ok_03-06-2022.zip)

AlCalzone, 2022-03-11:
1. Node 89 sporadically takes >10s to be reached (or rather fails to be reached).
2. When trying to send something to node 112, the controller failed to send for a short time.

So there are two different symptoms here, but I think both are caused by the same problem: Node 90 is spamming meter reports like crazy. I see meter reports from that one roughly every 500-1000 ms, even if nothing changed. It also seems like it is sending the same reports from both endpoint 0 and endpoint 1, doubling the already unnecessary high amount of traffic.

This can definitely be considered by the controller as being jammed (at least on pre-7.17.1, although I'm not sure if today's 7.17.2 will improve this), but it most certainly has a negative influence on your network's responsiveness, since other nodes won't get enough airtime to transmit.

Turn the reporting interval on Node 90 wayyyyy down, ideally configure it to report on significant changes only (like +/- 5W), maybe add a backup reporting interval of 1h if that is possible. Also check the associations to node 1 and see if there is unnecessary duplication.

RJ-Make, 2022-03-11:
Thank you for taking a look at this. 

I think there must be something wrong with that node (90), it was already set to 5 watts, and it keeps reverting back when I change it. 

I'm going to mess with it tomorrow. 

Thanks again, and I'll report back.

RJ-Make, 2022-03-13:
So I've updated the firmware on Node 90 (all my ZEN20's) and found a bug. 

Zwave-js reverts parameter #2 back to default.  I confirmed this as a zwave-js bug by including it back into my HomeSeer setup (Z-NET) and It not only doesn't incorrectly identifies the 2 USB ports as switches, I can also update #2 without it reverting to default. 

So I'll open a new bug report for the ZEN20. 

I'm going to keep this open, as I've just updated my ZST10 700 Stick to 7.17.2 and will leave debug on for a few weeks.  Fingers crossed.

RJ-Make, 2022-03-21:
I'm closing this, as with FW 7.17.2 my network has been responsive without major issue.

closed 2022-03-21 by RJ-Make