zwave-js/zwave-js #4193: Potential Series 700 residual issue - 31 out of 84 devices healed. Healing stopped after ~10 min even if it run for 19hrs

alexruffell, 2022-02-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Running network heal

What do you observe?

After 18-19 hours the network heal has not completed and it appears it has only healed 31 nodes out of 84. I do have zwave battery operated devices but they are few compared to the mains powered ones. Either way it quit healing within 10 minutes of it starting and the remaining 19hrs it doesn't seem to have done anything.

`	Line 235: 2022-02-04T05:47:48.784Z CNTRLR   [Node 005] healed successfully
	Line 338: 2022-02-04T05:48:05.596Z CNTRLR   [Node 006] healed successfully
	Line 434: 2022-02-04T05:48:18.255Z CNTRLR   [Node 009] healed successfully
	Line 534: 2022-02-04T05:48:31.985Z CNTRLR   [Node 010] healed successfully
	Line 632: 2022-02-04T05:48:46.841Z CNTRLR   [Node 011] healed successfully
	Line 727: 2022-02-04T05:48:58.071Z CNTRLR   [Node 012] healed successfully
	Line 825: 2022-02-04T05:49:09.347Z CNTRLR   [Node 013] healed successfully
	Line 921: 2022-02-04T05:49:20.550Z CNTRLR   [Node 014] healed successfully
	Line 1019: 2022-02-04T05:49:35.050Z CNTRLR   [Node 016] healed successfully
	Line 1113: 2022-02-04T05:49:48.206Z CNTRLR   [Node 017] healed successfully
	Line 1223: 2022-02-04T05:50:01.648Z CNTRLR   [Node 019] healed successfully
	Line 1318: 2022-02-04T05:50:16.896Z CNTRLR   [Node 024] healed successfully
	Line 1413: 2022-02-04T05:50:31.730Z CNTRLR   [Node 025] healed successfully
	Line 1557: 2022-02-04T05:50:47.506Z CNTRLR   [Node 027] healed successfully
	Line 1651: 2022-02-04T05:51:04.503Z CNTRLR   [Node 031] healed successfully
	Line 1815: 2022-02-04T05:51:16.274Z CNTRLR   [Node 035] healed successfully
	Line 1919: 2022-02-04T05:51:28.789Z CNTRLR   [Node 036] healed successfully
	Line 2019: 2022-02-04T05:51:40.157Z CNTRLR   [Node 037] healed successfully
	Line 2143: 2022-02-04T05:51:55.220Z CNTRLR   [Node 040] healed successfully
	Line 2379: 2022-02-04T05:52:07.773Z CNTRLR   [Node 041] healed successfully
	Line 2682: 2022-02-04T05:52:20.585Z CNTRLR   [Node 042] healed successfully
	Line 2864: 2022-02-04T05:52:33.773Z CNTRLR   [Node 045] healed successfully
	Line 3020: 2022-02-04T05:52:47.177Z CNTRLR   [Node 046] healed successfully
	Line 3133: 2022-02-04T05:52:59.711Z CNTRLR   [Node 048] healed successfully
	Line 3229: 2022-02-04T05:53:11.343Z CNTRLR   [Node 049] healed successfully
	Line 3385: 2022-02-04T05:53:23.492Z CNTRLR   [Node 052] healed successfully
	Line 3515: 2022-02-04T05:53:36.034Z CNTRLR   [Node 053] healed successfully
	Line 3626: 2022-02-04T05:53:48.498Z CNTRLR   [Node 054] healed successfully
	Line 3786: 2022-02-04T05:54:03.435Z CNTRLR   [Node 055] healed successfully
	Line 4103: 2022-02-04T05:55:27.257Z CNTRLR   [Node 074] healed successfully
	Line 4282: 2022-02-04T05:55:42.825Z CNTRLR   [Node 076] healed successfully`

BTW, PLEASE add Node Id in some table where I can figure out what device I am looking for without having to use the diagnostic file.

I have been following:

META-Issue: Problems with 700 series (healing, delays, neighbors, ...) 🚧 #3906
https://github.com/zwave-js/node-zwave-js/issues/3906

And I have updated my Aeotec ZStick 7 stick to the firmware discussed in the #3906 prior to this heal. Other than this, the network seems to work fine and fast.

What did you expect to happen?

Finish heal. (Feature request: would be nice to get a report of pass/fail! without having to run a debug log)

Steps to reproduce the behavior:
  1. Run a heal

[config_entry-zwave_js-05dbf40f9ae35bc85f09b184fab65497.json.txt](https://github.com/zwave-js/node-zwave-js/files/8007015/config_entry-zwave_js-05dbf40f9ae35bc85f09b184fab65497.json.txt)

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version: 8.10.2
Server Version: 1.14.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I believe I might have seen a network heal finish once or twice. I have been using HA for 6 months or so.

### Attach Driver Logfile

[unfinished network heal log.txt](https://github.com/zwave-js/node-zwave-js/files/8007013/unfinished.network.heal.log.txt)

AlCalzone, 2022-02-06:
> BTW, PLEASE add Node Id in some table where I can figure out what device I am looking for without having to use the diagnostic file
> (Feature request: would be nice to get a report of pass/fail! without having to run a debug log)

That's an issue for https://github.com/home-assistant/core - this is just the driver and zwavejs2mqtt does show node IDs and the heal status.

That said, I can't see the end of the healing in your log. There seems to be a multi-hour pause while Node 083 is being healed:
```
2022-02-04T05:55:46.299Z DRIVER « [Node 100] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -93 dBm
                                  └─[BasicCCSet]
                                      target value: 255
Subscribed to Z-Wave JS Log Messages…
2022-02-04T14:39:50.080Z SERIAL « 0x011700a800014c0e320221640000faf804b00000faf800a567                (25 bytes)
```
Maybe using zwavejs2mqtt for diagnosing the heal would be better. At least there you can write a logfile which doesn't stop when the browser feels like.

alexruffell, 2022-02-10:
@AlCalzone I upgraded to ZWaveJS2MQTT, a process that took over a day as it seemed to have gotten stuck for over a day interviewing. Strangely it was mostly stuck on mains powered dimmers. I force restarted the add-on and they all showed up as healed instantly. It then took some time to complete the battery operated devices which I tried to wake up where possible. For the last few stragglers that refused to work, I excluded and included them back.

The mesh network now looks like this:

[image]

The 3 devices that do not have a direct route to the controller are all new Zooz 700 series battery operated sensors. Two are in the garage and one in the kitchen. I suspect that an individual heal would allow them all to reach the controller directly, but heals do not appear to be working on those sensors (No confirmation heal is even run).

Moving from ZwaveJS to ZWaveJS2MQTT was easy, but not 100% fool proof. For some reason ZWaveJS kept restarting even though I had turned that off, and by doing so it would hog the device and port. I ended up pointing the old integration to another USB port (this solved the device issue) and then stopped the integration for the nth time immediately before starting the MQTT version. After the about 2 days of waiting and work to allow it to finish the interviewing process, I was up and running happily.

@AlCalzone I really appreciate your work, and that of the extended team, developing ZWaveJS* Thanks!