zwave-js/zwave-js #4197: McoHome MH3901-Z Thermostat

jeroenvdwaal, 2022-02-06:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

McoHome MH3901-Z

### Manufacturer ID

0x015f

### Product Type

0x3910

### Product ID

0x5102

### Firmware Version

2.1

### Do you have a manual?

[MH3901-Z.pdf](https://github.com/zwave-js/node-zwave-js/files/8010275/MH3901-Z.pdf)

AlCalzone, 2022-02-06:
oh, you already have a PR :)

jeroenvdwaal, 2022-02-06:
Yes, but I was a little bit struggling getting the PR and Issue linked with
each other :-)

PR #4198 (2022-02-09): feat(config): configuration file MCO Home 3901-Z thermostat
fixes #4197
closed 2022-02-09 by AlCalzone