zwave-js/zwave-js #4201: Zooz ZST10 700 after firmware update to 7.17.1 still with nodes dropping off network

Snuffy2, 2022-02-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Spontaneously at times but also when multiple Zwave commands are sent (ex. a Home Assistant Automation to turn off all switches at midnight.

What do you observe?
The nodes fail and show as dead and the commands sent to them fail or occur many seconds to minutes later.

### Device information

Manufacturer: Zooz
Model name:  ZST10 700
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.11.2
`zwavejs2mqtt` branch: 6.5.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Upgraded firmware to 7.17.1 per #3906

This was worse with the 7.15 firmware but even with 7.17.1, it seems that 1+ node is always dead (incorrectly) and multiple nodes will drop off at the same time when multiple zwave commands are sent. 

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Attaching logs but truncating/combining from 2/5/22 11:00pm - 2/6/22 1:00am (Eastern) [2022-02-06T04:00:00 - 2022-02-06T06:00:00].

Lots of errors:
- [Security2CCMessageEncapsulation] [INVALID]
- error: Duplicate command
- Dropping message with invalid payload
- Message authentication failed, cannot decode command. Requesting a nonce...
- [Security2CCNonceGet] [INVALID]
- No SPAN is established yet, cannot decode command. Requesting a nonce...
- the node is unknown or not initialized yet, discarding...
- The node did not respond after 3 attempts, it is presumed dead
- Timed out while waiting for a response from the node (ZW0201)

I don't really know where to go from here.

[zwavejs_2022-02-05.log](https://github.com/zwave-js/node-zwave-js/files/8011934/zwavejs_2022-02-05.log)
I'll email my network key for decoding

markfrancisonly, 2022-02-07:
Does rolling back to zwavejs2mqtt 6.0.3 (zwave-js 8.8.3) resolve your dead nodes issue?

Snuffy2, 2022-02-11:
So my USB Sticks are attached to a 10ft USB Extension cord and moved far from any other electronics.

That being said, I've had a multitude of other problems since I opened this issue. I attempted to move my NVM settings from a Zooz ZST10 700 to a Zooz ZST10 (not 700). However, it looks like the ZST10 I got was DOA or defective and wouldn't connect to any nodes. It wouldn't connect after restoring the 700 series NVM nor after factory resetting and attempting to pair from a blank stick. Unfortunately, after I moved back to the ZST10 700 somehow this also wouldn't connect to any nodes.

All of this started because my Nortek combo stick failed. Now, I have a Nortek Combo stick that is failed, a Zooz ZST10 700 that is flaky, and a ZST10 that is DOA. Not having good luck.

I have another ZST10 (not 700) coming in the mail today and will hopefully be able to use this until the 700 series issues are better sorted out. I will be starting from scratch with this stick coming in today and re-pairing my ~90 devices.

Thus, I'm not going to be able to help for now in debugging the 700 series issues. Sorry and thanks!

closed 2022-02-11 by Snuffy2