zwave-js/zwave-js #4206: Dead Nodes after 7.17.1 firmware (ZST10).

kars85, 2022-02-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown at this moment. Time and network traffic seem to be key components. For example, if I trigger a lot of traffic from my newly added Nanomote Quad, I can get nodes to appear dead within a few hours.
What do you observe?
The affected node is Node 4 (today). Yesterday, before I had setup silly logging across the network, it was Node 16
What did you expect to happen?
For the network to remain resilient despite increased traffic.
Steps to reproduce the behavior:
Unknown at this time - but causing an influx of high traffic seems to be a trigger.

### Device information

Manufacturer: Leviton
Model name: ZW15R
Node ID in your network: 4

Zooz ZST10 is on 7.17.1.344

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.4.1
zwave-js: 8.11.2
home id: 3745883352
home hex: 0xdf45a4d8

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-02-08.log](https://github.com/zwave-js/node-zwave-js/files/8025229/zwavejs_2022-02-08.log)
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8025230/zwavejs_current.log)

kars85, 2022-02-08:
Doing some additional configuration with my Minimote Quad, and are seeing more mains nodes dropping off now, too. Node 23 is the latest dead node.
[zwavejs_2022-02-08_morenodes.log](https://github.com/zwave-js/node-zwave-js/files/8025570/zwavejs_2022-02-08_morenodes.log)

AlCalzone, 2022-02-08:
Meeeehhh. Do you have a Zniffer? This looks like the 700 series issue all over.

kars85, 2022-02-08:
I am unversed on the Zniffer realm, but do have a couple Aeotec Z-Stick 5's laying around. Would love to assist if there's a guide to follow showing the necessary processes you're looking for. If not, I'll Google and give it best effort here in the coming day(s). 

Agreed, something is still not right on the 700 series. I chalked my first dead node up to bad mojo, but now I'm throwing what I originally intended at controller and it's not going well.

AlCalzone, 2022-02-08:
I've heard that the UZB3 is heaps better as a Zniffer, but it might be worth a try. After all, you just want to capture what the other stick sends.

@jmgiaever since you went through the Zniffer installation a few times lately, what's your recommendation?

@kars85 you'll can download Simplicity Studio in the meantime

jmgiaever, 2022-02-08:
Hi,

I've only tried the UZB3 as a zniffer, but it works great. Installation is simple too.

I've experienced that the PC Programmer sometimes fails flashing the dongle, but then it's already recognized as a zniffer module in the Zniffer software. However an unuseful device, but the Zniffer software will recommend upgrading it to the latest - and it will do it for you. Then it starts sniffing right after the upgrade. 

Not sure if it's the case with the Z-stick though.

AlCalzone, 2022-02-08:
@kars85 if you contact me on Slack, I can send you the Z-Wave Programmer and the Zniffer firmware. This stuff is hard to find on the internet - it's out somewhere...

jmgiaever, 2022-02-08:
It's available at [this page](https://www.silabs.com/development-tools/wireless/z-wave/embedded-development-kit) in the grey section («footer»?):

[image]

kars85, 2022-02-08:
@jmgiaever - went to that page, logged in, but won't let me download.

[image]

@AlCalzone - have Simplicity Studio ready and pinged you offline for further instruction.

kpine, 2022-02-08:
If you follow the instructions in that yellow text you will have access to the Z-Wave tools. Download (or at least start the download) of one of the [Z-Wave SDK](https://siliconlabs.force.com/s/validatezwsdk?ContentId=0681M000007yaWr%20&ContentType=ESDK&rURL=https%3A//www.silabs.com/products/development-tools/software/z-wave/embedded-sdk.html) packages and it will prompt you to accept a EULA. Once accepted, you can download the other tools.

kars85, 2022-02-09:
@kpine - Noticed that - well executed plan of attack on that front. Unfortunately, I'm finding more guidance online about Zniffing with a UZB3 and not my Z-Stick Gen5, so at a bit of a loss trying to combine bits/pieces on flashing my Gen5 with the Zniffer FW.

kpine, 2022-02-09:
If I recall correctly, I followed [these instructions](https://community.silabs.com/s/article/z-wave-500-programming-uzb3-controller-stick?language=en_US) for my Gen5, despite the warning about other manufacturers. It worked but... in my experience the Gen5 was not a good Zniffer. Comparing a UZB3 and a converted Gen5 from the same location, plugged into the same PC with extension cord, the Gen5 had much worse results. The UZB3 was capturing most of the traffic without a problem, but the Gen5 was missing messages and seeing a lot of CRC errors. It doesn't really make much sense to me why that was the case, maybe you'll have different results.

kars85, 2022-02-09:
Thanks - I think I'll invest in a UZB3 for the cause here. The least I can do for you and the team's efforts in straightening this out. In the meantime, I'll try some of the links you posted and see what results I get tomorrow. I have two Z-Stick 5's, and not worried about donating one to the cause if one is locked in afterwards, should Zniffing work.

One of my 5's is on the latest Aeotec firmware, the other is likely on a predated S2 capable FW for now.

Stay tuned...

jmgiaever, 2022-02-09:
Hi @kars85 - You can reach out to me if you're in need of an UZB3 zniffer. I can offer you one pre-flashed at a good price :)

kars85, 2022-02-09:
@jmgiaever - thanks for the offer. Are you on Slack? I was able to get my Z-Stick 5 flashed (after a LOT of work) and able to be read in Zniffer. Quality of the packet grabs is TBD. DM me in Slack to talk offline.

jdiegmueller, 2022-02-10:
> Meeeehhh. Do you have a Zniffer? This looks like the 700 series issue all over.

I have a small 10-node dev network setup using a Silicon Labs UZB7.  After upgrading it to 7.17.1 I am seeing similar behavior.  I can trigger nodes to go offline when I heal.  I can bring them right back online with a refresh.  

I did a bunch of interview/refresh/heals and grabbed zwavejs logs and Zniffer captures.  If there's anything specific you need done or collected, let me know and I'll do a second pass.

[doogles_zwavejs-4206.zip](https://github.com/zwave-js/node-zwave-js/files/8037733/doogles_zwavejs-4206.zip)

AlCalzone, 2022-02-10:
Thanks @jdiegmueller - I've already suspected that the commands used for healing were not fixed. I'll check out your capture later and forward relevant info to Silabs if it confirms my suspicion.

AlCalzone, 2022-02-10:
@jdiegmueller ok your issue is actually this one: https://github.com/zwave-js/node-zwave-js/issues/4190

kars85, 2022-02-10:
I am relocating my Z-Stick 5 Zniffer closer to my controller by using a spare Win10 laptop. Can anyone provide the "signed" uzb.inf driver? I had to spool up my Win7 VM to get the Zniffer working on my desktop via the uzb.inf driver I was able to find

kpine, 2022-02-10:
The drivers are included with the Zniffer download.

kars85, 2022-02-10:
As this will be my first attempt at packet inspection/capture, is it acceptable to let Zniffer collect for a day period, or do I have to try to actively generate the controller collapse for a shorter period of time (minutes). My primary concern with a longer capture period would be the .zlf file getting out of hand, so want to confirm best practice(s).

kars85, 2022-02-11:
Here's a capture for about an hour period today where I'm trying to configure a Nanomote Quad (Node 34). The network appears to have collapsed on itself. Please let me know if any of this helps
[zwavejs_2022-02-11.log](https://github.com/zwave-js/node-zwave-js/files/8049705/zwavejs_2022-02-11.log)
[2-11-20_zniffer.zlf.zip](https://github.com/zwave-js/node-zwave-js/files/8049717/2-11-20_zniffer.zlf.zip)
.

kars85, 2022-02-15:
Haven't done anything out of the ordinary with the network the past couple days as far as configuring my minimote quad. Just letting automations do their thing. Last night, towards the end of the end of the attached log, I had Node 23 (Leviton DZ6HD) go dead.
[zwavejs_2022-02-14.log](https://github.com/zwave-js/node-zwave-js/files/8072841/zwavejs_2022-02-14.log)

kars85, 2022-03-09:
Node 12 (BE469ZP lock) went dead last night (3/8) at 10:07PM. Had Zniffer running thankfully, so I've attached that and the silly log for that day.

Had a mains powered Leviton fan controller go dead about a week ago, too, but didn't have Zniffer capturing on my UZB3 yet
[zwavejs_2022-03-08.log](https://github.com/zwave-js/node-zwave-js/files/8215537/zwavejs_2022-03-08.log)
[node12_zniff.zlf.txt](https://github.com/zwave-js/node-zwave-js/files/8215540/node12_zniff.zlf.txt)
.

AlCalzone, 2022-03-11:
@kars85 I think there is a part missing between the two logfiles. the first one ends at `2022-03-08T12:00:56.530Z` with node 12 being alive, the next one starts 6 hours later at `2022-03-09T06:00:00.002Z` with node 12 already dead.

closed 2022-03-28 by zwave-js-assistant[bot]