zwave-js/zwave-js #4207: [feat] Identify dead devices

The-Deep-Sea, 2022-02-08:


AlCalzone, 2022-02-08:
The usecase behind this issue makes sense, but I'm not decided yet what is the best way to do this.

jmgiaever, 2022-02-08:
> My question wasn't aimed at waking up devices.

No, but to be able to ping a device it needs to support to be woken up. That's why I though maybe mark it as «presumed dead» when

1. the device haven't reported alive within the wake-up interval since zwjs was started
2. responded to the ping within the wake-up interval, that was sent if 1 was the case

raman325, 2022-02-08:
> **Additional context** Wouldn't it be correct to set the entities of dead nodes to "unavailable" when working with Home Assistant? Except possibly the node status.

Right now all entities but the node status sensor get marked as unavailable when a device is marked dead by the driver. We were actually just having a separate discussion about whether this is the right behavior because there are cases where the driver may think the device is dead but the next action against the device would cause it to respond and make the node alive again, but there's a catch 22 because the entity is unavailable in HA so you can't perform normal actions against the device to revive it (you can ping it, but there's no UI component to do that so it's not as easy of a recovery setup).

In your example, the false alarm above would be after the fire alarm was marked dead but you put the battery back in - something needs to happen to tell the driver that the node is alive, otherwise it's still dead even though it's functioning again.

A polling mechanism reduces the false positives in either direction (we think it's dead but it's alive or we think it's alive but it's dead) but comes at a cost of battery life and extra processing cycles. I don't know what the optimal solution is here either but these two discussions are somewhat related and I will be keeping an eye on this accordingly.

The-Deep-Sea, 2022-02-09:
> No, but to be able to ping a device it needs to support to be woken up.

Okay, that's correct. However, in the case of Zigbee2mqtt, sleeping devices are not pinged either.
Only devices that are, or should be, alive are pinged (usually devices that are connected to the mains).

---

> Right now all entities but the node status sensor get marked as unavailable when a device is marked dead by the driver.

I tested it: you are right. I don't know why I remembered it differently.

> something needs to happen to tell the driver that the node is alive, otherwise it's still dead even though it's functioning again.

At least the devices I tested report back automatically as soon as they have power.
Otherwise, passive (battery powered) devices should do it at the latest with the next wakeup interval and active (mains powered) devices are actively pinged if you implement it as with Zigbee2mqtt.

> A polling mechanism reduces the false positives in either direction [...] but comes at a cost of battery life and extra processing cycles.

If it only pings mains powered devices, it will not affect a battery.
But I can't say what about FLiRS devices, I don't know (yet) about them. But if it is configurable, every user can decide for himself whether he prefers a longer battery life or more "security", possibly also per device.
Basically, this (a dead device) should be the exception anyway.

AlCalzone, 2022-02-09:
> At least the devices I tested report back automatically as soon as they have power.

I have at least one that doesn't. Only when you physically interact with it, or try to control it via Z-Wave.

To reiterate, we need to solve these points:
- detect when a battery powered (sleeping) device hasn't woken up for too long
- detect when a mains powered device no longer reacts to pings
- detect when a FLiRS device no longer reacts, but at a reduced frequency compared to mains powered or you'll drain the battery

And at the same time we need to
- avoid false-positives for fragile meshes
- avoid introducing unnecessary delays by pinging unreachable devices:
	a) because the timeout can take several seconds
	b) because the most used stick (sometimes?) has a firmware bug where attempting communication with a dead device can delay the serial API response that should come immediately and throw the driver and stick out of sync

bagobones, 2023-10-03:
I have two questions:

1. the Z-wave JS web UI exposes a last active time stamp however this does not appear to be exposed in home assistant via the native HA API (haven't tired MQTT).. Even just having this would let me create an alert for devices that have not checked in for more than 12 hours (way longer than any of my battery devices should go without checkin)
2. As already mentioned [zigbee2mqtt availability](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html) settings have two time based settings 1 short one defaulted to 10 min for active devices (routers/mains powered), and another for passive devices (mostly battery powered) which is 1500 min AKA 25 hours. short of a hand held remotes, lightbulbs and maybe an alarm/siren (special class) I would suspect MOST device check in at least once and awhile. You might even be able to get an idea by looking at larger existing installs and looking at those last active time stamps to generate some useful info.
3. Do dead devices not just turn alive again on their next check-in? Is the fear primarily that marking something dead will prevent it from being available for use?

As there is a fear of false positives why not implement a broad Active / Passive approach but let users opt into it as a beta feature and provide feedback.. it may be a good idea to allow excluding device classes and individual devices if it becomes a problem.

gruntpartystyle, 2024-11-07:
Just to make sure I understand this... if a battery powered device fails to check in with the controller it will stay "asleep" into perpetuity? Isn't that in direct contrast with the principles of zwave guaranteeing reliability?
I trust that my battery operated devices report in every 6h (or whatever I set), so I trust my controller will throw a trantrum if the device doesn't check-in after a while... I rather have a warning of a suspected dead node which failed to check-in after x missed intervals than a sensor which is dead since weeks with no indication at all.
AFAIK, battery powered devices have a wake-up parameter/interval. The controller is aware of this parameter as it can read this parameter? So why not at least flag the device as AWOL after it has missed its expected check-in? 

[image]

1. Make it a global parameter that x missed check-ins means trouble, ... 
2. ...have device specific setting to override the global one for certain devices / deactivate the check.
3. If a device doesn't have a wake-up interval, use a system default 24h or whatever
4. ...which can also be customized per device if required.

AlCalzone, 2024-11-07:
> So why not at least flag the device as AWOL after it has missed its expected check-in?

Because this issue is still open and a solution has not been implemented yet.

VVVVVVVip, 2025-01-14:
I think a working report/notification for devices battery level would go a long way here to solve some of this at least.

None of my battery devices reports any change in battery level, and I found this quite annoying, especially since the battery sensor entity is discovered.
I moved from a Vera Egde to HA with a Aeotec Z‐Stick Gen5 USB Controller ZW090/ JS Add-on and it took me a while to realize that there is no reports from the devices.
For devices with temp sensors I now monitor if the temp hasn't changed for 24h and send a notification to check the device battery.

dannytrigo, 2025-08-06:
I'm really surprised this feature isn't available yet... 

I just checked my nodes and realised a door sensor has not been working since January, and its still marked as sleeping

jeffjensen, 2025-08-06:
While I too look forward to this feature, in the meantime I created a card that lists entities that haven't checked-in recent enough, allowing me to easily determine dead batteries and other problems.  Maybe it helps you for now too:
```
entity_id: /sensor.*_sensor_last_seen/
"last_updated: "> 840m
```