zwave-js/zwave-js #4220: [question] Device only update status after tapping (off) twice (manual)

junior466, 2022-02-10:


AlCalzone, 2022-02-10:
@zwave-js-bot add compat flag to 0x0063:0x4944:0x3031:3.37
```jsonc
manualValueRefreshDelayMs: 250
```

AlCalzone, 2022-02-10:
@junior466 If you have zwavejs2mqtt installed, you can test [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/73d6de0df6a4fda7619eb4d37b8748f899418972/packages/config/config/devices/0x0063/zw3003_ge_12724.json) by following [this guide](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files) and let me know if that works.

If not, also let me know - then I'll just merge and you can test on the next update.

junior466, 2022-02-10:
@AlCalzone Thanks for the prompt (possible) fix. I will try it as soon as I can today. And just to be sure, where will my ‘app-root’ be if installed using Snap on Ubuntu?

AlCalzone, 2022-02-10:
@jmgiaever?

jmgiaever, 2022-02-10:
`/var/snap/zwavejs2mqtt/<revision>/` where each revision contains data (backup) from each. You should most likely use the highest revision, or the symlink `current`.

junior466, 2022-02-10:
> 

So inside there should be a ‘store’ folder where I create a ‘config’ folder and upload? Apologies but away from desk.

jmgiaever, 2022-02-10:
Hi, no, that folder is your «store folder». Everything essential is created by z2m itself, but if you need to store files, use the config folder specified in the «config priority path» within the zwave section in the setting. In my case it's set to (by default): `/var/snap/zwavejs2mqtt/340/config`. 

**However** this path is not automatically created. Create it with `sudo mkdir <path you have>`. 

Note that `340` is the revision number. Snap will increase/decrease this value whenever you're upgrading or downgrading. 

For what I'm aware you're not able to upload files through the Z2M UI. If you have to upload from your local computer, use scp or something like that to upload it to your server in an SSH like way

```
scp /my/local/file username@my-host:<path you have>
```

AlCalzone, 2022-02-10:
You can edit files in the UI, which is enough in this case
https://github.com/zwave-js/node-zwave-js/issues/4188#issuecomment-1035491752

junior466, 2022-02-11:
@AlCalzone I have followed the instructions but no luck. Attached are the logs. I tried it a couple times between 7:10PM and 7:30PM eastern time. 
One thing I didn't do is re-interview that device as per the guide since I was afraid of loosing my configuration. If it must be done, let me know and I will try it. 
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8045141/zwavejs2mqtt-store.zip)

AlCalzone, 2022-02-11:
Doesn't look like the config was applied. Did you restart zwavejs2mqtt? If so, please keep logging enabled there and download the logfile from the store so I can see what happens during startup.

junior466, 2022-02-11:
I made sure to restart it at the command line after clicking save. I have logging enabled since yesterday. I just finished restarting and attached the logs. I interacted with the switch today between 4:20PM and 4:25PM eastern time. 
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8051720/zwavejs2mqtt-store.zip)

junior466, 2022-02-12:
@AlCalzone Anything else you recommend I try? Any more logs needed?

AlCalzone, 2022-02-15:
```
2022-02-11T21:23:19.367Z CNTRLR   [Node 031] Node does not send unsolicited updates; refreshing actuator and sen
                                  sor values in 250 ms...
2022-02-11T21:23:19.617Z CNTRLR » [Node 031] requesting current switch state...
```

Looks like the change was applied, but maybe 250 ms is still not enough. How long does that device need for dimming?

AlCalzone, 2022-02-15:
Can you try something ridiculously long, like 10000? I wonder if the delay has any influence on how close the reported value is to the actual brightness.

Also how is the dimming configured?
[image]

blhoward2, 2022-02-15:
This means it should take one 30ms step. I just confirmed that's what mine are set to and they barely ramp down at all. It's practically instant.

AlCalzone, 2022-02-15:
Interesting... we should definitely mention this in the device config comments when we can pin the issue down further.

junior466, 2022-02-15:
So what @blhoward2 said is I should change the step to 30ms? What setting exactly? There are multiple Dim Rate Steps.

blhoward2, 2022-02-15:
It’s already at 30 ms. The unit is 10ms. There are two that matter. Manual is the local switch. Zwave is zwave. I’d try two or three steps at 30 ms. Then if that doesn’t work try 1-3 steps at 10 ms. Too slow and the get will catch it on the ramp but just play around with it.

junior466, 2022-02-15:
I think I want to try a factory reset before going further just in case. Should I just reset the switch manually or do I need to do anything from within the control panel?

blhoward2, 2022-02-15:
Do it at the switch. Then replace failed node and include it again.

junior466, 2022-02-15:
So I configured `Dim Rate Steps (Z-Wave Command) ` to 2 and `Dim Rate Steps (Manual) ` to 2 and later tried a value of `3` on each and no luck. Instead the brightness goes to 18 when turned on and 240 when turned off.

junior466, 2022-02-15:
@AlCalzone I tried your suggestion of `3000` and it works perfectly. Since I can reproduce the results in the configuration panel as suggested by @blhoward2, is there any downside to using the configuration file? And if not, how do I apply the same solution to another switch? Not sure how the configuration file should look like.

AlCalzone, 2022-05-04:
@junior466 did you get this resolved somehow? I'm not convinced that the linked PR is the correct fix for this, so I'll probably not merge it.

junior466, 2022-05-05:
The suggestion [here](https://github.com/zwave-js/node-zwave-js/issues/4220#issuecomment-1040321979) has been working fine.

AlCalzone, 2022-05-06:
Ok thanks. I'm hesitant to use this as the "official" fix though, since you seem to be the only one so far with the problem.

junior466, 2022-05-17:
It appears that the issue is only with a WD500Z-1 switch. Manual control doesn't update Home Assistant at all. I tried resetting the switch and re-pairing it.

AlCalzone, 2022-05-18:
Could be that you need to set up a lifeline association from the switch to the controller. For some  older switches, zwave-js doesn't know which association would be needed.

closed 2023-01-26 by AlCalzone

lee-costa, 2023-09-25:
So recently I migrated from the SNAP package version to a docker compose file. Brought back all of my devices but noticed the issue was back. I followed the steps again and configured the JSON file `"manualValueRefreshDelayMs":` value to 30000 but nothing this time. 

Attached are my logs and here is my device configuration screen.
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/12720307/zwavejs_current.log)
[image]

lee-costa, 2023-09-26:
...and figured out the solution again by going through the comments. The sweet spot for me is setting `manualValueRefreshDelayMs:` to `3000`. Here's the json file with this already set. 

Hopefully this helps someone (and myself) down the road.
[ZW3003.json.zip](https://github.com/zwave-js/node-zwave-js/files/12732004/ZW3003.json.zip)

TrendyGuy, 2025-04-18:
@lee-costa @AlCalzone Sorry to post to such an old issue. I am having this same issue with my GE 12724. I have a new install of Home Assistant OS with a Zooz 8-- Z-Wave USB stick and Z-Wave JS UI installed. Everything seems to be working great except for this.

I cannot figure out out to add the below to my setup:

"manualValueRefreshDelayMs": 1000

Any assistance would be greatly appreciated to get this fixed.

lee-costa, 2025-04-18:
@TrendyGuy It has been a while since I last created any custom configurations but I believe the documentation that would help is located [here](https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files) and [here](https://zwave-js.github.io/zwave-js/#/config-files/file-format). I am sure @AlCalzone can guide you further.