zwave-js/zwave-js #4227: [Bug]: Restoring a backup of the NVM from a Z-Stick Gen5 device to a UZB7 fails conversion.

Madj42, 2022-02-11:


AlCalzone, 2022-02-11:
@Madj42 let me know if this is still relevant. #4093 has been fixed.

Madj42, 2022-02-13:
Thank you.  I now get farther but it seems like it still has issues.  I put the details in 4093.

Danny2100, 2022-02-15:
I'm having issues on an aeotec stick 7. I also have a zooz I'm going to try that now.

[image]

AlCalzone, 2022-02-15:
Make sure you have the latest driver version. I very recently fixed a bug in this regard.

Danny2100, 2022-02-15:
> Make sure you have the latest driver version. I very recently fixed a bug in this regard.

Yes I updated zwavejs2mqtt yesterday.

AlCalzone, 2022-02-16:
Ok please send me your NVM backups (500 and 700) so I can check what's wrong.

MoogleTroupe, 2022-02-18:
I have this same issue trying to restore a 500 stick  to a Aeotec 7 stick, I e-mailed you the nvm files.

AlCalzone, 2022-02-18:
Looks like zwavejs2mqtt didn't have the patch included yet. The new release today will have it, please try again with it.

Dinth, 2024-08-17:
Hi, it seems that the problem persists at least for me.
My Zstick is updated to v1.20 but still getting the same error:
```
2024-08-17 17:09:55.600 INFO APP: GET /health/zwave 301 0.417 ms - 191
2024-08-17 17:09:55.887 INFO Z-WAVE: Controller status: Convert NVM progress: 83%
2024-08-17 17:09:57.639 INFO Z-WAVE: Controller status: Convert NVM progress: 84%
2024-08-17 17:09:59.134 INFO Z-WAVE: Controller status: Convert NVM progress: 85%
2024-08-17 17:10:00.885 INFO Z-WAVE: Controller status: Convert NVM progress: 86%
2024-08-17 17:10:02.636 INFO Z-WAVE: Controller status: Convert NVM progress: 87%
2024-08-17 17:10:04.130 INFO Z-WAVE: Controller status: Convert NVM progress: 88%
2024-08-17 17:10:05.881 INFO Z-WAVE: Controller status: Convert NVM progress: 89%
2024-08-17 17:10:07.628 INFO Z-WAVE: Controller status: Convert NVM progress: 90%
2024-08-17 17:10:09.376 INFO Z-WAVE: Controller status: Convert NVM progress: 91%
2024-08-17 17:10:10.887 INFO Z-WAVE: Controller status: Convert NVM progress: 92%
2024-08-17 17:10:12.638 INFO Z-WAVE: Controller status: Convert NVM progress: 93%
2024-08-17 17:10:14.388 INFO Z-WAVE: Controller status: Convert NVM progress: 94%
2024-08-17 17:10:16.138 INFO Z-WAVE: Controller status: Convert NVM progress: 95%
2024-08-17 17:10:17.627 INFO Z-WAVE: Controller status: Convert NVM progress: 96%
2024-08-17 17:10:19.378 INFO Z-WAVE: Controller status: Convert NVM progress: 97%
2024-08-17 17:10:21.129 INFO Z-WAVE: Controller status: Convert NVM progress: 98%
2024-08-17 17:10:22.640 INFO Z-WAVE: Controller status: Convert NVM progress: 99%
2024-08-17 17:10:24.389 INFO Z-WAVE: Controller status: Convert NVM progress: 100%
2024-08-17T16:10:25.455Z CNTRLR   Restoring NVM backup...
2024-08-17T16:10:25.626Z CNTRLR   Turning RF on...
2024-08-17 17:10:25.635 INFO APP: GET /health/zwave 301 0.801 ms - 191
2024-08-17 17:10:25.883 INFO Z-WAVE: The given data does not match the NVM size - cannot restore! (ZW0322) restoreNVM undefined
```
[image]
[image]

AlCalzone, 2024-08-17:
I'm going to need the driver log on loglevel debug to investigate.

Dinth, 2024-08-17:
Sorry im not sure if thats the log you need... but no matter what logging options i choose (tried several combinations of General and Driver logging settings), i can see the exact conversion logs in Debug window if Zwave logging is set to silly and "Log to file" is disabled, but as soon as enable "Log to file", the conversion logs dont seem to be saved to file. 

[z-ui_2024-08-17 (1).log](https://github.com/user-attachments/files/16646280/z-ui_2024-08-17.1.log)
[zwavejs_2024-08-17 (1).log](https://github.com/user-attachments/files/16646281/zwavejs_2024-08-17.1.log)

Dinth, 2024-10-22:
Sorry, i have just missed the notification about the reply, i will try to check this week.

Dinth, 2024-10-24:
It still doesnt work, here's the latest log 
[zwavejs_2024-10-24.log](https://github.com/user-attachments/files/17508745/zwavejs_2024-10-24.log)

AlCalzone, 2024-10-25:
Can you make an NVM backup of both the Gen5 and the UZB7 and share that?

Dinth, 2024-10-25:
Here is the NVM backup of Gen5, i will try to make a backup of UZB7 over the weekend
[NVM_20241024143517.tar.gz](https://github.com/user-attachments/files/17526127/NVM_20241024143517.tar.gz)

Dinth, 2024-10-26:
Another NVM backup of Gen5 as after the last night update the new backup has different checksum (still doesnt work though)
[NVM_20241026174856.tar.gz](https://github.com/user-attachments/files/17531523/NVM_20241026174856.tar.gz)

When backing up Razberry, the size of the backup is 0 bytes

AlCalzone, 2024-10-29:
> When backing up Razberry, the size of the backup is 0 bytes

That would explain why it fails. I'm confused though - is it an UZB7 or a Razberry 7?

Dinth, 2024-10-30:
So firstly let me apologise - Ive got no idea why i got so confused, but i indeed have Razberry 7 Pro not UZB7 :( I think it was some kind of self-suggestion, i once wrote UZB7 and then stuck with that in my mind. I hope this didnt cause a lot of trouble for you

AlCalzone, 2024-10-30:
According to https://z-wave.me/support/uzbrazberry-firmwares/, there are quite a few firmware updates available for the Razberry 7 Pro, one of which reads
> Changes NVM size to comply with the standard SiLabs firmwares and allows backup restore.

Might be worth a try before we continue investigating.

Dinth, 2024-10-30:
Originally i have seen that website and assumed that since my Razberry has firmware 7.5, then its newer than the ones available online... but my SDK is much older, so maybe my 7.5 is really 7.05, so definetelly worth trying...
[image]

On another subject, will OTA update in ZwaveJS work for a controller? Since im using my Razberry through TubesZB PoE adapter, im not sure how to upgrade it

AlCalzone, 2024-10-30:
You have to use the OTW (over the wire) update functionality. It **should** work with Razberry controllers too, but I can't give a guarantee. Using the "special tool" linked from the firmwares site would be safer, but I'm not sure how to point that to a remote serial port.

Dinth, 2024-11-01:
Thank you, it worked! 
And sorry again for not checking that FW version before - i was confident that my version was newer, because i didnt know how Zwave interprets 0 in version numbers

AlCalzone, 2024-11-01:
There are no leading zeros in version numbers. That's just marketing.

closed 2025-05-08 by AlCalzone