zwave-js/zwave-js #4228: Vivint Element CT200X Thermostat Air Temperature Not Updating

randominternetuser, 2022-02-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

ZwaveJS attempts to query the multilevel sensor air temperature with a scale of Celsius and the device responds in Fahrenheit out of specification causing the query to fail. Re-Interviewing the device does not fix the issue however it does cause the air temperature sensor to be updated once correctly.

What do you observe?

Air temperature sensor values not updating after initial interview values.

What did you expect to happen?

Air temperature sensor values to update correctly when polled. 

Steps to reproduce the behavior:
  1. Go to device in Z-Wave JS panel
  2. Click on 'Refresh' under Multilevel Sensor v6
  4. See error in debug logs and value is not updated.

### Device information

Manufacturer: Radio Thermostat Company of America (RTC)
Model name:  Vivint Element Thermostat w/ other sensors (CT200X)
Node ID in your network: 6 & 8

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.11.2
`zwavejs2mqtt` branch: 6.5.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/8052341/zwave_js.log)

randominternetuser, 2022-02-12:
I should also note I've tried setting preferred scale for this type in Z-Wave settings and it still fails with an extra message:

``2022-02-12T03:28:41.896Z CNTRLR [Node 006] Preferred scale 1 not supported for sensor type 1, using the first
supported scale``

AlCalzone, 2022-02-18:
There is this in the logs
```
2022-02-12T00:13:54.555Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -87 dBm
                                  └─[MultilevelSensorCCReport] [INVALID]
                                      error: Unsupported sensor type Fahrenheit or corrupted data
```
which seems weird. Please change the loglevel to "debug" and re-interview the device, then post a log of the re-interview.

randominternetuser, 2022-02-21:
Here is a debug log filtered for this device with several refresh of the multi level sensor data and a full interview of the device. After the interview the temperature is missing now under the sensor data and you can see a few of the above errors in the log.
[zwavejs_2022-02-20.log](https://github.com/zwave-js/node-zwave-js/files/8106100/zwavejs_2022-02-20.log)

AlCalzone, 2022-02-21:
The problem is that the device only reports support for the Celsius scale:
```
2022-02-21T03:49:04.787Z SERIAL « 0x010d00a8000106043106010100a7c9                                    (15 bytes)
                                                          ^^-- Celsius, would have to be 03 for Celsius+Fahrenheit
                                                        ^^---- Air temperature      
2022-02-21T03:49:04.789Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[MultilevelSensorCCSupportedScaleReport]
                                      sensor type:      Air temperature
                                      supported scales:
                                      · Celsius
2022-02-21T03:49:04.790Z CNTRLR « [Node 006] received supported scales:
                                  · Celsius
```
and no support for any Humidity scales:
```
2022-02-21T03:49:04.830Z SERIAL « 0x010d00a8000106043106050000a7cc                                    (15 bytes)
                                                          ^^-- No supported scales, would have to be 01 for %
                                                        ^^---- Humidity
2022-02-21T03:49:04.832Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[MultilevelSensorCCSupportedScaleReport]
                                      sensor type:      Humidity
                                      supported scales:
```
which is likely due to an incorrect encoding of the scales field, I'd assume an off-by-one error in the bitmask, or the firmware interpreting this as an array instead of a bitmask. I've seen this happen before.

For Multilevel Sensor and Meter CC, zwave-js ignores any reports with an unsupported type or scale, because this has lead to many issues with power meters that suddenly report water consumption or something strange due to corrupted RF frames. And these no longer go away without re-interviewing the device.

Can you try contacting the manufacturer support and see if there are any firmware updates fixing this bug?

PurpleIsEverything, 2022-02-21:
Yeah I've noticed it's a bit strange in that department ZwaveJS asks for Celcius scale and it clearly responds in Fahrenheit. If you try and change the scales in HA or ZwaveJS with the scales config and or on the thermostat itself you'll loose data entirely.

Unfortunately I suspect it will be extremely unlikely the manufacturer will support this device in any way outside of use with their alarm system. I'll try and contact the OEM RTC but they don't even list this model on their website and that's probably also a dead end.

Anything that can be done to make devices like these function properly would extremely appreciated. 

These devices work fine in every other manner and I'd like to be able to use them rather than scrap them like the alarm company installers that toss more of these than they care to admit.

randominternetuser, 2022-02-21:
EDIT: Apologizes Work Account

Yeah I've noticed it's a bit strange in that department ZwaveJS asks for Celcius scale and it clearly responds in Fahrenheit. If you try and change the scales in HA or ZwaveJS with the scales config and or on the thermostat itself you'll loose data entirely.

Unfortunately I suspect it will be extremely unlikely the manufacturer will support this device in any way outside of use with their alarm system. I'll try and contact the OEM RTC but they don't even list this model on their website and that's probably also a dead end.

Anything that can be done to make devices like these function properly would extremely appreciated.

These devices work fine in every other manner and I'd like to be able to use them rather than scrap them like the alarm company installers that toss more of these than they care to admit.

PR #4568 (2022-05-04): feat: add compat flag to disable strict meter/sensor/scale validation
fixes: #4228
closed 2022-05-04 by AlCalzone