zwave-js/zwave-js #4229: Go-Control (Linear) NGD00Z-4 Missing attributes

pfunkmallone, 2022-02-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Attributes which existed at one point, appear to be missing.

Coinciding with an HA update and zwavejs2mqtt update, I noticed that the battery status value began reporting "undefined".

IIRC, it was a true/false for battery alert.

### Device information

Manufacturer: Go Control/Linear
Model name: NGD00Z-4
Node ID in your network: 33

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.11.4
`zwavejs2mqtt` branch: 6.5.1

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Nothing particular to the node in question

### Did this work before?

Yes (please describe)

### Attach Driver Logfile

[zwavejs_2022-02-12.log](https://github.com/zwave-js/node-zwave-js/files/8052702/zwavejs_2022-02-12.log)

pfunkmallone, 2022-02-21:
I assume you mean clicking "refresh" on the Notification v4 menu on the node correct? Yes, I've done that. Still "undefined".

AlCalzone, 2022-03-02:
Sorry, this slipped through the cracks. I meant the refresh on the **Battery** menu. In any case, a driver log of a re-interview wouldn't hurt to see.

closed 2022-03-13 by zwave-js-assistant[bot]