zwave-js/zwave-js #4230: Missing device configuration: Nexia One Touch NX1000

AB398, 2022-02-12:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Nexia One Touch NX1000

### Manufacturer ID

0x0178

### Product Type

0x5343

### Product ID

0x4735

### Firmware Version

1.26

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1344/

### Do you have a manual?

http://www.nexiahome.com/wp-content/uploads/2015/07/One-Touch.pdf

AlCalzone, 2022-02-13:
@zwave-js-bot import config 1344 from zwa

AlCalzone, 2022-02-13:
The bot has created a PR but the file only serves as a starter template. It needs to be cleaned up to conform with our style guide, see above.

For anyone who wants to help out:
- fork the repo if you haven't already. If you have, synchronize your fork so you get the new branch.
- edit branch `import-config-1644762285` in your fork
- create a pull request with your branch `import-config-1644762285` as the source and our branch `import-config-1644762285` as the target

PR #4231 (2022-04-20): feat(config): add Nexia One Touch NX1000
fixes: #4230

**TODO:**
- [x]	Change PR title to be more specific
closed 2022-04-20 by AlCalzone