zwave-js/zwave-js #4243: Release anonymized, aggrigated usage statistics?

kdknigga, 2022-02-16:
Has there been any thought about releasing some of the statistics the [statistics-server](https://github.com/zwave-js/statistics-server) is collecting?

Perhaps something similar to [Home Assistant Analytics](https://analytics.home-assistant.io/).

It would be super interesting to see what trends there are in the Zwave community, things like, but not limited to:

* What percentages of devices are 300, 500, and 700 series
* Popular device models
* How many nodes there are in an average network
* How "healthy" networks tend to be (hop counts, message loss, etc)

Thanks!

AlCalzone, 2022-02-21:
We do share some statistics from time to time on our [Slack](https://join.slack.com/t/zwave-js/shared_invite/zt-8ns655f6-d407vtI~KjU~1z11jyaQ9Q) in the #statistics channel. We don't share a live view or something because some of the queries are quite heavy.

To answer some of the questions:

**What percentages of devices are 300, 500, and 700 series**
This is impossible to tell at the moment, since we don't record the SDK version and such

**Popular device models:**
At the time of writing:
[image]

**How many nodes there are in an average network**
13.6 currently, which has been a **very** steady (+/- 0.1) average since we started collecting the data.

**How "healthy" networks tend to be (hop counts, message loss, etc)**
We don't collect that either. See https://zwave-js.github.io/node-zwave-js/#/data-collection/data-collection?id=usage-statistics for an overview of what is collected.

kdknigga, 2022-02-21:
Thanks!

I'll watch the Slack channel for more interesting stats!

closed 2022-02-21 by kdknigga