zwave-js/zwave-js #4247: Qubino Mini Dimmer ZMNHHD1 not sending status updates for certain CCs

Schlurcher, 2022-02-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Toggling the wall switch doesn't send a status update for the binary switch CC.

I think this problem is very closely related to #3666.

### Device information

Manufacturer: Qubino 
Model name: Mini Dimmer ZMNHHD1 
Node ID in your network: 21

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.11.4
`node-zwave-js` branch: ???
`zwavejs2mqtt` branch: 6.5.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8084679/zwavejs_current.log)

AlCalzone, 2022-02-17:
Unlike https://github.com/zwave-js/node-zwave-js/issues/3666, the associations do get set up here. It also seems to send power reports, so the association is working.

I can't see anything obviously wrong, so maybe this is a configuration issue on the device. Please contact the Qubino support for help.

Side note: Isn't this a dimmer? In this case it should be sending Multilevel Switch reports, no?

Schlurcher, 2022-02-17:
It's configurable as a switch or dimmer. It's currently configured as a switch since I don't have a dimmable light (yet).
[image]

I'm pretty sure it didn't work when setting it as a multilevel switch either, but I will try it once more.

I will also raise a ticket with Qubino.

flemmingss, 2022-02-17:
This might be a little off-topic, but I also has some problems with ZMNHHD1 , so I tried to exclude it to re-include it. But no mater what I do I am not able to re-add it to my network. This one was originally added with openzwave and later migrated to open zwave2mqtt

Schlurcher, 2022-02-17:
Okay, I've tested it as a dimmer. Configured like this it sends a multi level report.

```
2022-02-17T12:35:48.078Z SERIAL « 0x010b0004001803260363ca0067                                        (13 bytes)
2022-02-17T12:35:48.079Z CNTRLR   [Node 024] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
2022-02-17T12:35:48.082Z SERIAL » [ACK]                                                                   (0x06)
2022-02-17T12:35:48.084Z DRIVER « [Node 024] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 99
```

Binary switch is unaffected by changes to the multilevel switch, unless manually refreshed.
[image]

Seems like the binary switch report is just not sent out at all. 
On a side note, it seems a bit odd that the binary switch exists in dimmer mode at all.

AlCalzone, 2022-02-17:
Okay, then this is a case for Qubino 🤷🏻‍♂️

Schlurcher, 2022-02-17:
Yep, I'll report back when I have news.

Schlurcher, 2022-02-17:
Okay, after fiddling around with it some more (up to node 29 now), including, excluding and changing parameters like a mad lad, it seems to work now. Not going to touch it anymore now. 

Sorry for the falsish alert. :(

Anyways, theres a paramter in the ui (parameter 6, load type) that isn't mentioned anywhere in the manual and seems to time out when trying to configure it.

[image]
[image]
https://qubino.com/wp-content/uploads/2019/09/[Qubino_Mini-Dimmer-PLUS-extended-manual_eng_3.4_09092019.pdf](https://qubino.com/wp-content/uploads/2019/09/Qubino_Mini-Dimmer-PLUS-extended-manual_eng_3.4_09092019.pdf)

closed 2022-03-01 by Schlurcher