zwave-js/zwave-js #4266: Unresponsive network on 700 series stick (fw 7.17.1)

alterbridge86, 2022-02-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown

What do you observe?
The network becomes unresponsive until zwavejs is restarted.

What did you expect to happen?
Network to remain functional.

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: zwavejs2mqtt - 6.5.2 / zwavejs 8.11.6
`node-zwave-js` branch: latest
`zwavejs2mqtt` branch: latest

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Moved to an Aeotec 700 series stick. 

Note: I'm aware the 700 series sticks have issues currently, I moved over intentionally to try it out after the latest firmware (7.17.1) as the issues were supposed to be largely resolved, however it appears there may still be some problems. 

### Did this work before?

Yes (please describe)

### If yes, where did it work?

On the Aeotec 500 series stick.

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8109989/zwavejs_current.log)

The unresponsive network begins around 13:11:41.

alterbridge86, 2022-02-21:
What's indicative of the issue from my end is Node 008 and Node 014 seem to be repeating the same messages over and over starting at 13:11:41 until I restarted zwavejs about an hour later. Just looking at the zwavejs log, from 00:00 to 13:11 there was ~5400 lines of log. From 13:11 to the service restart at 14:14 there was nearly 210,000 lines of log created. I'd think this was a node issue but the fact that it stops after restarting zwavejs and not touching the nodes makes me think something else is going on.

I'm not able to reliably trigger this behavior, but every time it has happened has been during the same automation which does send a couple of commands rapid fire. Node 13 is a door sensor and Node 14 is a RGB LED controller. I have it set that when Node 13 reports the door open, it immediately sends a command to turn on Node 14. When the door is closed, there is a 5s delay then it will turn off Node 14 so long as the door isn't opened again prior to that. This has also occurred on Node 10/11, which have the same arrangement and automation (10 is door sensor, 11 is RGB LED controller).

When this happened previously I tried to send commands via the zwavejs2mqtt webui and it also was not being processed, though I really should have tried this again so it was captured in the log here. If there are no further insights about the Node 8/14 being stuck broadcasting messages this issue pretty regularly repeats every day or two, I will attempt more controls outside of Home Assistant.

AlCalzone, 2022-02-21:
> What's indicative of the issue from my end is Node 008 and Node 014 seem to be repeating the same messages over and over starting at 13:11:41 until I restarted zwavejs about an hour later.

Those are not the same messages though. Note how the session ID for the SupervisionReport changes.
Just looking at node 14, each time, something sets the targetValue for node 14 to 99:
```
2022-02-21T13:11:38.174Z DRIVER » [Node 014] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      209
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 149
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 1
                                      └─[SupervisionCCGet]
                                        │ session id:      47
                                        │ request updates: true
                                        └─[MultilevelSwitchCCSet]
                                            target value: 99
                                            duration:     default
```
The node confirms execution:
```
2022-02-21T13:11:38.251Z DRIVER « [Node 014] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -89 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 25
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      1
                                      │ destination: 0
                                      └─[SupervisionCCReport]
                                          session id:          47
                                          more updates follow: true
                                          status:              Working
                                          duration:            1s
```
One second later it reports that it is done:
```
2022-02-21T13:11:39.267Z DRIVER « [Node 014] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -90 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 26
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      1
                                      │ destination: 0
                                      └─[SupervisionCCReport]
                                          session id:          47
                                          more updates follow: false
                                          status:              Success
                                          duration:            0s
```
Then (I'm not sure why, this shouldn't happen), the driver verifies the change:
```
2022-02-21T13:11:39.303Z DRIVER » [Node 014] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      210
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 150
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 1
                                      └─[MultilevelSwitchCCGet]
...
2022-02-21T13:11:39.455Z DRIVER « [Node 014] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -90 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 27
[... 36 lines cut ...]
                                  │ callback id:      211
                                  └─[BinarySwitchCCSet]
                                      target value: false
2022-02-21T13:11:42.715Z DRIVER » [Node 008] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      213
                                  └─[BinarySwitchCCGet]
2022-02-21T13:11:42.764Z DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -75 dBm
                                  └─[BinarySwitchCCReport]
                                      current value: false
...
2022-02-21T13:11:43.000Z DRIVER » [Node 008] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      214
                                  └─[BinarySwitchCCSet]
                                      target value: false
...
```

I'd say this is an automation that went rogue or something.

alterbridge86, 2022-02-21:
Thanks for the analysis! I didn't realize it was _sending_ those state changes to the devices, not just receiving reports. I was thinking to myself why is this thing telling me it's set the value to 99 over and over. 

I suppose restarting the zwavejs container is enough to break the connection to HA and interrupt whatever crazy thing is going on there that is sending this command repeatedly, which led me to the wrong conclusion that zwavejs was the culprit. 

I'm going to close this one out, and go ratchet up my logging in HA to catch the next time this happens. Thanks again for the help checking into this!

closed 2022-02-21 by alterbridge86

AlCalzone, 2022-02-21:
The arrows are telling you the direction of the command:
Outgoing: `DRIVER » [Node 008]`
Incoming: `DRIVER « [Node 008]`

alterbridge86, 2022-02-21:
LOL that's brilliant...seems so obvious now. Thanks for the tip! 

General note, appreciate you taking the time to explain the _analysis_ rather than just throwing out solutions/diagnoses. Helps me learn the logs and troubleshooting process so I can do it myself next time.