zwave-js/zwave-js #4271: ECO-DIM Z-Wave Led dimmer not dimming properly

fifise77, 2022-02-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Hello,

The device ID is  1073-2-514 (0x0431-0x0202-0x0002)

When I try to set a new target value on the node ECO-DIM Z-Wave Led dimmer (nod14) most of the time the value I asked is not set properly after mutiply request the dimmer finaly apply the ask value

Steps to reproduce the behavior:
  1. Go to the node
  2. type a new [14-38-0] target value 
  3. click on tha rrow to send the new value
  4. the current value doesn't reflect the target value
 
In the drivers lof for example I requested a "target value" of 30 and the "current value" stay on 60 I request a second time to set the "target value" at 60 and it work, I than ask to set the "target value" at 70 and only on the 3 try did the "current value" finaly got to 70.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8112376/zwavejs2mqtt-store.zip)
[node_14.json.txt](https://github.com/zwave-js/node-zwave-js/files/8112379/node_14.json.txt)

### Device information

Manufacturer: ECO-DIM
Model name: Z-Wave Led dimmer
Node ID in your network: 14

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.5.2
zwave-js: 8.11.6
home id: 25483355
home hex: 0x184d85b

Home Assistant 2021.12.5

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-02-21.log](https://github.com/zwave-js/node-zwave-js/files/8112398/zwavejs_2022-02-21.log)

AlCalzone, 2022-02-28:
Here's a breakdown of what happens in your log:

zwave-js tells the dimmer to go to 30. It uses supervision, so it gets notified whether the command was handled:
```
2022-02-21T22:25:30.328Z DRIVER » [Node 014] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      2
                                  └─[SupervisionCCGet]
                                    │ session id:      19
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 30
```

The device replies with "Success", so zwave-js changes the currentValue to the targetValue you set before:
```
2022-02-21T22:25:30.373Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          19
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s
2022-02-21T22:25:30.377Z CNTRLR   [Node 014] [~] [Multilevel Switch] currentValue: 60 => 30         [Endpoint 0]
```

But 20ms later the device sends a report with the old status:
```
2022-02-21T22:25:30.401Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 60
```
and as a result, zwave-js sets the currentValue back to 60:
```
2022-02-21T22:25:30.397Z CNTRLR   [Node 014] [~] [Multilevel Switch] currentValue: 30 => 60         [Endpoint 0]
```

TL;DR: That's a device bug. Contact the manufacturer to get this sorted out.

closed 2022-03-03 by zwave-js-assistant[bot]

fifise77, 2022-03-05:
Thank for your help, I am in contact with manufacturer

stephanvierkant, 2023-12-06:
@fifise77 Is your problem solved? I'm having the same issue.