zwave-js/zwave-js #4273: Missing device configuration: Philio Technology Corp / Z-WAVE Remote PH-PSR03

jamespreedy, 2022-02-22:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Philio Technology Corp / Z-WAVE Remote PH-PSR03

### Manufacturer ID

0x013c

### Product Type

0x0008

### Product ID

0x0027

### Firmware Version

1.13

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1407?selectedFrequencyId=1

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/1407/PSR03_ABC_version_Manual.pdf

I have paired a Philio PSR03 remote with zwavejs2mqtt and it included fine.
However it shows as unknown for both manufacturer and model.

The link on the device configuration page points here, where the device is not found in the database.
https://devices.zwave-js.io/?jumpTo=0x013c:0x0008:0x0027:1.13

However, I believe this is the entry for it?
https://devices.zwave-js.io/?jumpTo=0x013c:0x0008:0x001d:0.0

Under device info, Home Assistant shows ...
Firmware: 1.13
Z-Wave Info
Source: Z-Wave JS
Device ID: 80
Device Status: Asleep
Device Ready: Yes
Highest Security: None
Z-Wave Plus: Version 1

There are three entities created for this remote.
(The remote has a single button marked "PANIC", nothing else)
sensor.annex_panic_button_battery_level
binary_sensor.annex_panic_button_low_battery_level
sensor.annex_panic_button_node_status

I've included my node dump.
[node_80.txt](https://github.com/zwave-js/node-zwave-js/files/8120359/node_80.txt)

Thanks for your help!

AlCalzone, 2022-02-23:
@zwave-js-bot import config 1407 from zwa

jamespreedy, 2022-02-23:
Awesome sauce - thank you very much! 🍺

PR #4274 (2022-02-24): feat(config): add Philio PSR03-A, -B and -C
fixes: #4273

**TODO:**
- [x]	Change PR title to be more specific
closed 2022-02-24 by AlCalzone