zwave-js/zwave-js #4276: Alarm.com ADC-T2000 Configuration Parameter 5

klyoku, 2022-02-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The configuration for parameter 5 has some form of mask/offset, so that the valid range is not actually -100 to 100. For example, 704643072 is actually 0. As a result, zwavejs gives an error when trying to set this configuration parameter:
[image]

Here is how it should be set written by the developer of the device handler from smartthings:
https://community.smartthings.com/t/release-alarm-com-z-wave-thermostat-adc-t2000/74833/28

### Device information

Manufacturer: 0x0190
Model name: adc-t_2000

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

8.11.6

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

to be attached when required.

AlCalzone, 2022-02-23:
I wonder if this is the same as for the ADC-T3000, that was added here: https://github.com/zwave-js/node-zwave-js/pull/4131
I guess parts of the new config from there need to be carried over. @EiNSTeiN- you don't have a T2000 by any chance? :)

EiNSTeiN-, 2022-02-23:
I have 2 of those but not connected to HA right now... Last time I connected one, I did notice the zwave float values and I made a note to open a PR when I get ready to transfer my T2000 thermostats to HA.

If someone else wants to tackle this first, it should be a matter of adding a `0xffff00` bitmask [like this](https://github.com/zwave-js/node-zwave-js/blob/2694346fd30847082a89829ab43e4997fb7fd106/packages/config/config/devices/0x0190/adc-t_3000.json#L72-L118) on every configuration parameter that represents a temperature in the adc-t2000 config.

blhoward2, 2022-04-16:
I don't think that you're trying to set a valid value. Our parameter matches the zwave alliance DB, OpenHAB, and the SmartThigns handler that you referenced. At SmartThings, the parameter is the same as ours:

<Value type="byte" byteSize="4" index="5" label="Calibration Temperature" min="-100" max="100" value="0" setting_type="zwave" fw="" disabled="false">
 <Help>
Calibration Temperature Range (in deg. F) Precision is tenths of a degree.
Range: -100 to 100
Default: 0
</Help>
</Value>

klyoku, 2022-04-16:
I've tried to set it with -100 to 100 in the beginning. It says the parameter has been updated, but the thermostat doesn't reflect the change, and if I refresh the Z-Wave device configuration page, I see the number is reverted. See below screenshots:

This is when I set it to 20. As you can see, no error is shown:
[image]

The thermostat reading is not reflected after the change:
[image]

And if I go back to the device configuration page, the number has been reverted. (note this used to say 704643072 when I initially opened this issue. It has recently changed to truncate the digits)
[image]

Let me know how I can submit a more detailed report/log if you suspect I've done something wrong instead of the bitmask issue as suggested by @EiNSTeiN- [EiNSTeiN](https://github.com/EiNSTeiN-)

blhoward2, 2022-04-17:
That looks accurate to me? It's at 22.4 and you've set two tenths precision. That allows it to exceed the set temp for the defined range (dead band). Part of the difficulty in telling is that the device is looking for Fahrenheit but you're HA is in Celsius. 

This wouldn't be the first device to report back an improper value. We'd need to see the zwavejs logs at a silly level.

EiNSTeiN-, 2022-04-17:
Keep in mind the value is encoded as a zwave float. The top 8 bits are the size/scale/precision, the bottom 8 bits are unused, the middle 16 bits are the actual value. Everything except the middle 16 bits are read only so all you need to do is shift the value you want to set by 8 bits (multiply by 256). Instead of 100, enter 25600 and that should set that zwave float value to 10.0.

blhoward2, 2022-04-17:
That is not true. Not on this device per the manual or the Smartthings handler.

EiNSTeiN-, 2022-04-17:
Yes it is. What else does 704643072 decode to if not a zwave float. It is 0x2A000000, where 0x2A is the correct precision/size/encoding for tenths of a degree, Fahrenheit, 2 byte value.

blhoward2, 2022-04-17:
We've seen lots of devices before that report back something completely erroneous. All I'm saying is so far we have a screenshot with one value being set. The fact it's reporting that back doesn't mean that is the correct format to set it.

EiNSTeiN-, 2022-04-17:
Gotcha, I agree I haven't tested. Can do it tomorrow as I've recently connected my T2000 to HA

AlCalzone, 2022-04-27:
@EiNSTeiN- any new insights?

EiNSTeiN-, 2022-04-28:
Sorry for the delay

I had to increase `maxValue` for one of the parameter to test this. I changed parameter 5 from 704643072 (0x2A000000) to the new value 704645632 (0x2A000A00) which should set the calibration temp to 10 (in 0.1F increments, so 1 degree).

I tried setting the value to 704668672/0x2A006400 (or how I expect the value 100 would be encoded) for parameter 5, this works and the value sticks, but anything above 100 (e.g. 101 would be encoded as 0x2A006500) does not stick and reverts back, which points to what I would expect if the value is encoded into the second and third bytes like for the ADC-3000.

I also tested setting the value back to 0x2A000000 and then to 2A0064FF (i.e. 100 for the value, and all bits set for the lower byte). This sets the value to 704668672/0x2A006400 as we would expect if the bottom byte is unused or readonly.

closed 2022-04-29 by AlCalzone

bachya, 2022-07-09:
Can someone clearly explain how to translate these "Z-Wave float" values into something sensible? I'm not following. I have an ADC T-2000 and currently, several parameters contain these types of values (e.g., 704643072) – it's unclear how to alter them. Thanks!