zwave-js/zwave-js #4283: Impossibile to include/Replace/Exclude Nodes

candrea77, 2022-02-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Exclusion / Inclusion process

What do you observe?
Unable to complete the inclusion / exclusion / replace process

What did you expect to happen?
To be able to include / exclude / replace any node

Steps to reproduce the behavior:
  1. Have a failed node to replace
  2. Replace failed node
  3. Selected the node to replace
  4. Selected S2 Authenticated
  5. Start the replace process
  6. See in the Log "Controller is busy and cannot handle this inclusion request right now..."

Please note , currently if I move my Aeotec Zwave USB Key to a PC with Silab PC controller 5 , I can do inclusion/exclusion/replace of any zwave device without any issue.
AEON Labs Z‐Stick Gen5 USB Controller ZW090 v1.2

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.5.2
zwave-js: 8.11.6

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs2mqtt_2022-02-25.log](https://github.com/zwave-js/node-zwave-js/files/8143149/zwavejs2mqtt_2022-02-25.log)

AlCalzone, 2022-02-25:
That's not a driver log

candrea77, 2022-02-25:
> That's not a driver log

Sorry for the mistake, I've updated the file into the original post

AlCalzone, 2022-02-25:
In the beginning, Node 152 got included successfully. I don't see any normal inclusion processes after that.

When you try to replace node 54, it looks like the SmartStart listening mode doesn't get disabled beforehand, so all I'm seeing are SmartStart inclusion requests and not the normal one that gets sent when you press a button. I think this is actually a bug.

PR #4559 (2022-05-03): fix: pause SmartStart listening mode before adding/removing/replacing nodes
...and improve the logs for AddNodeRequests while we're at it. Turns out what we had before looked like it paused SmartStart but it didn't actually do it.

fixes: #4283
closed 2022-05-03 by AlCalzone