zwave-js/zwave-js #4308: Sunricher LED-Driver not responding reliably

Maes152, 2022-03-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I've been using ZwaveJS2MQTT since the start. It's worked incredibly reliable for me in all cases. I have 30 z-wave nodes.
I recently purchased 2 z-wave LED drivers KLEMKO LD-CC-ZV-MO-50 [product page](https://www.klemko.nl/producten/leddriver-constante-stroom-z-wave-voor-enkel-kleurige-ledverlichting-instelbaar-met-dipswitches-250ma-1500-ma-50w/860238)
This is a rebrand of the Sunricher ZV9102FA-CCT (according to the manufacturer information in ZwaveJS2MQTT). I cannot find this product on Sunricher's page but I found this OpenHab page: [github](https://github.com/openhab/org.openhab.binding.zwave/blob/main/doc/sunricher/zv9102facct_0_0.md).
Also, the product picture doesn't align with the product I have as you can see on the KLEMKO website. I believe it's actually a rebrand of [this product](https://www.sunricher.com/50w-constant-current-z-wave-led-dimmable-driver-srp-zv9105-50w-cc.html).

Anyway. The problem is that these devices do not respond in a clear fashion as I'm used to from ZwaveJS. Sometimes the command doesn't come through at all. I'm observing timeouts from the log.

If I operate the devices individually they -usually- respond fine, albeit with a delay. I've created a light group in home assistant that executes my commands on both devices simultaneously but this creates an unacceptable delay as well as often times 1 of the 2 devices not receiving the command at all. 

Placing the devices right next to the controller showed the same behavior.
I am still using an (old) Aeotech Z‐Stick Gen5 (ZW090)

### Device information

Manufacturer: KLEMKO   // Sunricher
Model name: KLEMKO LD-CC-ZV-MO-50 // ZV9102FA-CCT
Node ID in your network: 39 & 40

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.5.2
zwave-js: 8.11.6

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Multiple heals
Exclude/include without security
Multiple re-interviews
Physically changed location (same behaviour even when directly next to controller)

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/8163191/zwave_js.log)
NodeID's to look for: 39 & 40. 
This is a log of me turning on the light group. Dimming the lights from 10% to 100%, shutting off the lights. 
In this instance, shutting off the lights did not come through correctly.

AlCalzone, 2022-03-02:
There seems to be at least some connectivity problem between the controller and node 39. Here's an example of the message exchange, commented:

```
Turn light to 74:
2022-03-01T16:22:43.684Z DRIVER » [Node 039] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      106
                                  └─[SupervisionCCGet]
                                    │ session id:      44
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 74
                                        duration:     default

Device updates its status:
2022-03-01T16:22:44.070Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[BasicCCReport]
                                      current value: 74

Then it confirms that it is executing the command, half a second later:
2022-03-01T16:22:44.629Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          44
                                      more updates follow: true
                                      status:              Working
                                      duration:            1s

Another second later it tells the driver that it is done:
2022-03-01T16:22:46.351Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          44
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s

and again:
2022-03-01T16:22:46.771Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          44
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s

and again:
2022-03-01T16:22:49.007Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          44
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s
...
2022-03-01T16:22:49.200Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          44
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s

meanwhile not responding to the value refresh:
2022-03-01T16:22:46.362Z DRIVER » [Node 039] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      108
                                  └─[MultilevelSwitchCCGet]
...
2022-03-01T16:22:58.771Z CNTRLR   [Node 039] Timed out while waiting for a response from the node (ZW0201)
```

This is going to be hard to diagnose without a Zniffer, but just judging from that behavior, I assume that the controller's acknowledgements and/or commands often don't reach the device. This could be an issue with the controller placement - try following the first section here: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

Maes152, 2022-03-03:
I will purchase an USB extension cord to make sure this isn't the problem. However, it would surprise me greatly. I've had 28 nodes for a very long time without any issues. One of those nodes is all the way in the back of my garden.

I've also tried placing these 2 led drivers next to the controller but unfortunately the behaviour did not change.

AlCalzone, 2022-03-03:
> However, it would surprise me greatly

If I had to guess, the success rate of this is roughly 80%. A lot of issues can be explained by interference and noise, which USB ports create a lot of - especially around the frequency that Z-Wave uses.

Maes152, 2022-03-03:
That certainly sounds promising. I'll receive the cord on Saturday and will keep you posted.

AlCalzone, 2022-03-04:
Don't forget to heal the network when you install the cord and move the controller - this may change a few routes.

Maes152, 2022-03-04:
That's a good tip, thank you.