zwave-js/zwave-js #4330: Scene Actuator Configuration Report with SceneID = 0 should not be dropped, used by Leviton VRCS2

j9brown, 2022-03-07:
I'm having trouble writing code to refresh the state of a Leviton VRCS2 when its buttons are pressed.  Unfortunately, this device does not implement Hail and does not send Basic Set messages when the load state changes in response to scene activation so it must be handled a different way.

The Leviton VRCS2 consists of three nodes: one scene controller and two loads.

The scene controller has two association groups and two scenes.  By default the scene controller's association groups are associated with each of the loads.

The loads each have one association group which can be used as a lifeline to inform the hub about state changes.

When the user presses a button on the scene controller, the associated load receives a Scene Activation message and toggles itself on or off.  The load then sends a Scene Actuator Configuration Report to its association group to confirm the scene activation.

When the load is toggled on by a scene activation, the Scene Actuator Configuration Report has the button's associated scene ID as was configured in the Scene Controller Configuration.

When the load is toggled off by a scene activation, the Scene Actuator Configuration Report has a Scene ID of 0.

In principle, we can use these reports to trigger a refresh of the load's state (and I previously did this before upgrading to Zwave JS from OZW).  Unfortunately, Zwave JS does not persist Scene Actuator Configuration Reports that contain a Scene ID of 0 so it's not possible to handle the case when the load is toggled off by a scene activation.

According to the spec, a Scene ID of 0 is valid and it refers to the currently active scene.  Given that Zwave JS doesn't always know the currently active scene, I think it'd be ok to simply handle Scene ID 0 in the same way as any other Scene ID (from 1 to 255) and report the level and dimming values accordingly.

https://github.com/zwave-js/node-zwave-js/blob/67e716a2fcf6477f10afb96fec8978cae91f990e/packages/zwave-js/src/lib/commandclass/SceneActuatorConfigurationCC.ts#L446

j9brown, 2022-03-07:
Come to think of it, the way these reports are modeled as values doesn't make much sense.  The report indicates a current state (which scene is active) and the level and dimming duration that were applied when activated.

I think it would make more sense to remove the 1..255 values indexed by scene id and instead add a single "currentScene" value whose properties are the scene id, level, and dimming duration similar to how currentValue works for the basic set CC.

AlCalzone, 2022-03-07:
The values are currently modeled in a way that makes sense with what's in the specification:
> The Scene Actuator Configuration Command Class is used to **configure scenes** settings for a node supporting an actuator Command Class

Each of these 2*255 values allow configuring the settings for one scene. The way I understand it, a `Scene Actuator Configuration Report` with scene ID 0 (no scene active) is meant as a response to a `Scene Actuator Configuration Get` with scene ID 0 (which scene is active?)

IMO it would have made more sense to use the `Scene Activation Command Class` to report the current scene, which a lot of other devices also do. Its whole purpose is to indicate (or rather control, but reality is a bit different) which scene should be active.

I guess what could be done is what you proposed and add a `sceneId` value which contains the ID of the last activated scene. This would get updated for responses to a GET with scene ID 0 and unsolicited reports which the driver did not query. The downside of this approach is that re-transmitted responses to a GET will be interpreted as unsolicited and cause an unwanted change of this value.

j9brown, 2022-04-26:
More notes for later.

When the scene ID for a button is configured to 0, the scene controller sends `Basic Set` to the associated nodes when the button is pressed.  The button's LED indicator automatically tracks the state of the associated local loads.

When the scene ID for a button is configured to non-zero, the scene controller sends both `Basic Set` and `Scene Activation Set` (with the given scene ID) to the associated nodes when the button is pressed.  The button's LED indicator does not automatically track the state of the associated local loads.

When the hub is associated with both groups in the scene controller, both buttons' LED indicators become coupled and turn on/off together so they no longer track the state of the associated local loads correctly.  Unsure if the configured scene IDs have any effect on this behavior.

I think I'm kind of stuck.  I want local load control (for latency and robustness), I want the button LED indicators to track the local loads correctly, and I want to avoid polling the load state.  Stuff breaks if the scene IDs and associations aren't configured just right.

I don't suppose we could add a compat flag to remap that weird `Scene Actuator Conf Report` to `Hail`?

AlCalzone, 2022-04-27:
I'm not a fan of having a compat flag for each buggy device, but I still think the approach outlined in my last comment could help you:

> I guess what could be done is what you proposed and add a `sceneId` value which contains the ID of the last activated scene. This would get updated for responses to a GET with scene ID 0 and [...]

This would mean that you have a value you could base an automation off and query the load states yourself, like you used to.

Side note: Getting the manufacturer to fix their mess is out of the question?

j9brown, 2022-04-28:
Ahh yes what you propose could work.  Even just an event notification with no special value semantics or retention would work for my case.

Unfortunately the device is no longer in production as far as I know and it seems pretty unlikely they would publish a firmware update (it might not be OTA updatable anyhow).

Thanks for spending the time to look into this with me.

j9brown, 2022-05-06:
Coincidentally, I captured an example of the deserialization issue while reporting a different bug.  Here's the log from that, in case it's helpful for writing a test case later on:

```
2022-05-06T19:55:45.180Z SERIAL « 0x010b00040021032c0200b50049                                        (13 bytes)
2022-05-06T19:55:45.182Z DRIVER   Dropping message because it could not be deserialized: SceneActuatorConfigurat
                                  ionCCGet: deserialization not implemented (ZW0320)
2022-05-06T19:55:45.182Z SERIAL » [ACK]     
```