zwave-js/zwave-js #4333: [ClockCCSet] after [ClockCCReport] sets device clock back one hour when minutes are rounded up from 59 to 00.

p3nguin53, 2022-03-08:
Bug occurs when updating time on Aeotec ZWA023 device.

Example of problem from zwave log  - 1st ClockCCSet is ok, 2nd ClockCCSet has wrong hour:

2022-03-07T00:59:22.072Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[ClockCCReport]
                                      clock setting: Sunday, 19:00
2022-03-07T00:59:22.074Z CNTRLR   [Node 016] detected a deviation of the node's clock, updating it...
2022-03-07T00:59:22.078Z SERIAL » 0x010b001310048104f23b250993                                        (13 bytes)
2022-03-07T00:59:22.078Z DRIVER » [Node 016] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      9
                                  └─[ClockCCSet]
                                      clock setting: Sunday, **18:59**

2022-03-07T00:59:37.740Z DRIVER « [Node 015] [REQ] [ApplicationCommand]
                                  └─[ClockCCReport]
                                      clock setting: Sunday, 19:00
2022-03-07T00:59:37.741Z CNTRLR   [Node 015] detected a deviation of the node's clock, updating it...
2022-03-07T00:59:37.745Z SERIAL » 0x010b00130f048104f200250ab4                                        (13 bytes)
2022-03-07T00:59:37.745Z DRIVER » [Node 015] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      10
                                  └─[ClockCCSet]
                                      clock setting: Sunday, **18:00**

I found the related section of code in node-zwave-js  Node.ts  at lines 3081-3130 in the version that I looked at.

**If this code is updated to increment the hour when the minutes rounds up from 59 to 00, then it should also set the day to the next day when the time rounds up from 23:59 to 00:00.**

I am a newbie to HomeAssistant so I am only reporting this problem and requesting that someone fix this please.
My Zwave Log is in UTC time.  I think there is an existing request to display Local Time in this log.

[zwave_jslog_4.log](https://github.com/zwave-js/node-zwave-js/files/8201729/zwave_jslog_4.log)

p3nguin53, 2022-03-12:
I'm not great at deciphering code.  Will these changes handle rounding to midnight?  For example if 'now' is Tuesday 23:59:35, what will the device clock be set to?

 
 

Closed #4333 via #4346.

p3nguin53, 2022-03-12:
I'm not great at deciphering code.  Will these changes handle rounding to midnight?  For example if 'now' is Tuesday 23:59:35, what will the device clock be set to?

I originally replied to you thru a forwarded mailbox.  Sending this again in case my 1st reply didn't go thru properly.

AlCalzone, 2022-03-12:
They won't but thanks for mentioning it. Will put up another PR shortly.

PR #4346 (2022-03-10): fix: increment hours when necessary while rounding time
fixes: #4333
PR #4354 (2022-03-12): fix: let JS handle the date computation for rounding times
fixes: #4333
(for real this time)
closed 2022-03-12 by AlCalzone