zwave-js/zwave-js #4335: RangeError after trying to include vision ZD2102EU-5 door/window sensor

VerbruggenBart, 2022-03-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have trouble including the vision ZD2102EU-5, which led me to look at the log-files (pointed there by the home assistant community)

In the zwave js to mqtt add-on I noticed that the node (now node 4) did not seem to be interviewed properly, even after waking the device up manually for a significant amount of times (it only stays awake for 10secs, so I kept repeating this while eating).
However, it kept saying unkown manufacturer/unknown product and there is no sensor available, except once I got it to show the battery somehow.

Obviously I hoped the device would show up and I would have a binary sensor for the open/close status.

How I got to this point:
1) I excluded the device from the old 'gateway'
2) I started inclusion through the zwave js to mqtt interface 
3) I put in the battery in the sensor to get it into inclusion mode (in a second attempt I just opened the lid with the same effect)
4) I waited
5) I tried to wake up the device a lot
6) I tried to re-interview
7) I tried to get help from the home assistant community
8) I started silly loggin
9) I noticed the error below and got pointed here :)

The error that got me sent here:

2022-03-08T05:44:13.655Z DRIVER   RangeError: The value of "offset" is out of range. It must be >= 0 and <= 14. 
                                  Received 15
                                      at new NodeError (node:internal/errors:371:5)
                                      at boundsError (node:internal/buffer:86:9)
                                      at Buffer.readInt8 (node:internal/buffer:424:5)
                                      at parseRSSI (/opt/node_modules/zwave-js/src/lib/controller/SendDataShared
                                  .ts:91:22)
                                      at new BridgeApplicationCommandRequest (/opt/node_modules/zwave-js/src/lib
                                  /controller/BridgeApplicationCommandRequest.ts:81:24)
                                      at Function.from (/opt/node_modules/zwave-js/src/lib/message/Message.ts:22
                                  8:15)
                                      at Driver.serialport_onData (/opt/node_modules/zwave-js/src/lib/driver/Dri
                                  ver.ts:2135:18)
                                      at SerialAPIParser.emit (node:events:520:28)
                                      at SerialAPIParser.emit (node:domain:475:12)
                                      at addChunk (node:internal/streams/readable:315:12)

### Device information

Manufacturer: Vision
Model name: ZD2102EU-5
Node ID in your network: 4 (in first attempt it was node 3)

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Z-wave js:
Driver Version: 8.11.6
Server Version: 1.15.0

Z-wave to MQTT:
version: 0.35.2

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It worked with the fifthplay gateway, however fifthplay stopped its operation and I wanted to recycle some of the devices.
Obviously I excluded them from the fifthplay gateway first.

### Attach Driver Logfile

[zwavejs_2022-03-08.log](https://github.com/zwave-js/node-zwave-js/files/8204634/zwavejs_2022-03-08.log)
[zwavejs_2022-03-07.log](https://github.com/zwave-js/node-zwave-js/files/8204639/zwavejs_2022-03-07.log)

AlCalzone, 2022-03-08:
Sounds like a bug in the serial parser with a probably easy fix. Will try to get this fixed tonight.

closed 2022-03-08 by AlCalzone

AlCalzone, 2022-03-08:
Fixed on master and 8.11.9 (which is going to be released in a few minutes)

VerbruggenBart, 2022-03-09:
Wow, what a quick response!
Wonderful! 👏

PR #4337 (2022-03-08): fix: parsing of BridgeApplicationCommandRequest w/o RSSI
fixes: #4335