zwave-js/zwave-js #4340: Missing device configuration: <enter manufacturer and device name>

jlmaners, 2022-03-09:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Minoston Outdoor Plug Dimmer

### Manufacturer ID

0x0312

### Product Type

0xFF07

### Product ID

0xFF03

### Firmware Version

1.00

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4254

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/4254/MP22ZD-manual-20210728.pdf

AlCalzone, 2022-03-09:
@zwave-js-bot import config 4254 from zwa

AlCalzone, 2022-03-09:
The bot has created a PR but the file only serves as a starter template. It needs to be cleaned up to conform with our style guide, see above.

For anyone who wants to help out:
- fork the repo if you haven't already
- otherwise synchronize your fork so it is up to date with this repo
- edit branch `import-config-1646811815` in your fork
- create a pull request with your branch `import-config-1646811815` as the source and our branch `import-config-1646811815` as the target

blhoward2, 2022-04-16:
This device is already in our database.

closed 2022-04-20 by AlCalzone