zwave-js/zwave-js #4341: Support bidirectional Manufacturer Proprietary messages

j9brown, 2022-03-09:
**Is your feature request related to a problem? Please describe.**

I'm writing [some code](https://github.com/j9brown/node-zwave-pe653) to update the firmware of my Intermatic PE653 spa controller.  The protocol is quite simple but it requires sending and receiving `Manufacturer Proprietary` messages.

The `zwave-js` API already offers a [sendData function](https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/ManufacturerProprietaryCC.ts#L37) to send `Manufacturer Proprietary` messages but it does not currently support receiving replies, which I need for this project.

**Describe the solution you'd like**

I'd like an API that sends a manufacturer proprietary command and returns the payload of the received manufacturer proprietary response.  Say, something like `sendAndReceiveData`.  We can safely assume that the response will have the same `manufacturerId` as the command that was sent.

I think this could be implemented by creating a subclass of `ManufacturerProprietaryCC` that is suitably decorated to expect a `ManufacturerProprietaryCC` in response.  Or perhaps `DynamicCC` could be used here.  I'm not very familiar with the zwave-js API.

**Describe alternatives you've considered**

I considered trying to implement this functionality on top of the available `zwave-js` API but it seems pretty tricky because the library heavily relies on reflection and I'm poking at it from the side via `zwavejs2mqtt`.  So an actual `zwave-js` API function would be best.

**Additional context**

My little afternoon project: [node-zwave-pe653](https://github.com/j9brown/node-zwave-pe653)

AlCalzone, 2022-03-09:
I haven't tested it recently, but it should be possible to subclass the ManufacturerProprietaryCC and override the following methods on the subclass:
- [expectsCCResponse](https://github.com/zwave-js/node-zwave-js/blob/47eb088858129708b07e97c9bcfdd9ea9294bfc5/packages/zwave-js/src/lib/commandclass/CommandClass.ts#L868)
- [isExpectedCCResponse](https://github.com/zwave-js/node-zwave-js/blob/47eb088858129708b07e97c9bcfdd9ea9294bfc5/packages/zwave-js/src/lib/commandclass/CommandClass.ts#L888)
Then you should be able to `sendCommand` that CC and receive the response in return.

It could make more sense to bake this in though, like we already do for the [Fibaro CC](https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/manufacturerProprietary/Fibaro.ts). That way you could also expose CC specific [API methods](https://github.com/zwave-js/node-zwave-js/blob/93aa398d8df97f35e17cfa0d000ee785a2015cd6/packages/zwave-js/src/lib/commandclass/ManufacturerProprietaryCC.ts#L52). I'm not very happy with how the correct subclass is selected currently, but since usage of the ManufacturerProprietaryCC is pretty low, we can "hardcode" the selection for now.

j9brown, 2022-05-06:
Coincidentally, I captured an example of the deserialization issue while reporting a different bug.  Here's the log from that, in case it's helpful for writing a test case later on:

```
2022-05-06T19:55:45.180Z SERIAL « 0x010b00040021032c0200b50049                                        (13 bytes)
2022-05-06T19:55:45.182Z DRIVER   Dropping message because it could not be deserialized: SceneActuatorConfigurat
                                  ionCCGet: deserialization not implemented (ZW0320)
2022-05-06T19:55:45.182Z SERIAL » [ACK]     
```

j9brown, 2022-05-06:
Yeah, I saw the manufacturer proprietary hardcoding for Fibaro.  There's repeated logic in several places and it looks pretty fragile.  Rather than pile on to what's there, I think I'd rather hold off until there's a more robust delegation scheme in place for such extensions.

So instead, I might try to implement a general purpose `sendAndReceiveData` method along the lines of the existing `sendData` method since it doesn't require as much glue and can be used with any device.

PR #4586 (2022-05-11): feat: add `sendAndReceiveData` method to `ManufacturerProprietaryCCAPI`
Adds an API that can be used to send ManufacturerProprietaryCC messages that expect responses from the device.

fixes: #4341
closed 2022-05-11 by AlCalzone