zwave-js/zwave-js #4352: Aeotec Z-Stick 700 with FW 7.17.2.0 - "Duplicate command" errors after backup restoration

mundschenk-at, 2022-03-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Yesterday, after updating the stick to FW 7.17.2.0, I restored a current NVM backup from my 500 series stick. The driver started up fine and the nodes seem to work OK. Unlike previous attempts, there were no dead nodes and no automatic re-interviews.

However, when checking the log files this morning, I noticed that it does contain a lot of "Duplicate command" errors, and not (just) from the [Sensative Strips Guard 700](https://github.com/zwave-js/node-zwave-js/issues/4053). Could this be a side effect of a "Health Check" attempt I made (under the erroneous assumption that certain issues had been fixed since the last time I tried)? I did restart the driver afterwards though, to stop the network flooding.

### Device information

Manufacturer: Aeotec
Model name:  Z-Stick 700
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:  8.11.6
`zwavejs2mqtt` branch: 6.5.2
`zwavejs2mqtt community addon`: 0.36.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Switched back to Z-Stick 700

### Did this work before?

Yes (please describe)

### If yes, where did it work?

With Z-Stick 5gen+

### Attach Driver Logfile

[zwavejslog_20220312.zip](https://github.com/zwave-js/node-zwave-js/files/8237826/zwavejslog_20220312.zip)

mundschenk-at, 2022-03-12:
> It looks like you attached a logfile, but its filename doesn't look like it a driver log.

It is.

AlCalzone, 2022-03-12:
For me, the logs look like the nodes don't get the expected acknowledgement from the controller and retry sending. Also there are some instances where the controller can't reach a node while the same node is sending reports to the controller, which would support my suspicion further.

I don't see any mention of you doing a network heal. Even if the backup restores the known routes, the physical differences between sticks may result in incorrect routes being used for attempting communication with the nodes.

mundschenk-at, 2022-03-12:
I had not done a network heal when that log file was created, no. I've since started one, takes some time for the battery powered nodes. I'll upload tomorrow's logfile which should be mostly after the healing (only the smoke detector is not done yet, not sure when it's scheduled for its daily wakeup).

mundschenk-at, 2022-03-12:
Addendum: According to the network graph, range still appears to be much worse with 700 series (though it's been advertised as having an improved range). I think I mentioned this previously in another ticket, I've a Strips Guard 700 on a window in the same room as the controller, maybe 5 m distance that takes 2 hops with the 700 series and only 1 hop with the 5gen+. This might explain the issues before the network heal (but not why range is so abysmal with the newer stick).

mundschenk-at, 2022-03-13:
Here's today's log, @AlCalzone. The network heal finished at 01:00 with the last remaining node (45 - the others were healed before the log turnover), but unfortunately to me it looks very similar (lots of invalid payloads/duplicate commands):
[zwavejslog_20220313.zip](https://github.com/zwave-js/node-zwave-js/files/8239478/zwavejslog_20220313.zip)

AlCalzone, 2022-03-13:
Not sure if you we through this already - did you follow the instructions for controller placement mentioned here?
https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

mundschenk-at, 2022-03-13:
We didn't, but I know about it and the stick's attached to a 1,5 m extension cord (above a wooden door frame, so no metal surface nearby either).

AlCalzone, 2022-03-13:
Ok. What bugs me a bit are the low RSSI values (-88...-100 dBm), which are probably not much above the background noise, and they gradually get worse. Maybe this is less of a controller placement issue than a mesh strength issue.

You can see that at least node 18 has a stronger connection than most others:
```
2022-03-13T09:11:21.169Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            228
                                    transmit status:        OK, took 100 ms
                                    repeater node IDs:      18, 10
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -89 dBm
                                    ACK RSSI on repeaters:  -63 dBm, N/A
                                                            ^^       ^^^ node 10
                                                            ^^ node 18
                                    ACK channel no.:        0
                                    TX channel no.:         0
```

In zwavejs2mqtt, on the network map, you have the ability to test the connection quality to individual nodes. I think this is your best bet now - identify nodes with a bad connectivity (might be the repeaters, not necessarily the "problematic" nodes) and try to optimize their connection.

mundschenk-at, 2022-03-13:
How would increase the controller's TX power? The health check on the network map is not working for me (it rapidly swamps the network, similar to https://github.com/zwave-js/node-zwave-js/issues/4130).

This is a relatively small appartment and the non-battery nodes are fairely evenly spaced. Node 18 is in the same direction as Node 10, but the latter is nearer to controller (basically they are located on opposite sites of the same wall: 1  -- wall -- "empty" room -- 10 -- wall -- 18 -- other room).

mundschenk-at, 2022-03-13:
For comparison, this is the log from a few days ago using the Gen5+ controller (also Aeotec). Everything is the same except the controller (i.e. same nodes and placement):
[zwavejslog_20220309.zip](https://github.com/zwave-js/node-zwave-js/files/8239777/zwavejslog_20220309.zip)

AlCalzone, 2022-03-14:
> How would increase the controller's TX power?

This is a bit tricky, since no UI exposes it AFAIK. You can do it via code though:
[image]

I suggest the following:
1. Go to the log tab, open log in a new window
2. Go to the driver function dialog, execute this:
   ```js
   await driver.controller.getPowerlevel()
   ```
3. Look at the log and remember the two numbers. Should be something like -3.3 and 0 I think - mine are messed up (outside the -12.8...+12.7 dBm range).
4. execute this:
   ```js
   await driver.controller.setPowerlevel(
     3, // the normal powerlevel plus 3 (or so, need to test)
     0 // the output power at 0 dBm, unchanged!
   )
   ```

I would start with a slight adjustment of the normal powerlevel, e.g. +3 dBm. Note that increasing the powerlevel of a radio device may be against the law, so don't go overboard.

AlCalzone, 2022-03-14:
> The health check on the network map is not working for me (it rapidly swamps the network

Any chance I could see a log of that?

AlCalzone, 2022-03-14:
> I've triggered the health check a few minutes ago

Ok that's https://github.com/zwave-js/node-zwave-js/issues/4075

> What's normal here? GetPowerlevel returned these values

I'm trying to find that out. Unfortunately my test stick has **very** strange values there. Will let you know once I've found the defaults

mundschenk-at, 2022-03-14:
> > What's normal here? GetPowerlevel returned these values
> 
> I'm trying to find that out. Unfortunately my test stick has **very** strange values there. Will let you know once I've found the defaults

Is the power level of a Gen5 stick comparable (or interesting)? Then I could switch back to my other stick.

Also, regarding network strength, do you see similar issues in the [Gen5 log](https://github.com/zwave-js/node-zwave-js/issues/4352#issuecomment-1066105431) or is it the 700 series stick?

AlCalzone, 2022-03-15:
From what I've seen in the NVM data, the settings for 500 and 700 series sticks aren't comparable.

In your Gen5 log, I don't see RSSI for incoming messages, but the ACKs have RSSI values in the 50s/60s, roughly 30-40 dBm higher than on the 700 series. This could be caused by incorrect settings on the 700 series stick.

Start by setting the TX power to:
```
normal powerlevel: 0
measured: +3.3
```
which are the default settings. For good measure also set the RF region to EU, you never know...
```js
await driver.controller.setRFRegion(0)
```

AlCalzone, 2022-03-15:
@mundschenk-at It might be that you're also affected by this bug: https://github.com/zwave-js/node-zwave-js/issues/4367
In that case you won't be able to change the settings on your own without editing the NVM backup first. I can do that for you if you send me the NVM backup via e-mail.

mundschenk-at, 2022-03-15:
> @mundschenk-at It might be that you're also affected by this bug: #4367
> In that case you won't be able to change the settings on your own without editing the NVM backup first. I can do that for you if you send me the NVM backup via e-mail.

Yes it definitely looks that way:
```
2022-03-15T15:12:04.787Z DRIVER « [RES] [SerialAPISetup]
                                    command: GetRFRegion
                                    region:  Unknown
```

AlCalzone, 2022-03-15:
Sure, see here for details: https://github.com/zwave-js/node-zwave-js/issues/3906#issuecomment-1068031298

mundschenk-at, 2022-03-15:
I tried this: Made a new 700 series backup, converted to JSON and replaced `"rfConfig": null` with the object from https://github.com/zwave-js/node-zwave-js/issues/3906#issuecomment-1068031298. It worked, in that `GetRFRegion` returns `Europe` and the power levels are now `0.0 dBm` and `3.3 dBm`. 

However, the whole network started to be reinterviewed immediately. As a result, several nodes had incomplete information and/or security classes. I manually re-interviewed them (using the "reset security classes" option) and was able to fix most of them (two are missing, one is a smoke detector that I would need a ladder to wake up, the other an Aeotec Water Sensor 6, which is quite buggy when using `S0` security - I may have to reinclude that, maybe dropping security this time). Unfortunately, even after this, things are not really good. 

I've tried starting a network heal, but that has stalled after just 4 nodes (i.e. not even all mains powered nodes have been healed).

mundschenk-at, 2022-03-15:
> > Unfortunately, even after this, things are not really good.
> > I've tried starting a network heal, but that has stalled after just 4 nodes .
> 
> Can you post another log so I can see for myself? You don't have to encrypt them, unless you re-include a secure device and/or change user codes while you're logging.

Driver restarts as well (home ID), no? Since these are pretty long logs (whole day), I feel more comfortable with encrypted uploads.

AlCalzone, 2022-03-15:
The home ID is just a number that's transmitted in plaintext via radio anyways. If someone is standing near your network with a Zniffer, they will see it.
Encrypted is fine for me though.

mundschenk-at, 2022-03-15:
Here it is: 
[zwavejslog_20220315.zip](https://github.com/zwave-js/node-zwave-js/files/8256942/zwavejslog_20220315.zip)

(Sorry, a bit messy due to repeated re-interview/heal attempts.)

AlCalzone, 2022-03-15:
From what I'm seeing in the logs, the heal is slowly progressing. You had a full interview of node 99 going on in parallel, which delays the process quite a bit.

The RSSI values didn't change, but that's expected, since you didn't change anything physical. I guess now you can try [this](https://github.com/zwave-js/node-zwave-js/issues/4352#issuecomment-1066524645) and see if the nodes can "hear" the controller better then.

mundschenk-at, 2022-03-15:
> The RSSI values didn't change, but that's expected, since you didn't change anything physical. I guess now you can try [this](https://github.com/zwave-js/node-zwave-js/issues/4352#issuecomment-1066524645) and see if the nodes can "hear" the controller better then.

Just to be clear, do you mean I should initially try `3, 0` as the parameters (as in the example), or rather `3, 3.3` (`3` added to the previous "normal powerlevel" of `0.0`, `0` added to the previous "output power at 0 dBm`)?

mundschenk-at, 2022-03-15:
Here's the new log. I think it is somewhat better, but node 10 still appears to be problematic (though I don't know why, it's a FGWP102, though you might consider it a legacy device - I didn't know how to read the specifications properly when ordering my first nodes). 

Also, when doing a health check for node 11, node 80 started freaking out completely. This one _is_ a bit remote (at the bottom of a cupboard at the other end of the flat), but there are plenty of repeaters in between if necessary.

[zwavejslog_20220315a.zip](https://github.com/zwave-js/node-zwave-js/files/8257418/zwavejslog_20220315a.zip)

AlCalzone, 2022-03-15:
Hm you're still running into the infinite healthcheck loop. Probably best if you wait for the next zwavejs2mqtt beta release, this will include a few fixes for this.

mundschenk-at, 2022-03-15:
Yes, but some health checks were successful (i.e. for nodes 18, 70, 100). Any improvements visible over the previous logs (signal strength-wise)? I have a hard time reading/comparing those.

Node 80 is an Aeotec smart plug set to report pretty frequently so that I know when a relative low-power device is on. I've never seen it spam the network this way, though. To me it appears to start with `2022-03-15T21:36:21.864Z CNTRLR « [Node 080] Beginning Transport Service RX session #14...`, but I don't see what's triggering it.

AlCalzone, 2022-03-16:
> Yes, but some health checks were successful (i.e. for nodes 18, 70, 100). Any improvements visible over the previous logs (signal strength-wise)? I have a hard time reading/comparing those.

We're not going to see improvements in the signal strength. Making the stick "shout louder" doesn't make it "hear better". My hope was that the higher power leads to the acknowledgements reaching the nodes, so they stop re-transmitting the messages over and over.

The health checks are meant as a tool for you to identify nodes with problematic connectivity and potentially re-arrange a few nodes to improve on that. The next zwavejs2mqtt beta release will include fixes for the infinite loops and also show the actual routes used when you click on a node in the network map.

mundschenk-at, 2022-03-16:
> We're not going to see improvements in the signal strength. Making the stick "shout louder" doesn't make it "hear better". My hope was that the higher power leads to the acknowledgements reaching [...]

Sorry, I worded that wrong. But did we see such an improvement? Otherwise, it appears that Z-Stick 7's antenna design (or something) is simply suboptimal compared to the Z-Stick Gen5+.

AlCalzone, 2022-03-16:
I don't think its much better. The routing reports almost always indicate the same nodes though where the route fails: 18 -> 69, 69 -> 80 and 18 -> 80. Might be worth looking into.

mundschenk-at, 2022-03-16:
That's weird. 18 and 69 are in completely different rooms with lots of stuff in between (kitchen appliances) and no way the best routes to take. Same for 69 and 80.

18 -> 80 is somewhat legit (same room), although "80" is the leaf node, location-wise. The arrow indicates the direction, i.e. 18 hops to 80 which should hop to the controller? Or is it the other way around?

AlCalzone, 2022-03-16:
> Or is it the other way around?

It is. These reports indicate the routing from the controller to the nodes.

> 18 and 69 are in completely different rooms with lots of stuff in between (walls and kitchen appliances) and no way the best routes to take. Same for 69 and 80.

Try doing single node heals for those (and other potential nodes along the routes), step by step. Maybe this helps the controller figure out better routes.

mundschenk-at, 2022-03-18:
I've tried healing individual nodes in between, but the routes have not become any better as far as I can see. Yesterday, I switched back to the 500 series stick, which seems to work much better overall. (I am helping Sensative debug the Strips Guard 700 firmware, and I don't want to change too many variables at the same time). I'll raise a ticket with Aeotec what's going on with their 700 series sticks.

Just so that I read the logs correctly, the `ACK RSSI` in this example is what the controller receives from the repeater node (`70`), and `ACKK RSSI on repeaters` is what the repeater node received from the target node?

```
2022-03-18T01:37:12.129Z DRIVER « [REQ] [SendData]
                                    callback id:            61
                                    transmit status:        OK, took 90 ms
                                    repeater node IDs:      70
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -68 dBm
                                    ACK RSSI on repeaters:  -91 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
```

AlCalzone, 2022-03-18:
That's almost correct. For the example you showed it is, but if there are multiple repeaters, you're going to have multiple measurements in the "on repeaters" line. E.g. if the route is (1 -> 70 -> 71 -> 68), then you'll see (for example)
```
repeater node IDs:      70, 71
ACK RSSI:               -68 dBm
ACK RSSI on repeaters:  -91 dBm, -88 dBm
```
where -88 is the RSSI of the ACK that 71 received from 68
and -91 is the  RSSI of the ACK that 70 received from 71
and -68 is the  RSSI of the ACK that the controller received from 70

AlCalzone, 2022-03-18:
> I'll raise a ticket with Aeotec what's going on with their 700 series sticks.

Let me know what they say.

mundschenk-at, 2022-03-18:
Small follow-up question: What's a ”reasonable” RSSI? `-91 dBm` is pretty low, too low?

AlCalzone, 2022-03-18:
AFAIK this depends on the background RSSI. The higher that is, the higher the RSSI should be. The [lifeline health check](https://zwave-js.github.io/node-zwave-js/#/api/node?id=checklifelinehealth) is modeled after what Silabs' PC Controller does and they recommend an RSSI margin (`<RSSI> - <background RSSI>`) of 17 dBm or more.

mundschenk-at, 2022-03-20:
Is there a driver command to dump the routing tables? Aeotec asked if I could provide those.

AlCalzone, 2022-03-20:
The current state is included in the NVM backup. Make a fresh one, then convert it to json and copy all entries for `lwr` and `nlwr`.

mundschenk-at, 2022-04-07:
I am still waiting on meaningful response from Aeotec. Their first level support has made some suggestions for a test setup, but it's not practical investment wise and  I don't think they have looked at the provided logfiles at all.

mundschenk-at, 2022-04-23:
@AlCalzone At Aeotec's advice I've created a test setup with just the 700 controller and a single node. Today I've run health checks with the node in increasing distance from the controller (in 1 m intervals). When I couldn't do straight line-of-sight anymore due to space constraints, I put the node about a further 2 m away, with a thin wall and some furniture (speakers) in between. (That's the measurement from `2022-04-23T09:19:45.952Z`). Then I made an NVM backup to have a look at some of the internal settings after the factory reset. 

Some time later I did a health check again without moving either the controller or the node. The results were pretty weird, as `GetRoutingInfo` now returned loads of neighbors (in a network that never had more than two nodes, and with node numbers not corresponding to any nodes of the ”real” network running on the gen5+ stick either). Is this the controller malfunctioning completely? Some weird interaction because of the backup?

[zwavejs_2022-04-23_weird_routing.log](https://github.com/zwave-js/node-zwave-js/files/8546689/zwavejs_2022-04-23_weird_routing.log)

I'm running the test environment on an RPi4 with docker (`node-zwave-js 9.0.4`).

AlCalzone, 2022-04-26:
The weird neighbors look like random data from the controller memory IMO. This information is stored as a bitmask and maybe the controller is reading the data from a weird location with uninitialized/random memory.

I'm not sure why a backup would cause this though, since it is just reading.

mundschenk-at, 2022-04-26:
Maybe it was not the backup but an incomplete reset (since GetRFRegion returned `unknown`)?

AlCalzone, 2022-04-27:
Could be... if you convert the JSON I sent you back to a binary NVM file and restore that, does `GetRFRegion` return something useful again?

mundschenk-at, 2022-04-27:
I can try to that, but I already called `SetRFRegion`, so now it shows `Europe` already (weirdly, the `output power at 0 dBm` has changed from `14.8 dBm` with region `unknown` to `3.3 dBm` with region `Europe`): [zwavejs_2022-04-23_region_tx_power.log.zip](https://github.com/zwave-js/node-zwave-js/files/8577433/zwavejs_2022-04-23_region_tx_power.log.zip)

If it helps, I can now also provide Zniffer logs (not for this driver log, but going forward).

AlCalzone, 2022-04-28:
That sounds like it already had the other RF values in there, but didn't read them correctly for some reason. Zniffer logs won't help as this communication is USB-only.

mundschenk-at, 2022-05-05:
A partial update on the spamming node 80 mentioned earlier: It appears that node was in a weird state (maybe it didn't like the changing controllers or something), even with my 500 series controller. It was directly accessible by the controller, yet a health check timed out. Healing it didn't change anything about this behavior. After un- and replugging it, suddenly the health check reports a solid 10, as it should.

mundschenk-at, 2022-06-24:
So it looks like it was the stick after all. After a lot of back-and-forth with Aeotec (taking measurements and comparing logfiles), they agreed to send me a new stick from China. With my test node, I got a solid 8 on the health check now, as it should. (I took a backup of the stick before adding the first node, so we can look at the RF settings if those are still of interest to you.)

I'll have to update the firmware and restore a backup from the Gen5+, but it's looking good.

mundschenk-at, 2022-06-24:
@AlCalzone 
Unfortunately, I was too hasty (and forgot to take a separate backup after the firmware update). Like last time, after restoring my network backup from the Gen5+ stick, range was really bad even after healing (non-battery) nodes one by one. I got 10/10 on the health check for 700 series repeater in the same room, but very bad results for the a node next to the test node in another room (which also gets 10/10 with the Gen5+). 

I then tried restoring the "clean" backup from before the firmware update and got the test node to connect again, but with a much worse S/N ratio. I converted the backups to JSON to look at the RF config and found this:

Initial backup:
```
                "rfConfig": {
                        "rfRegion": 0,
                        "txPower": 9.9,
                        "measured0dBm": 0,
                        "enablePTI": null,
                        "maxTXPower": null
                },
```

After firmware update with restored backup (and the test node added again):
```
                "rfConfig": {
                        "rfRegion": 0,
                        "txPower": 9.9,
                        "measured0dBm": 0,
                        "enablePTI": 0,
                        "maxTXPower": 14
                },
```

Here's the log file of the first backup, first test with > 25 dBm and the test after firmware update and restore with  < 18 dBm (some positions etc.): [700-before-and-after-restore.zip](https://github.com/zwave-js/node-zwave-js/files/8980581/700-before-and-after-restore.zip)

For comparison, the Gen5+ stick with another node that sits in the same power outlet gets > 29 dBm S/N.

AlCalzone, 2022-06-27:
Unfortunately I cannot find the password anymore, can you send it again?

Regarding the RF config settings:
Since both are essentially the same, I'm a bit surprised that it makes such a difference. `enablePTI` (useless for our purposes) and `maxTXPower` (AFAIK always 14) are added by a newer firmware, so it is expected that they were `null` before.

`txPower` and `measured0dBm` have somewhat strange values. I'd expect them to be 0.0 (tx power) and +3.3 (measured). Might be worth a try changing them to normal values first.

mundschenk-at, 2022-06-27:
> Unfortunately I cannot find the password anymore, can you send it again?

Sure!

> `txPower` and `measured0dBm` have somewhat strange values. I'd expect them to be 0.0 (tx power) and +3.3 (measured). Might be worth a try changing them to normal values first.

I've experimented a bit with this already, but it doesn't seem to make any difference with my test node. Reported RSSI values are much worse than with the 500 series stick. From [this discussion](https://siliconlabs.force.com/community/s/question/0D58Y00008GEzWtSAL/hellois-it-possible-to-improve-power-of-new-z-stick-700-or-do-something-with-it-i-replaced-the-z-stick-500-but-from-that-moment-some-devices-turn-dead-each-time-and-then-come-back-again), I gather I'm not the only one experiencing something like this.

mundschenk-at, 2022-07-02:
@AlCalzone, did you receive the PW?

AlCalzone, 2022-07-05:
Yeah I did. What strikes me as a bit odd is that the second test has a worse SNR margin, but the latency and route changes are better than the first test. The difference in SNR margin is caused by a change in ACK RSSI which is ~10 dBm higher for the first tests. I'm not sure what causes this - maybe the calibration changed somehow?

That said, I did my own measurements with 4 different controllers yesterday (Z-Stick 7, UZB 3 and 7, and a Razberry 7 Pro). The Z-Stick 7 performed significantly worse than the others (20-35% less range, 15-30 dBm worse SNR margin) and had troubles including a node in some locations where the others had no issues.

mundschenk-at, 2022-07-05:
OK, so it looks a like this is an inherent feature of the Z-Stick 7, i.e. a bad design (maybe too small a package for a good antenna)?

mundschenk-at, 2022-07-21:
Anything I can do/measure to help the investigation?

AlCalzone, 2022-07-22:
I did a couple of tests with a node in a fixed location where I changed the powerlevel settings of the stick:
TX power: -9.9; 0.0; +9.9
measured: -9.9; 0.0; +9.9
(combination of both)
In addition I checked TX power +9.9 / measured -2.3, which was recommended by Aeotec.

Each time I did a health check with 10 rounds and took the mean and std. deviation of the SNR margin.

You could check what this does for yours - in my case the changes were insignificant. Repeating the set of tests later yielded a much larger difference than the settings themselves.

mundschenk-at, 2022-07-30:
Just so that I understand the intent correctly, I should measure powerlevel combinations like this:
* ( -9.9 / -9.9 )
* ( -9.9 / 0.0 )
* ( -9.9 / +9.9 )
* ...
* ( +9.9 / +9.9 )

I'm asking because trying to set the powerlevel to (-9.9 / 0.0 ) results in a reading of ( 15.7 / 0.0 ) when returning to the config values from another screen in Z2M.

mundschenk-at, 2022-07-30:
OK, I've just run the whole series. Without actually calculating variance or standard deviation, the SNR sets did not look any different to me. Some where a bit better or worse, but not in any way that I would call "significant" from eyeballing it. If it helps, I can provide the actual measurements. 

I did notice that I could net set negative values for either of the parameters even when using the driver function directly. Trying to set a negative value returned a positive value upon `await driver.controller.getPowerlevel()`, with no obvious correlation between the input and output values.

AlCalzone, 2022-08-17:
Weird, I didn't get any notification that you answered. I'll let Aeotec know that you were able to reproduce my findings.

fhriley, 2022-08-29:
I'm also having the low SNR issues with an Aeotek z-stick 7. I'm upgrading from a Nortek HUSBZB-1, which as far as I know had no SNR issues since it worked great for the few years I used it. With the Aeotek I'm unable to get through a full interview on my front door lock. Power levels in zwavejs2mqtt are reported in the -85-91 dBm range. No surprise it's not working well. Even with direct LOS with < 15 ft, the SNR is poor. I've got a Zooz 700 series arriving today to see if it rectifies the low SNR. 

> txPower and measured0dBm have somewhat strange values. I'd expect them to be 0.0 (tx power) and +3.3 (measured).

Mine reports the same as @mundschenk-at, 9.9 and 0, respectively.

[7 more replies]

closed 2023-04-19 by AlCalzone