zwave-js/zwave-js #4359: [Bug]: Qubino smart light switch relay missing

thecobra666, 2022-03-14:
### Checklist

- []  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Zwavejs2Mqtt version

6.5.2

### ZwaveJS version

zwave-js: 8.11.6

### Describe the bug

I am unable to control the relay inside the device. There should be an endpoint, endpoint1 and endpoint2. The latter should be the relay.

Now I am only able to control the led light. The binary switch disables the led instead of the relay switch. 

2nd bug parameter: [46-112-0-3] Touch Functionality: Disable Alarm: is set to Enabled (1) but is not listed as default. Yet the manual states that 1 is the default.

Parameter no 3. – turning off alarming
Values (size is 1-byte dec):
• Default value 1

```
2022-03-13 10:15:42.887 INFO ZWAVE: Node 46: value updated: 37-0-currentValue false => true
2022-03-13 10:15:43.120 INFO ZWAVE: Node 46: value added: 32-0-currentValue => 255
2022-03-13 10:15:43.143 INFO ZWAVE: Node 46: value added 46-32-0-currentValue => 255
2022-03-13 10:15:44.826 INFO ZWAVE: Node 46: value updated: 50-0-value-66049 0 => 38.9
2022-03-13 10:15:53.350 INFO ZWAVE: Node 11: metadata updated: 49-0-Power
2022-03-13 10:15:53.353 INFO ZWAVE: Node 11: value updated: 49-0-Power 167 => 139.1
2022-03-13 10:15:57.877 INFO ZWAVE: Node 46: value updated: 37-0-currentValue true => false
```

This is me turning on the relay with the touch controls and trying to turn it off again with the button in zwavejs2mqtt. It targets the same value yet, it is different. I'm unsure what happend here. Already tried doing a re-interview. This was working before I did a docker upgrade from 6.5.0 to 6.5.2. 

### To Reproduce

Nvt

### Expected behavior

Have the ability to control the relay. 

### Additional context

[node_46.txt](https://github.com/zwave-js/zwavejs2mqtt/files/8239287/node_46.txt)
[Luxy_Smart_Switch.pdf](https://github.com/zwave-js/zwavejs2mqtt/files/8239289/Luxy_Smart_Switch.pdf)
_

AlCalzone, 2022-03-14:
@thecobra666 you probably forgot this?
https://github.com/zwave-js/node-zwave-js/issues/4062

> Qubino devices have the ability to enable/disable endpoints via configuration parameters. However, the changes only show up after excluding and re-including the device. This is written very prominently in the manual, but I'm guessing many users don't read them if they have the nice config parameters in the UI.
Excluding and re-including should be the first troubleshooting step after re-interviewing when Qubino devices and missing endpoints values/entities appear together.

thecobra666, 2022-03-14:
> @thecobra666 you probably forgot this? #4062

Hi, this is not mentioned in the manual but is indeed required for the mini dimmer. There are also no settings I needed to change on the device. It's installed as is and was working as such. But to rule it out, I'll exclude and include the device again tonight and report back. If not okay I'll enable the silly debug log and attach it also.

AlCalzone, 2022-03-14:
It could also be that the driver determined the endpoints to be unnecessary and does not expose them. Make sure to capture a driver log (level debug) of the inclusion (or maybe a re-interview before that), so I can check if that's the case.

thecobra666, 2022-03-14:
[zwavejs_2022-03-14.log](https://github.com/zwave-js/node-zwave-js/files/8248121/zwavejs_2022-03-14.log)
[zwavejs2mqtt_2022-03-14.log](https://github.com/zwave-js/node-zwave-js/files/8248123/zwavejs2mqtt_2022-03-14.log)

Attached the requested logs. I have re-interviewed it more then once and excluded - included it again. It keeps detecting as node 46. I'm also unable to get it back in unless I click re-interview and reset security classes

AlCalzone, 2022-03-14:
Hmm, there is a bit of chaos in the log, at the very least two parallel interviews of the same node - however you managed that 😅. It doesn't look like the exclusion reset the device, so you'll have to factory reset it again after excluding.

Please make another log when you include again after that, ideally delete the old logfile beforehand.

thecobra666, 2022-03-15:
[zwavejs_2022-03-14.log](https://github.com/zwave-js/node-zwave-js/files/8251102/zwavejs_2022-03-14.log)
[zwavejs2mqtt_2022-03-14.log](https://github.com/zwave-js/node-zwave-js/files/8251103/zwavejs2mqtt_2022-03-14.log)

I was not able to clear the logs before. I did do a factory reset and included it again. Same issue.

AlCalzone, 2022-03-15:
(removed your logs because they contain the network key)

AlCalzone, 2022-03-15:
@zwave-js-bot add compat flag to 0x0159:0x0008:0x0052:0.0
```jsonc
// This device has two endpoints which are considered "unnecessary",
// but the root endpoint doesn't expose all their functionality.
preserveEndpoints: "*"
```

AlCalzone, 2022-03-15:
@thecobra666 Can you try [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/546a47fd9090142ed718f83f749e852ba0b024ad/packages/config/config/devices/0x0159/zmnkad1.json) as [described here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files).

If it works, I can include the change in the next release.

thecobra666, 2022-03-15:
> @thecobra666 Can you try [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/546a47fd9090142ed718f83f749e852ba0b024ad/packages/config/config/devices/0x0159/zmnkad1.json) as [...]

Will test tonight and report back.

thecobra666, 2022-03-15:
I receive this error when dropping the code. 
 error 
2022-03-15T19:09:29.119Z CONFIG   version 8.11.6
2022-03-15T19:09:31.890Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2022-03-15T19:09:32.205Z CONFIG   Error parsing config file zmnkad1.json: Could not parse config file /usr/src/a
                                  pp/store/config/templates/master_template.json: ENOENT: no such file or direct
                                  ory, open '/usr/src/app/store/config/templates/master_template.json'
                                  Import stack: 
                                    in #state_after_power_failure_off_prev
                                    in /usr/src/app/store/config/zmnkad1.json (ZW0340)
2022-03-15T19:09:32.439Z CNTRLR   querying controller IDs...
2022-03-15T19:09:32.462Z CNTRLR   received controller IDs:
                                    home ID:     0xcd00c016
                                    own node ID: 1
2022-03-15T19:09:32.462Z CNTRLR   querying API capabilities...
2022-03-15T19:09:32.480Z CNTRLR   received API capabilities:

AlCalzone, 2022-03-15:
Sorry, I already merged. Can re-open if the file doesn't fix the issue.

I always forget to mention that, you need to [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/templates/master_template.json) too in a subfolder `templates` next to the file you're testing.

thecobra666, 2022-03-15:
> Sorry, I already merged. Can re-open if the file doesn't fix the issue.
>
> I always forget to mention that, you need to [this [...]

It took a restart after asking the interview but now it's working fine. Thanks! Maybe update the docs to include the template file?

AlCalzone, 2022-03-15:
> Maybe update the docs to include the template file?

When I get to it at some point, I'll make it so the embedded config DB will be used as a fallback. Then that's not necessary anymore.

PR #4366 (2022-03-15): fix(config): preserve all endpoints for Qubino ZMNKAD1 Luxy Smart Switch
fixes: #4359

**TODO:**
- [x]	Change PR title to be more specific