zwave-js/zwave-js #4362: [question] Fibaro Electric Consumption and power slow to update

Rouzax, 2022-03-14:


Rouzax, 2022-03-14:
> Maybe it's worth getting the plugin (?) devs onboard, is it https://github.com/moroen/ZWave-MQTT-plugin ?

You mean opening an issue there?

AlCalzone, 2022-03-14:
> You mean opening an issue there?

Or asking the dev(s) to join the discussion here and figure out what needs to change where and how. Not sure who's the right one to talk to for this, do you know who?

Rouzax, 2022-03-14:
I've reached out to the Slack channel, thinking about opening an issue on https://github.com/domoticz/domoticz/issues

Rouzax, 2022-03-14:
@AlCalzone, Would be useful to have your input over there as well.

gizmocuz, 2022-03-16:
Issue is solved by finding a Watt sensor looking at the **unique_id** .... (prioritizing command class 49 (COMMAND_CLASS_SENSOR_MULTILEVEL) in front of 50 (COMMAND_CLASS_METER)

AlCalzone, 2022-03-16:
Great! Closing here then.

closed 2022-03-16 by AlCalzone