zwave-js/zwave-js #4363: FIBARO Walli Double Switch FGWDSEU-221: "Parameter 40" and "Parameter 41" cannot be set to ">8"

kysr, 2022-03-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
It is not possible to set configuration parameter 40 and 41 of the FIBARO Walli Double Switch FGWDSEU-221 to a value in range 0..15. It is possible to set this parameter 40 and 41 in range 0..8, but that does not cover all capabilities.

What do you observe?
In the manual of the device:
https://manuals.fibaro.com/wp-content/uploads/2019/06/FGWDSEU-221-T-EN-v1.2-1.pdf
Page 30 shows that values 1, 2, 4 and 8 can be added to each other, resulting in a value in range 0..15.

In the documentation Z-Wave JS Config DB Browser:
https://devices.zwave-js.io/?jumpTo=0x010f:0x1b01:0x1000:0.0
Parameter 40 and 41 can only be 1, 2 ,4 and 8. This is wrong.

What did you expect to happen?
To be able to set the parameter of the device.

Steps to reproduce the behavior:
  1. In Home Assistant preferences select the device,
  2. Enter Z-Wave Device Configuration,
  3. Try to set parameter 40 or 41 to a value > 8
  4. An error occurred. Must provide a value within the target range (Min: 0, Max: 8)

### Device information

Manufacturer:  Fibaro
Model name: FGWDSEU-221
Node ID in your network: 14

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.52

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

The logfile is not needed.

kysr, 2022-03-14:
I know I did not provide a logfile, but I don't know where to find it. And I think it offers no added value for the investigation of the problem.

AlCalzone, 2022-03-14:
Parameters 40 and 41 are both modeled as 4 separate "partial" parameters, which each set one of the 4 bits, so all values from 0 to 15 are possible to be entered (although indirectly):
[image]

Not sure which version of the driver and configuration DB is in use, maybe you need to update?

kysr, 2022-03-14:
Then I misinterpreted the documentation of Z-Wave JS Config DB Browser. Sorry. I am rather new to this software...

I now updated from  Z-Wave JS  0.1.52 to 0.1.54 which uses DB Version: 8.11.6. I re-interviewed the device.

In Home Assistant: 
This does work:
service: zwave_js.set_config_parameter
data:
  parameter: '40'
  value: '8'
target:
  device_id: <device-id>

But this does not work:
service: zwave_js.set_config_parameter
data:
  parameter: '40'
  value: '1'
  bitmask: '0x08'
target:
  device_id: <device-id>
Gives: "Failed to call service zwave_js.set_config_parameter. Unknown error"

In the Home Assistant Log, I can see:

2022-03-14 21:46:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547411547376] Error handling message: Unknown error
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 27, in _handle_async_response
await func(hass, connection, msg)
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 534, in handle_execute_script
await script_obj.async_run(msg.get("variables"), context=context)
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1253, in async_run
await asyncio.shield(run.async_run())
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 354, in async_run
await self._async_step(log_exceptions=False)
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 372, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 575, in _async_call_service_step
await service_task
File "/usr/src/homeassistant/homeassistant/core.py", line 1630, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1667, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 354, in async_set_config_parameter
zwave_value, cmd_status = await async_set_config_parameter(
File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 65, in async_set_config_parameter
raise NotFoundError(
zwave_js_server.exceptions.NotFoundError: Configuration parameter with value ID 14-112-0-40-8 could not be found

AlCalzone, 2022-03-15:
I've asked in the HomeAssistant Discord if someone can help you out with this, as I'm not familiar with the details here.

kpine, 2022-03-15:
@kysr Please download and attach the Diagnostic Data for this device (accessible from the Device page in HA).

kysr, 2022-03-15:
@kpine This is the Diagnostics Data of the device:

[zwave_js-ddc3bfaf1801e8128d3d330448105449-WALLI SWITCH FIBARO SINGLE SWITCH-d65784307617795b12b8dc6521032850.json.txt](https://github.com/zwave-js/node-zwave-js/files/8255148/zwave_js-ddc3bfaf1801e8128d3d330448105449-WALLI.SWITCH.FIBARO.SINGLE.SWITCH-d65784307617795b12b8dc6521032850.json.txt)

kpine, 2022-03-15:
It's not the same device. This is the URL the diag dump shows. https://devices.zwave-js.io/?jumpTo=0x010f:0x1a01:0x1000:5.2

AlCalzone, 2022-03-15:
So is this a double switch or a single switch? The diag dump does seem to indicate its a single switch.

kpine, 2022-03-15:
Doesn't seem like there's such a thing as a "Single Switch". OZW calls it `FGWDS221 Walli Double Switch (Single switch mode)`. Looks like your device config was imported from [Open Smarthouse](https://opensmarthouse.org/zwavedatabase/1094), and even thre the attached manuals are for a "Double Switch".

It is confusing, because the ZWA DB says "FIBARO Walli Single Switch" in the description, but the manuals are still for the "Double Switch". Also doesn't help that Fibaro itself markets it simple as "Walli Switch", despite the manual.

AlCalzone, 2022-03-15:
Yeah, the updated config file also mentions both Single Switch and Double Switch mode.

AlCalzone, 2022-03-15:
@kysr next update will contain the change, don't forget to re-interview after you pick that up.

closed 2022-03-15 by AlCalzone

kysr, 2022-03-15:
@AlCalzone Thx for the update. 

If you go to https://devices.zwave-js.io/, it is possible to see DB Version 8.11.7. But is it possible to see the same documentation of DB Version 8.11.6 somewhere? (the version that Home Assistant uses at this time)

kpine, 2022-03-15:
> It is this device: FIBARO WALLI DOUBLE SWITCH FGWDSEU-221
> https://devices.zwave-js.io/?jumpTo=0x010f:0x1b01:0x1000:0.0

That is not your device. Your device has the `0x1a01` product type, which goes to this DB entry: https://devices.zwave-js.io/?jumpTo=0x010f:0x1a01:0x1000:5.2. They may have the same product name, but they have different IDs. If you look at the config parameters (as of right now), they don't match the device you are looking at, which is why it doesn't work.

kysr, 2022-03-15:
Ah okay. I did not notice... I just looked at the product type on the box of the device...
[image]

I did not know a the device can get another product type...

PR #4370 (2022-03-15): fix(config): merge redundant config files for Fibaro Walli Double Switch
fixes: #4363