zwave-js/zwave-js #4365: missing RF unlock operation notification

phairplay, 2022-03-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
RF unlock operation notification only appears if the door was previously RF Locked.
What do you observe?
listening to the zwave notification nothing appeared when remote unlocking. i locked the door manually and noticed notifications were working.
What did you expect to happen?
RF unlock operation notification to appear every time
Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error
[Zwave.log](https://github.com/zwave-js/node-zwave-js/files/8248447/Zwave.log)
ive attached a file that shows me trying to capture it remote unlocking. you will see manually locked appears twice, in between that i remote unlocked the door. yet when tried remote locking then unlocking i saw a notification stating so.

### Device information

Manufacturer: Yalw
Model name: Conexis L1
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.54
`node-zwave-js` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

`Event 14 fired 18:01:
{
    "event_type": "zwave_js_notification",
    "data": {
        "domain": "zwave_js",
        "node_id": 3,
        "home_id": 3809197039,
[... 328 lines cut ...]
        "user_id": null
    }
}
Event 0 fired 17:45:
{
    "event_type": "zwave_js_notification",
    "data": {
        "domain": "zwave_js",
        "node_id": 3,
        "home_id": 3809197039,
        "device_id": "399c54aefbc5207f692af385e964c814",
        "command_class": 113,
        "command_class_name": "Notification",
        "label": "Access Control",
        "type": 6,
        "event": 1,
        "event_label": "Manual lock operation",
        "parameters": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-03-14T17:45:45.591414+00:00",
    "context": {
        "id": "c1e6d7769db91e07b1ea3c93656c23b1",
        "parent_id": null,
        "user_id": null`

phairplay, 2022-03-14:
uploaded log from integration

AlCalzone, 2022-03-15:
First off: The driver does not generate any notifications unless the device sends them. Inconsistent behavior like this would have to be fixed in the device firmware.

I'm not sure I follow your description though. In the log I see two consecutive RF unlock attempts, both times the lock sends a Door Lock Report and a Notification with alarmType 25.
Then at the end there is a RF lock attempt, but this time there's no automatic report from the node at all and the driver validates the change by querying the lock status.

phairplay, 2022-03-17:
apologies I may not of given you the correct information in the logs.

I posted the problem on HA forum and someone else mentioned they had the same issue, and another stated everything was fine. 
https://community.home-assistant.io/t/yale-z-wave-module-2-and-usb-z-wave-stick/64073/366

Would I be correct in saying the issue would be the same regardless if I was running ZWAVE-JS or ZWAVE-JSMQTT?

AlCalzone, 2022-03-17:
> Would I be correct in saying the issue would be the same regardless if I was running ZWAVE-JS or ZWAVE-JSMQTT?

Both share the same Z-Wave implementation (this one), so yes.

phairplay, 2022-03-17:
Thank you
Also thank you for your support. I'll close this issue as it doesn't seem to be caused by zwave js

closed 2022-03-17 by phairplay