zwave-js/zwave-js #4407: Firmware Update Timeout (30seconds) too low for some deices on a busy network

warthog9, 2022-03-28:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Attempting to do a firmware update on an Inovelli LZW60

What do you observe?

It starts sending packets (initiated from zwavejs2mqtt) runs for a bit fine, and then the firmware update hits, cleans up and a few seconds later gets a firmware request but zwave-js doesn't think that there's an ongoing firmware update and force aborts it.

What did you expect to happen?

Ideally it not time out

Steps to reproduce the behavior:
  1. Initiate update from zwavejs2mqtt
  2. Wait
  3. Firmware update times out

log entries:

2022-03-28T04:01:34.309Z CNTRLR   [Node 188] Firmware update timed out
2022-03-28T04:01:40.367Z CNTRLR « [Node 188] Received Firmware Update Get, but no firmware update is in progress
                                  . Forcing the node to abort...
                                  
I'd propose changing the timeout (30000ms or 30seconds) to at least 1minute, but more likely 5minutes, as 30 seconds seems to be just tight enough that the update can't complete, and I'm sure the LZW60 isn't the only device that may run into this.  In my own testing changing it to 20minutes (yes I recognize that's a smidge ridiculous) seems to resolve the issue

### Device information

Manufacturer: Inovelli
Model name: LZW60
Node ID in your network: 188

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: whatever zwavejs2mqtt pulls in, the timeout is identical in master when I looked
`zwavejs2mqtt` branch: master

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I made a change to node_modules/zwave-js/build/lib/node/Node.js to change the timeout to timeoutFirmwareUpdate from 30000 to  1.2e6 ms to test this, otherwise this is basically unchanged

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[log_output.txt](https://github.com/zwave-js/node-zwave-js/files/8359742/log_output.txt)

warthog9, 2022-03-28:
I'll note the log file is just the relevant portion of the firmware update for node 188, the rest of the traffic is somewhat irrelevant to this discussion

AlCalzone, 2022-03-31:
We can definitely investigate increasing the timeout to help with this.

This does sound a bit like intermittent connectivity issues of the switch though. The log doesn't contain any relevant details, since it's not on loglevel debug, but the delay between fragments 2 and 3 is already cutting it very close:
```
2022-03-28T03:56:25.875Z CNTRLR » [Node 188] Sending firmware fragment 2 / 4981
2022-03-28T03:56:49.945Z CNTRLR » [Node 188] Sending firmware fragment 3 / 4981
```
It could also be that the controller is waiting for something else here, not sure.

Do note that a firmware update is mostly controlled by the device that gets updated, which individually requests each firmware fragment. Z-Wave JS just considers the process to be stuck or aborted when it doesn't hear from the device for a while. The fact that it (probably) sometimes takes >20s to send a message to the controller is a bit bothersome IMO.

Have you used the mesh health check tools in zwavejs2mqtt yet? They can be found on the network map.acting up.

warthog9, 2022-04-11:
Quick check of the nodes (now that they are re-flashed and back in the network) looks like they are healthy and fine. (6,7,8,8,7 scores) so I don't think it was specifically an issue with the general health of the network (minor update)

AlCalzone, 2022-04-12:
Scores of 6 or 7 might already cut it a bit close - low latency but possibly not a strong link.
What are the min. powerlevel and SNR margins?

AlCalzone, 2023-06-08:
I literally quoted a part of the log where it took 20 seconds to reach the device 🤷🏻‍♂️

twasilczyk, 2023-06-08:
Sorry, as I'm re-reading what I wrote I might not express this clearly: I don't think it ever takes 20 seconds to send every fragment, not a few odd ones (and the rest takes hundreds of milliseconds). The rest of my response doesn't deny that sometimes it takes 30+, but rather tries to explain why it happens.

closed 2023-06-12 by zwave-js-bot

twasilczyk, 2023-07-09:
The change was finally pushed to zwave-js-ui - it works! Thank you very much, I was finally able to update my entire network without much fuss.

PR #5908 (2023-06-12): fix: increase fragment timeout during OTA update to 2 minutes
fixes: #4407