zwave-js/zwave-js #4413: Query CC versions for CCs that only exist on endpoints

AlCalzone, 2022-03-29:
See https://github.com/zwave-js/node-zwave-js/discussions/4402 for context

Essentially the device doesn't expose the `Multilevel Sensor CC` on the root device, so the driver never queries the CC version.
As a result, the endpoints are interviewed with `Multilevel Sensor CC V1`, which only queries the default sensor reading.

espenthingstad, 2022-03-29:
I see that each time I do a reload of the Zwave JS integration, then the values on the sensor are updated. 
See attached e.g. the consumption trendline.
[image]

AlCalzone, 2022-03-31:
> each time I do a reload of the Zwave JS integration, then the values on the sensor are updated

I'm not sure if HA re-queries the values then or why that is. I suggest opening an issue in https://github.com/home-assistant/core for this.

The issue you described in https://github.com/zwave-js/node-zwave-js/discussions/4402 will be fixed in the upcoming v9 release.

PR #4419 (2022-03-31): fix: query CC versions from endpoints if the node doesn't support the CC
fixes: #4413
closed 2022-03-31 by AlCalzone