zwave-js/zwave-js #4415: Sensor ZP3103 from Vision Security report not correctly interpreted

zanna1972, 2022-03-29:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The binary sensor entity, related to the shock sensor, doesn't change status from on to off.

What do you observe?
I set the zwave-js registry in debug mode, then I shake my device.
I can see two "events" within 10 seconds.
The first changes the value from 0 to 255 and the entity in Home Assistant changes from Secure to Intrusion.
The second event changes the value from 255 to 0 but the entity still remain in status Intrusion.

What did you expect to happen?
I expected that the status changes from Intrusion to Secure when the second event arrives.

This is the Zwave-js log:
[Problem ZS5101 change status.log](https://github.com/zwave-js/node-zwave-js/files/8373505/Problem.ZS5101.change.status.log)

### Device information

Manufacturer: Vision Security (Id: 265)
Model name:  ZS5101 (Id: 770)
Node ID in your network: 016

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Versione del driver: 8.11.9
Versione del server: 1.15.0
ID Home: 3591723216
URL del server: ws://core-zwave-js:3000

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[Problem ZS5101 change status.log](https://github.com/zwave-js/node-zwave-js/files/8373518/Problem.ZS5101.change.status.log)

zanna1972, 2022-03-29:
Hi, sorry I renamed the file but it's the log file I downloaded from Zwave-js registry in debug mode.
Is it good for you?

kpine, 2022-03-29:
Does the entity state not reset to off after 5 minutes?

Also, what is the device class? You can find this by downloading the Device Diagnostic file from the HA Device page.

zanna1972, 2022-03-29:
Hi kpine,
Yes the entity changes state to off after 5 minutes. A lot of time!
The device class is:
deviceClass": {
            "basic": {
              "key": 4,
              "label": "Routing Slave"
            },
            "generic": {
              "key": 32,
              "label": "Binary Sensor"
            },
            "specific": {
              "key": 1,
              "label": "Routing Binary Sensor"
            },

kpine, 2022-03-29:
> What do you mean for "other disabled sensors"? In this device?

In HA. What are the disabled sensors?

zanna1972, 2022-03-29:
There are the following entities in my device:
Sensor to detect shake
Sensor to detect cover remove
Sensor for air temperature (this device has not this sensor)
Binary sensor for shake (this is the one I'm using)
Binary sensor called Any (I don't know what is)
Sensor for battery
Binary sensor for low battery 
Sensor for status
Number basic (I don't know what is)

kpine, 2022-03-29:
Adding the [`enableBasicSetMapping`](https://zwave-js.github.io/node-zwave-js/#/config-files/file-format?id=enablebasicsetmapping) compat flag to the device configuration file would probably fix this. In that case, the "Any" binary_sensor is what you would use instead.

Unfortunately, since you aren't using zwavejs2mqtt, you can't test such a change. I would suggest attaching a log of the interview process. You can do that in HA by having two windows, one showing the Z-Wave JS Log window, and in the other window clicking the Re-Interview button on the device page.

zanna1972, 2022-03-29:
Sorry, I'm not sure to have understood all your explanation...
The enableBasicSetMapping is something you can add in the next release of Zwave-js?
At the moment the "any" binary sensor doesn't work at all. Never changes its status.
I don't know zwavejs2mqtt is.

kpine, 2022-03-29:
Yes, the flag could be added to a next release of Z-Wave JS. It it's the correct thing to do, it would make the Any sensor work properly.

I would suggest attaching a log of the interview process. You can do that in HA by having two windows, one showing the Z-Wave JS Log window, and in the other window clicking the Re-Interview button on the device page.

zanna1972, 2022-03-29:
Ok, probably I have the same problem with device ZG8101 from the same manufacturer. So, if you will modify the zwavejs version, please take into consideration also this model.
Tomorrow I will upload to this issue the full log of zwave registry in debut mode (from the shake until the off event, after 5 minutes) and also the HA log during interview. Is this what you need, correct? 
Thanks a lot

kpine, 2022-03-29:
Only need the interview log.

I have a ZG8101. It works fine.

zanna1972, 2022-03-29:
Ok, I try!
Thanks a lot

zanna1972, 2022-03-30:
Good morning,
I read something on Google to understand what ZWAVEJS2MQTT is. It's an alternative add-on to ZWAVE JS. Correct?
I'm actually using ZWAVE JS, do you suggest me to replace it with ZWAVEJS2MQTT? is there a how-to document?
I take a look at my ZG8101 device, I don't have an "any" entity and it works in the same way as ZP3103, I mean it switch to off after 5 minutes.
I followed your suggestion: opened two windows, one on the device where I click the re-interview button and the other on the Zwave-js log in debug mode. Then I shaked the sensor... in the log I only see what I have already attached to this issue: valure from 0 to 255 and 10 seconds later value from 255 to 0. But the binary sensor still remain in on.

zanna1972, 2022-03-30:
Hi AlCalzone,
I have cleaned the log I sent yesterday to you in order to evidence that there are two basicCCSet not only one:

**2022-03-29T16:38:52.229Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 255
2022-03-29T16:38:52.234Z CNTRLR   [Node 016] treating BasicCC::Set as a report
2022-03-29T16:38:52.236Z CNTRLR   [Node 016] [~] [Basic] currentValue: 0 => 255                     [Endpoint 0]

2022-03-29T16:39:04.918Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 0
2022-03-29T16:39:04.922Z CNTRLR   [Node 016] treating BasicCC::Set as a report
2022-03-29T16:39:04.923Z CNTRLR   [Node 016] [~] [Basic] currentValue: 255 => 0                     [Endpoint 0]**

If the message BasicCCSet is the responsible to change the binary_sensor status (this is what I have understood, but probably is not correct) I expected that the binary sensor return off after 10 seconds.

As I wrote before, I tryed to create the log for re-interview, but I'm not able to obtain it.
This is the procedure I followed:
- Open Home assistant and go to the device corresponding the node 016.
- Open another Home assistant and go to Integration, Zwave JS, logs panel
- In logs panel select log level=debug
- Click the re-interview button of the device
- Shake the device to wake it up
- Wait to see something in the log panel concerning node 016 and download the log

Unfortunately nothing was written in the log panel concerning re-interview.
Is there another procedure to obtain the log you need?

zanna1972, 2022-03-30:
Hi kpine, ok I'll search a document to migrate from zwave js add-on to zwavejs2mqtt add-on.

Concerning the ZG8101 I have found the binary sensor you told me, and I have understood how it works.
When the device is placed vertically or horizontally the status is off, but if I place it horizontally but flipped over it turns to on.
This is fantistic to know if the garage door is open or close.
But I want to use this device also to detect shake in my alarm (If anyone tries to break my door to get into the garage) and I need to use the other binary sensor that detect this kind of movement... but the entity stay on for 5 minutes.

kpine, 2022-03-30:
> But I want to use this device also to detect shake in my alarm (If anyone tries to break my door to get into the garage) and I need to use the other binary sensor that detect this kind of movement... but the entity stay on for 5 minutes.

The ZG8101 is a tilt sensor, not a shock sensor. The other binary_sensor that stays on (named something like `Garage Door Tilt Sensor: Home Security - Sensor status`) is just reporting the same tilt events, however that one does not have any possible way to reset except for the 5-minute idle. You cannot work around that, so it's mostly useless. In fact, I disable that entity in HA, and only rely on the "Any" sensor.

zanna1972, 2022-03-30:
Ok, I agree with you, I disable the other binary sensor in ZG8101.
Concerning the other device ZP3103, I explain what happens: I bought 4 of them from the same vendor. He told me that are both ZS5101.
When included in Zwave network... 3 are ZP3103 and 1 is a ZS5101. They are externally identical.
The ZS5101, when shaked has a binary sensor that changes from off to on and come back to off after 10 seconds.
The ZP3101, as I already wrote, when shaked has a binary sensor that changes from off to on and come back to off after 5 minutes.
I want to send you the re-interview log, but I'm not able to obtain it. I wrote to AlCalzone in this issue the procedure I followed to obtain the log... probably is wrong because I don't see anything. Could you help me? Thanks a lot

AlCalzone, 2022-03-30:
> * Shake the device to wake it up

I'm preeeeeetty certain that does not wake it up and just causes a notification to be sent. Check the device manual how to actually wake it up.

zanna1972, 2022-03-30:
Ok, tomorrow morning I'll try.

closed 2022-04-11 by zwave-js-assistant[bot]