zwave-js/zwave-js #4417: [Bug] ConfigManager.lookupNotification(Type).lookupValue(Event) returning undefined for Event = 0

marcus-j-davies, 2022-03-29:
As discussed on slack.

2 examples (I am currently not including the entry if `lookupValue` yields `undefined`

Kwikset / Door Lock
https://devices.zwave-js.io/?jumpTo=0x0090:0x0001:0x0436:0.0

BRK Brands, Inc. / Smoke and Carbon Monoxide Alarm
https://devices.zwave-js.io/?jumpTo=0x0138:0x0001:0x0002:0.0

kpine, 2022-03-29:
Is this the reason why the ZCOMBO page is not showing mappings for alarm type 13?

marcus-j-davies, 2022-03-29:
@kpine,
That be correct.

`lookupValue` yields the metadata for the event, but is `undefined` for these instances.

```json
{
  "from": {
    "alarmType": 13,
    "alarmLevel": 255
  },
  "to": {
    "notificationType": 1,
    "notificationEvent": 0
  }
}

```

AlCalzone, 2022-03-30:
@marcus-j-davies I think the table needs another revision:
a) I think we should display the variable name along with the state's label:
[image]
Some of the notification states in the specs are poorly named so they don't make much sense without the `variableName`.
I propose:

| Notification Type | Notification Event |
| --- | --- |
| Smoke Alarm (0x01) | Sensor status: _Smoke detected - (0x02)_ |

b) `"notificationEvent": 0` is indeed **idle**, but it isn't possible to look it up like the others. Without any further context, 0 idles all variables of the same notification type, so the above example means Smoke Alarm -> idle. In the case that there are multiple states under the same notification type, **all** of them would go to idle. I don't think we have this case with the existing mappings though. With proper notifications there would be a possibility to selectively idle a single variable (V5+ if I remember correctly).
So, you'll have to special-case the 0 though and not look it up.

marcus-j-davies, 2022-03-31:
@AlCalzone 

that's all  done - I want to try and tidy it up however a bit later.
I think maybe this can be closed?

closed 2022-03-31 by AlCalzone