zwave-js/zwave-js #4418: Missing device configuration: FAKRO ZWK10

carom411, 2022-03-30:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

FAKRO ZWK10

### Manufacturer ID

0x0085

### Product Type

0x0001

### Product ID

0x0005

### Firmware Version

1.1

### Is the device listed on the Z-Wave Alliance website?

no

### Do you have a manual?

[zwk10.pdf](https://github.com/zwave-js/node-zwave-js/files/8378627/zwk10.pdf)

carom411, 2022-04-01:
I have find out the way how to operate with the ZWK10 remote controller. I have tested it and it works fine, it just do not show the battery level information.

`{
	"manufacturer": "Fakro",
	"manufacturerId": "0x0085",
	"label": "ZWK10",
	"description": "Z-Wave Wall Controller",
	"devices": [
		{
			"productType": "0x0001",
			"productId": "0x0005"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"associations": {
		"1": {
			"label": "Channel \"1234\" left",
			"maxNodes": 232
		},
		"2": {
			"label": "Channel \"1234\" right",
			"maxNodes": 232
		},
		"3": {
			"label": "Channel \"1\" left",
			"maxNodes": 232
		},
		"4": {
			"label": "Channel \"1\" right",
			"maxNodes": 232
		},
		"5": {
			"label": "Channel \"2\" left",
			"maxNodes": 232
		},
		"6": {
			"label": "Channel \"2\" right",
			"maxNodes": 232
		},
		"7": {
			"label": "Channel \"3\" left",
			"maxNodes": 232
		},
		"8": {
			"label": "Channel \"3\" right",
			"maxNodes": 232
		},
		"9": {
			"label": "Channel \"4\" left",
			"maxNodes": 232
		},
		"10": {
			"label": "Channel \"4\" right",
			"maxNodes": 232
		}
	}
}`

AlCalzone, 2022-04-01:
These remotes are often very basic in terms of functionality - and sometimes can't even be interviewed. I wouldn't be surprised if this just isn't possible.

austwhite, 2023-12-21:
I was looking at one of these controls as I have a couple of Fakro ZWS12 winders, which work flawlessly in Zwave JS by the way.
Did anyone work out how to get this controller to function in ZWave JS?