zwave-js/zwave-js #4450: Missing device configuration: HEATIT TF021

rathmannm, 2022-04-07:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

HEATIT

### Manufacturer ID

0x019b

### Product Type

0x0001

### Product ID

0x0001

### Firmware Version

1.8

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1234?selectedFrequencyId=-1

### Do you have a manual?

[Manual_Heatit_Z-Wave_FW 1.8_Ver2019-A-TF-021.pdf](https://github.com/zwave-js/node-zwave-js/files/8446575/Manual_Heatit_Z-Wave_FW.1.8_Ver2019-A-TF-021.pdf)

AlCalzone, 2022-04-08:
@zwave-js-bot import config 1234 from zwa

AlCalzone, 2022-04-08:
Hmm - looks like the IDs are identical with the TF016 we already have in the database.
https://products.z-wavealliance.org/products/1182?selectedFrequencyId=1

They even link to the same manual.

closed 2022-04-08 by AlCalzone

nesheimt, 2024-10-24:
The Heatit TRM2 thermostats is somewhat of a mess, because there is a "unofficiall" fw(isssued by Thermoflor) v1.92 that alters the thermostat. In zwave JS all TRM2 thermostats used the same driver, but I believe that that will not work. The official documentation can be found here: 
https://github.com/Heatit-Controls/no.thermofloor/tree/main/drivers/TF_Thermostat_FW1.92

AlCalzone, 2024-10-25:
That does not look like documentation

PR #4452 (2022-04-08): feat(config): extend Heatit TF016 config to include TF021
fixes: #4450

**TODO:**
- [x]	Change PR title to be more specific