zwave-js/zwave-js #4453: [feat]

PeteRager, 2022-04-08:


AlCalzone, 2022-04-08:
@PeteRager can you post a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) of you switching the fan? Loglevel `debug` or `silly` and attach it here as a file.
I want to confirm a suspicion.

PeteRager, 2022-04-08:
Here are two log files.  The difference in the second log file, is I have implemented application polling after the fan command in an HA automation - and we can see after the application poll the right value is reported.

[jasco fan off.txt](https://github.com/zwave-js/node-zwave-js/files/8452489/jasco.fan.off.txt)

[jasco fan off with app polling.txt](https://github.com/zwave-js/node-zwave-js/files/8452493/jasco.fan.off.with.app.polling.txt)

The automation is overly aggressive, since all it can trigger on is a value change of fan state, whenever it changes it poll the value.  So the 6->0 tirggers it, then the 0->6 triggers it, then since the first application poll returns a 6->0 that triggers it again.  Very messy....

Agree that the device isn't behaving in the best manner.  Seems like the code writer is reflecting the command back, then reflecting the current state before the command.  So yeah, i'd expect there's a bug in there that z-wave cert doesn't test for.

PR #4569 (2022-05-05): fix: cancel verification poll only when expected value is received
With this PR, we no longer blindly cancel value verification polls when an unsolicited report is received. Instead, for each scheduled poll, an expected value can/will be saved and value updates only cancel the poll when the value was changed to this expected value (assuming one is set).

fixes: #3523
fixes: #4453
closed 2022-05-05 by AlCalzone