zwave-js/zwave-js #4457: Dlink DCH-Z110 Window Sensor not working

danreg, 2022-04-10:
Today I've migrated my home assistant from deprecated zwave integration to zwave js integration.
All seems working fine except my DLINK DCH-Z110 Window / Door Sensors.

Air Temperature, Illuminance and Node Status seems to work. So I can see when device is sleeping or active. Also values for illuminance and temperature are changing.
But the important sensor to get the information wether window is opened or not is not working. With old zwave integration it was working. I guess that the raw values of the sensor are not interpreted in the right way.
Is there any possibility to debug the raw values?

AlCalzone, 2022-04-11:
You somehow managed to open the issue without filling out an issue template. If you'd selected "Bug Report", you would have been asked to provide the information necessary to debug this, including a driver log which includes what the driver sees from the device. So please try that again.

M0rfic, 2022-09-27:
Hi @AlCalzone,

I'm not exactly sure what the workflows you guys are using, so apologies if this is not the correct way to pursue this issue and please let me know if you'd rather continue here or something else.

Nonetheless, I may be able to provide more information as I found myself in the same situation. For starters, you can find below the logs relative to moving the sensor away from the magnet (similar thing if you bring it close) with all default settings. In this case it appears that 1 big message with a summary of all included sensor readings is generated, but it can't currently be handled. Even if the temp, or luminance is updated in HA, the door/window state remains `unknown`:

```
14:52:57.700 SERIAL « 0x0126000400051e8f010403800364097105000000ff06160005310503010706310 (40 bytes)
5010a004d7e0004
14:52:57.709 SERIAL » [ACK] (0x06)
14:52:57.715 CNTRLR [Node 005] [~] [Battery] level: 100 => 100 [Endpoint 0]
2022-09-27 14:52:57.718 DEBUG Z-WAVE: Node 5: mapping value(s) of property 'level' (level) from CC 128 (Battery) to node property 'minBatteryLevel
2022-09-27 14:52:57.719 DEBUG Z-WAVE: Node 5: mapping value(s) of property 'level' (level) from CC 128 (Battery) to node property 'batteryLevels
2022-09-27 14:52:57.721 INFO Z-WAVE: Node 5: value updated: 128-0-level 100 => 100
14:52:57.724 CNTRLR [Node 005] [~] [Battery] isLow: false => false [Endpoint 0]
2022-09-27 14:52:57.725 INFO Z-WAVE: Node 5: value updated: 128-0-isLow false => false
14:52:57.729 CNTRLR [Node 005] [~] [Notification] alarmType: 0 => 0 [Endpoint 0]
2022-09-27 14:52:57.731 INFO Z-WAVE: Node 5: value updated: 113-0-alarmType 0 => 0
14:52:57.736 CNTRLR [Node 005] [~] [Notification] alarmLevel: 0 => 0 [Endpoint 0]
2022-09-27 14:52:57.737 INFO Z-WAVE: Node 5: value updated: 113-0-alarmLevel 0 => 0
14:52:57.742 CNTRLR [Node 005] [Multilevel Sensor] Illuminance: metadata updated [Endpoint 0]
2022-09-27 14:52:57.744 INFO Z-WAVE: Node 5: metadata updated: 49-0-Illuminance
14:52:57.746 CNTRLR [Node 005] [~] [Multilevel Sensor] Illuminance: 9 => 7 [Endpoint 0]
2022-09-27 14:52:57.747 INFO Z-WAVE: Node 5: value updated: 49-0-Illuminance 9 => 7
14:52:57.751 CNTRLR [Node 005] [Multilevel Sensor] Air temperature: metadata updated [Endpoint 0]
2022-09-27 14:52:57.753 INFO Z-WAVE: Node 5: metadata updated: 49-0-Air temperature
14:52:57.755 CNTRLR [Node 005] [~] [Multilevel Sensor] Air temperature: 78 => 77 [Endpoint 0]
2022-09-27 14:52:57.761 INFO Z-WAVE: Node 5: value updated: 49-0-Air temperature 78 => 77
14:52:57.764 DRIVER « [Node 005] [REQ] [ApplicationCommand]
└─[MultiCommandCCCommandEncapsulation]
14:52:57.766 DRIVER Received a command that contains multiple CommandClasses. This is not supported yet! Discarding the message...
14:52:57.767 CNTRLR « [Node 005] TODO: no handler for application command
```

I then went and changed under `Configuration V1` the `[5-112-0-7-32] Disable Multi CC in auto report` setting to `[1] Disable`, and now it seems that multiple smaller messages are generated and handled correctly, including the state of the door/window, and I can see all the updated values in HA:

```
15:12:24.617 SERIAL « 0x010b00040005038003647e006f (13 bytes)
15:12:24.633 SERIAL » [ACK] (0x06)
15:12:24.639 CNTRLR [Node 005] [~] [Battery] level: 100 => 100 [Endpoint 0]
2022-09-27 15:12:24.641 DEBUG Z-WAVE: Node 5: mapping value(s) of property 'level' (level) from CC 128 (Battery) to node property 'minBatteryLevel
2022-09-27 15:12:24.642 DEBUG Z-WAVE: Node 5: mapping value(s) of property 'level' (level) from CC 128 (Battery) to node property 'batteryLevels
2022-09-27 15:12:24.645 INFO Z-WAVE: Node 5: value updated: 128-0-level 100 => 100
15:12:24.650 CNTRLR [Node 005] [~] [Battery] isLow: false => false [Endpoint 0]
2022-09-27 15:12:24.652 INFO Z-WAVE: Node 5: value updated: 128-0-isLow false => false
15:12:24.655 DRIVER « [Node 005] [REQ] [ApplicationCommand]
└─[BatteryCCReport]
level: 100
is low: false
15:12:24.666 SERIAL « 0x011100040005097105000000ff0617007e0002 (19 bytes)
15:12:24.669 SERIAL » [ACK] (0x06)
15:12:24.674 CNTRLR [Node 005] [~] [Notification] alarmType: 0 => 0 [Endpoint 0]
2022-09-27 15:12:24.677 INFO Z-WAVE: Node 5: value updated: 113-0-alarmType 0 => 0
15:12:24.683 CNTRLR [Node 005] [~] [Notification] alarmLevel: 0 => 0 [Endpoint 0]
2022-09-27 15:12:24.685 INFO Z-WAVE: Node 5: value updated: 113-0-alarmLevel 0 => 0
15:12:24.688 DRIVER « [Node 005] [REQ] [ApplicationCommand]
└─[NotificationCCReport]
notification type: Access Control
notification status: 255
notification state: Window/door is closed
15:12:24.693 CNTRLR [Node 005] [~] [Notification] Access Control[Door state]: 22 => 23 [Endpoint 0]
2022-09-27 15:12:24.695 INFO Z-WAVE: Node 5: value updated: 113-0-Access Control-Door state 22 => 23
15:12:24.709 SERIAL « 0x010d000400050531050301057e00bb (15 bytes)
15:12:24.711 SERIAL » [ACK] (0x06)
15:12:24.714 CNTRLR [Node 005] [Multilevel Sensor] Illuminance: metadata updated [Endpoint 0]
2022-09-27 15:12:24.715 INFO Z-WAVE: Node 5: metadata updated: 49-0-Illuminance
15:12:24.717 CNTRLR [Node 005] [~] [Multilevel Sensor] Illuminance: 5 => 5 [Endpoint 0]
2022-09-27 15:12:24.719 INFO Z-WAVE: Node 5: value updated: 49-0-Illuminance 5 => 5
15:12:24.721 DRIVER « [Node 005] [REQ] [ApplicationCommand]
└─[MultilevelSensorCCReport]
type: Illuminance
scale: Percentage value
value: 5
15:12:24.727 SERIAL « 0x010e00040005063105010a004c7e00fb (16 bytes)
15:12:24.729 SERIAL » [ACK] (0x06)
[... cut ...]
15:12:24.733 CNTRLR [Node 005] [~] [Multilevel Sensor] Air temperature: 76 => 76 [Endpoint 0]
2022-09-27 15:12:24.734 INFO Z-WAVE: Node 5: value updated: 49-0-Air temperature 76 => 76
15:12:24.736 DRIVER « [Node 005] [REQ] [ApplicationCommand]
└─[MultilevelSensorCCReport]
type: Air temperature
scale: Fahrenheit
value: 76
```
P.S. please be aware that I slightly manipulated the log to make sense, as sometimes it looks a bit weird when it wraps some of the lines, eg:
[image]

If there is any other information you require, please feel free to ask away.

BR

AlCalzone, 2022-09-27:
Oh, yeah that is helpful. You'll notice that the value updates do get through, but commands like the `Notification CC` that need more fine-grained handling do not get handled when inside a `Multi Command CC`. That should be straightforward though. The tricky thing would be handling nested commands that expect a response, because the responses would have to be collected and returned as one.

danreg, 2022-09-27:
I found in iobroker forum that setting notification type from notification report to sensorbinaryreport. But I was not able to change this configuration yet. (Was not applied to the device, tried to set this several times)

M0rfic, 2022-09-27:
@danreg I believe you only need to change the setting I suggested `Configuration V1` > `[5-112-0-7-32] Disable Multi CC in auto report setting` > `[1] Disable`. Remember to wake-up the device after clucking the apply/arrow button for the setting.

[image]

AlCalzone, 2023-04-19:
This should be fixed since v10.10

closed 2023-04-19 by AlCalzone