zwave-js/zwave-js #4458: GE Enbrighten Z-Wave Smart Hinge Pin Door Sensor (Jasco 32563)

F0rged, 2022-04-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
I upgraded from home assistant "ZWave" integration to "Z-WaveJS" integration.

**What do you observe?**
I was able to pair my GE/Jasco 32563 Hinge door pin sensor, however only the battery level is exposed.
The state of the door (open/closed) is not available.

**What did you expect to happen?**
I expected the state of the door (open/closed) to be available.

Steps to reproduce the behavior:
1. Install home assistant Z-WaveJS Integration
2. Pair GE/Jasco 32563 hinge door pin sensor
3. Note that the state of the door is not available.

### Device information

Manufacturer: GE / Jasco
Model name: 32563
Node ID in your network: 8

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version: 8.11.9
Server Version: 1.15.0
Devices ready: 7 / 12

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I upgraded from the zwave integration to Z-WaveJS

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It worked fine with the home assistant ZWave integration
[... 340 lines cut ...]
              },
              {
                "id": 122,
                "name": "Firmware Update Meta Data",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 90,
                "name": "Device Reset Locally",
                "version": 1,
                "isSecure": false
              }
            ],
            "interviewStage": "Complete",
            "deviceDatabaseUrl": "https://devices.zwave-js.io/?jumpTo=0x0063:0x4953:0x3032:5.20",
            "statistics": {
              "commandsTX": 6,
              "commandsRX": 3,
              "commandsDroppedRX": 0,
              "commandsDroppedTX": 1,
              "timeoutResponse": 0
            },
            "highestSecurityClass": -1
          },

F0rged, 2022-04-10:
> 👋 Hey @F0rged!
>
> Thanks for opening an issue! It doesn't look like you provided a logfile though. While not strictly necessary for every issue, having a [driver [...]

Thanks, I tried and this is all I seem to get, even though the log level is set to "Silly".

`Subscribed to Z-Wave JS Log Messages…`

AlCalzone, 2022-04-11:
> Thanks, I tried and this is all I seem to get, even though the log level is set to "Silly".
> 
> `Subscribed to Z-Wave JS Log Messages…`

That means that there was no communication. Battery powered devices generally need to be woken up for this.

Please try again with the logfile, this time trigger a re-interview and then wake up the device. The manual should tell you how to do that - usually requires pressing buttons on the device.

F0rged, 2022-04-12:
I rebooted and checked again today.  The extra sensors showed up.
Thanks!

closed 2022-04-12 by F0rged