zwave-js/zwave-js #4459: Missing device configuration: Cherubini ORA ZRX

katiuskt, 2022-04-11:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Cherubini ORA ZRX

### Manufacturer ID

0x0455

### Product Type

0x0004

### Product ID

0x0001

### Firmware Version

10.16

### Do you have a manual?

Product manual for this device (motor for awnings, pergolas and screens): [ORA_A4518_6N14_EN.pdf](https://github.com/zwave-js/node-zwave-js/files/8466004/ORA_A4518_6N14_EN.pdf)
Product URL: https://www.cherubini-group.com/en/products/motors-8/ora-rx-406

Cherubini has a lot more of ZWave products (https://www.cherubini-group.com/en/products/home-automation-systems-17/meta-320) but I can't see further information of any of them into the ZWave Alliance DB.

Please, find below, debug information from my ZwaveJS2MQTT system. I tried reading "Contributing configuration files" but I'm afraid it is definitely beyond my knowledge.
However, I will try my best if further details would be needed.
```
{
  "id": 25,
  "name": "Awning",
  "loc": "Office",
  "values": [
    {
      "id": "25-38-0-targetValue",
      "nodeId": 25,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "Target value",
      "stateless": false,
      "commandClassVersion": 4,
      "min": 0,
      "max": 99,
      "list": false,
      "value": 99,
      "lastUpdate": 1649566471983,
      "newValue": 99
    },
    {
      "id": "25-38-0-duration",
      "nodeId": 25,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "duration",
      "propertyName": "duration",
[... 882 lines cut ...]
  "isControllerNode": false,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 17,
    "specific": 7
  },
  "deviceId": "1109-1-4",
  "status": "Alive",
  "interviewStage": "Complete",
  "statistics": {
    "commandsTX": 1,
    "commandsRX": 0,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0
  },
  "lastActive": 1649566471995,
  "_name": "Office_Awning"
}
```

AlCalzone, 2022-04-12:
> I've just created this file but not sure if it might be enough (not sure how to test it in my current installation -Home Assistant Addon-)

You can test it using the zwavejs2mqtt addon, as described here:
https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files

It does look like it should work, but we cannot merge it as-is, since it doesn't conform to the [style guide](https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide).

Ideally, read that, apply it to your file, then open a PR with the new file. This way it can be automatically checked and we can give you instructions what to change where.
Here's an introduction to making pull requests: https://docs.github.com/en/get-started/quickstart/contributing-to-projects

> But PDF says parameter 35 is read-only.
> 
> I mention it because both parameters 33 and 36 are detected as read-only automatically by ZWaveJS which is right but wondering why parameter 35 doesn't

Without a config file, zwave-js tries to get the configuration info from the device itself. This is often suboptimal, e.g. have no labels like here. In this case it also didn't declare the parameter 35 as readonly it seems.

katiuskt, 2022-04-12:
Thank you @AlCalzone for your comments.

I tried my best to follow the guides but quite newbie here. Looking for further comments to keep learning.

closed 2022-04-20 by AlCalzone