zwave-js/zwave-js #4465: Issues with Leviton Decora Smart Z-Wave 600W Dimmer (DZ6HD)

F0rged, 2022-04-13:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The Light state appears in home assistant, but it always shows as "off".

What do you observe?
When I try toggle the switch on (physically), the home assistant device still shows as "off".
When I try toggle the switch on (via home assistant), the state goes to "on" briefly, then back to "off". Meanwhile the actual light itself never illuminates.

What did you expect to happen?
I expect the home assistant device state to reflect the actual state of the light
I expect the light to illuminate when I turn the device on from the app.
Everything worked fine on the old ZWAVE addon.  The problem began after upgrading to ZWaveJS.
I have re-interviewed the device and rebooted the system.  No effect.

Steps to reproduce the behavior:
1. Try to toggle the light on from the app, or vice versa.

### Device information

Manufacturer:  Leviton
Model name: Decora Smart Z-Wave 600W Dimmer (DZ6HD)
Node ID in your network: 14

Z-Wave Info
Device ID: 14
Device Status: Alive
Device Ready: Yes
Highest Security: None
Z-Wave Plus: Version 1

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version: 8.11.9
Server Version: 1.15.0
Home ID: 4282397224
Devices ready: 11 / 12

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I upgraded from the ZWAVE addon to the ZWaveJS Addon.
Everything was working perfectly before.

I see others are having similar issues.  See below post:
https://community.home-assistant.io/t/leviton-dz6hd-polling-not-working-under-z-wave-js/356753/9

### Attach Driver Logfile

[zwave_js (2).log](https://github.com/zwave-js/node-zwave-js/files/8477925/zwave_js.2.log)

AlCalzone, 2022-04-13:
This is what I can gather from your current log:
```
Node tells zwave-js to re-query its status:
2022-04-13T03:04:52.081Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[HailCC]
zwave-js does so:
2022-04-13T03:04:52.087Z DRIVER » [Node 014] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      41
                                  └─[MultilevelSwitchCCGet]
...
but the device doesn't know its current status:
2022-04-13T03:04:52.122Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: undefined
```

5 seconds later, the same happens multiple times and each time the node answers with a different value, which looks like it is dimming down (roughly 10% every half second):
```
2022-04-13T03:04:57.298Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 87
                                      target value:  87
                                      duration:      1s
2022-04-13T03:04:57.865Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 74
                                      target value:  74
                                      duration:      1s
2022-04-13T03:04:58.449Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 61
                                      target value:  61
                                      duration:      0s
2022-04-13T03:04:58.882Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 48
                                      target value:  48
                                      duration:      1s
... etc, down to 0:
2022-04-13T03:05:01.449Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 0
                                      target value:  0
                                      duration:      0s
```

Without knowing what should have happened instead, this doesn't look wrong to me.
I will need some additional, commented logs where you describe what you're doing when, e.g. like this:
```
Turn on the light physically:
... logs here ...

Turn on the light via HA:
... logs here ...
```

F0rged, 2022-04-15:
I have no idea why, but its working now....

F0rged, 2022-07-16:
Well, sometime since April, maybe after one of the upgrades (not sure which one), but it seems to be broken again.
Similar issue as before.  Its seems that home assistant can communicate with the light, but it fails to process the light's response and update the state of the light.

Here is a breakdown of what I am seeing:

---------------STATUS------------------
Actual State of Light Itself: Off
light.in_wall_600w_dimmer: "Off"
---------------------------------------

---------------Action Performed--------
Press Light Button in home assistant.
Expected behaviour: Actual light turns on, and "light.in_wall_600w_dimmer" changes to "Off"
---------------------------------------

---------------Start of ZWAVE Log-----------
Subscribed to Z-Wave JS Log Messages…
2022-07-16T20:31:38.083Z SERIAL » 0x010b001312042601ffff252bd8                                        (13 bytes)
2022-07-16T20:31:38.085Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      43
                                  └─[MultilevelSwitchCCSet]
                                      target value: 255
                                      duration:     default
2022-07-16T20:31:38.087Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T20:31:38.091Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T20:31:38.092Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:38.092Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T20:31:38.106Z SERIAL « 0x010500132b00c2                                                     (7 bytes)
2022-07-16T20:31:38.107Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:38.108Z DRIVER « [REQ] [SendData]
                                    callback id:     43
                                    transmit status: OK
2022-07-16T20:31:38.117Z SERIAL « 0x01080004001202820160                                              (10 bytes)
2022-07-16T20:31:38.118Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:38.118Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[HailCC]
2022-07-16T20:31:38.120Z CNTRLR   [Node 018] Hail received from node, refreshing actuator and sensor values...
2022-07-16T20:31:38.121Z CNTRLR » [Node 018] requesting current switch state...
2022-07-16T20:31:38.127Z SERIAL » 0x0109001312022602252cd8                                            (11 bytes)
2022-07-16T20:31:38.128Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      44
                                  └─[MultilevelSwitchCCGet]
2022-07-16T20:31:38.131Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T20:31:38.134Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T20:31:38.135Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:38.135Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T20:31:38.149Z SERIAL « 0x010500132c00c5                                                     (7 bytes)
2022-07-16T20:31:38.150Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:38.151Z DRIVER « [REQ] [SendData]
                                    callback id:     44
                                    transmit status: OK
2022-07-16T20:31:38.160Z SERIAL « 0x010b00040012052603646402c0                                        (13 bytes)
2022-07-16T20:31:38.161Z CNTRLR   [Node 018] [~] [Multilevel Switch] duration: {"value":1,"unit":"s [Endpoint 0]
                                  econds"} => {"value":2,"unit":"seconds"}
2022-07-16T20:31:38.162Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:38.164Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: undefined
2022-07-16T20:31:43.118Z SERIAL » 0x0109001312022602252dd9                                            (11 bytes)
2022-07-16T20:31:43.119Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      45
                                  └─[MultilevelSwitchCCGet]
2022-07-16T20:31:43.121Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T20:31:43.125Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T20:31:43.126Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:43.127Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T20:31:43.140Z SERIAL « 0x010500132d00c4                                                     (7 bytes)
2022-07-16T20:31:43.141Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:43.144Z DRIVER « [REQ] [SendData]
[... cut ...]
2022-07-16T20:31:43.153Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:31:43.154Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: undefined

---------------END OF ZWAVE Log-----------

---------------STATUS------------------
Actual State of Light Itself: On
light.in_wall_600w_dimmer: "Off"
---------------------------------------

F0rged, 2022-07-16:
Since home assistant always thinks the light is off, there is no way to turn it off using the home assistant dashboard.
However, if I call the service manually (), the light will turn off.

`Subscribed to Z-Wave JS Log Messages…
2022-07-16T20:45:54.948Z SERIAL » 0x010b00131204260100ff252e22                                        (13 bytes)
2022-07-16T20:45:54.949Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      46
                                  └─[MultilevelSwitchCCSet]
                                      target value: 0
                                      duration:     default
2022-07-16T20:45:54.952Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T20:45:54.955Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T20:45:54.957Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:45:54.958Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T20:45:54.971Z SERIAL « 0x010500132e00c7                                                     (7 bytes)
2022-07-16T20:45:54.972Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:45:54.973Z DRIVER « [REQ] [SendData]
                                    callback id:     46
                                    transmit status: OK
2022-07-16T20:45:54.978Z CNTRLR   [Node 018] [~] [Multilevel Switch] currentValue: 0 => 0           [Endpoint 0]
2022-07-16T20:45:54.981Z SERIAL « 0x01080004001202820160                                              (10 bytes)
2022-07-16T20:45:54.982Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:45:54.983Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[HailCC]
2022-07-16T20:45:54.985Z CNTRLR   [Node 018] Hail received from node, refreshing actuator and sensor values...
2022-07-16T20:45:54.986Z CNTRLR » [Node 018] requesting current switch state...
2022-07-16T20:45:54.994Z SERIAL » 0x0109001312022602252fdb                                            (11 bytes)
2022-07-16T20:45:54.995Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      47
                                  └─[MultilevelSwitchCCGet]
2022-07-16T20:45:54.997Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T20:45:55.000Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T20:45:55.001Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:45:55.002Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T20:45:55.016Z SERIAL « 0x010500132f00c6                                                     (7 bytes)
2022-07-16T20:45:55.016Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:45:55.017Z DRIVER « [REQ] [SendData]
                                    callback id:     47
                                    transmit status: OK
2022-07-16T20:45:55.027Z SERIAL « 0x010b00040012052603000001c3                                        (13 bytes)
2022-07-16T20:45:55.028Z CNTRLR   [Node 018] [~] [Multilevel Switch] targetValue: 0 => 0            [Endpoint 0]
2022-07-16T20:45:55.029Z CNTRLR   [Node 018] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                                  econds"} => {"value":1,"unit":"seconds"}
2022-07-16T20:45:55.030Z CNTRLR   [Node 018] [~] [Multilevel Switch] currentValue: 0 => 0           [Endpoint 0]
2022-07-16T20:45:55.031Z CNTRLR   [Node 018] Scheduled poll canceled because expected value was received
2022-07-16T20:45:55.031Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T20:45:55.032Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 0
                                      target value:  0
                                      duration:      1s
`

F0rged, 2022-07-16:
Not being an expert on this library, my eyes tend to gravitate to this:

After the light is turned off, the device sends this message with the current state of the light, and it is processed correctly.
```
2022-07-16T20:45:55.032Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 0
                                      target value:  0
                                      duration:      1s
```

But when the light is turned on, the device sends this message with the current state of the light, and it is "undefined", and processed incorrectly.
```
2022-07-16T20:31:43.154Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: undefined
```

F0rged, 2022-07-16:
Its also worth mentioning, the device is currently on the following firmware:
Firmware: 1.17
Device ID: 18
Device Status: Alive
Device Ready: Yes
Highest Security: None
Z-Wave Plus: Version 1

F0rged, 2022-07-16:
So I ran the [firmware update to v1.22](https://decorasmartsupport.zendesk.com/attachments/token/MFT4C61yHdtG43RDv3QkVHANg/?name=DZ6HD_v1_22.ota) from post [4101 ](https://github.com/zwave-js/node-zwave-js/issues/4101) and hopefully Im not speaking too soon, but it seems to have resolve the issue..

Here is the response after turning the light on:
```
Subscribed to Z-Wave JS Log Messages…
2022-07-16T21:18:02.049Z SERIAL » 0x010b001312042601ffff2544b7                                        (13 bytes)
2022-07-16T21:18:02.050Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      68
                                  └─[MultilevelSwitchCCSet]
                                      target value: 255
                                      duration:     default
2022-07-16T21:18:02.052Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T21:18:02.056Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T21:18:02.057Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T21:18:02.058Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T21:18:02.071Z SERIAL « 0x010500134400ad                                                     (7 bytes)
2022-07-16T21:18:02.072Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T21:18:02.073Z DRIVER « [REQ] [SendData]
                                    callback id:     68
                                    transmit status: OK
2022-07-16T21:18:02.082Z SERIAL « 0x01080004001202820160                                              (10 bytes)
2022-07-16T21:18:02.083Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T21:18:02.084Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[HailCC]
2022-07-16T21:18:02.086Z CNTRLR   [Node 018] Hail received from node, refreshing actuator and sensor values...
2022-07-16T21:18:02.087Z CNTRLR » [Node 018] requesting current switch state...
2022-07-16T21:18:02.092Z SERIAL » 0x01090013120226022545b1                                            (11 bytes)
2022-07-16T21:18:02.093Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      69
                                  └─[MultilevelSwitchCCGet]
2022-07-16T21:18:02.095Z SERIAL « [ACK]                                                                   (0x06)
2022-07-16T21:18:02.099Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-07-16T21:18:02.100Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T21:18:02.101Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-07-16T21:18:02.114Z SERIAL « 0x010500134500ac                                                     (7 bytes)
2022-07-16T21:18:02.115Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T21:18:02.116Z DRIVER « [REQ] [SendData]
                                    callback id:     69
                                    transmit status: OK
2022-07-16T21:18:02.125Z SERIAL « 0x010b00040012052603636302c0                                        (13 bytes)
2022-07-16T21:18:02.126Z CNTRLR   [Node 018] [~] [Multilevel Switch] targetValue: 255 => 99         [Endpoint 0]
2022-07-16T21:18:02.127Z CNTRLR   [Node 018] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                                  econds"} => {"value":2,"unit":"seconds"}
2022-07-16T21:18:02.128Z CNTRLR   [Node 018] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
2022-07-16T21:18:02.129Z CNTRLR   [Node 018] Scheduled poll canceled because expected value was received
2022-07-16T21:18:02.130Z SERIAL » [ACK]                                                                   (0x06)
2022-07-16T21:18:02.131Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 99
                                      target value:  99
                                      duration:      2s

```

The final message is no longer "undefined", but is actually showing values correctly now:
```
2022-07-16T21:18:02.131Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 99
                                      target value:  99
                                      duration:      2s
```

closed 2022-07-16 by F0rged